AltBeacon/android-beacon-library #772: Beacon detection doesn't start on OnePlus 6(API 28) even if the app is in foreground

chintan-mishra, 2018-11-12:
### Expected behavior
Beacon should be detected when the app is opened.

### Actual behavior
`BeaconConsumer` binds to the service but doesn't detect any beacon until after the phone is locked and opened up again.

### Steps to reproduce this behavior
Set `mBeaconManager.setEnableScheduledScanJobs(true);` and the detection will take place once every five minutes. In order to detect a beacon start lock your device by pressing the button and unlock it by entering the pin/pattern/etc.

However, setting `mBeaconManager.setEnableScheduledScanJobs(false);` starts the detection as soon as the app is opened. But this is not desired as [our app](https://8hoot.com) relies on scheduled scans for optimal user experience.

### Mobile device model and OS version
OnePlus 6 with OxygenOS 9.0.2 based off of Android P

### Android Beacon Library version
`implementation 'org.altbeacon:android-beacon-library:2.15.2'`

davidgyoung, 2018-11-13:
@chintan-mishra, I'd like to get this working on OxygenOS 9.0.2 but it may be tough since I do not have a test device.  Think you could help me with the troubleshooting?  If so, I'd suggest running the reference app with beaconManager.setDebug(true); then verify you see the same thing and capture a LogCat excerpt for about 10 seconds after scanning should start.

chintan-mishra, 2018-11-13:
I will love to help you troubleshoot the problem. Sure, I will capture the LogCat and share it.

On a different note, I was able to source a Xiaomi Redmi Y2(Android O, 8.1, API 27) of a friend. I am experiencing similar behavior in this device too.

[My application class where `BeaconManager`'s instance is created and used throughout the app](https://pastebin.com/igrGEnBD)
[The service which detects the Beacon](https://pastebin.com/6Wa37mdU). **Editing line 16 starts the scan**
[BeaconScanner class which implements the `BeaconConsumer` interface](https://pastebin.com/nq0f3U8D)

chintan-mishra, 2018-11-13:
@davidgyoung 

[Logs while detecting beacon by monitoring activity](https://pastebin.com/GWd1gguE)
[Logs while detecting beacon by ranging activity](https://pastebin.com/k50M2dMD)

Beacon was started using Locate app by Radius Networks on a Moto G4+ device

chintan-mishra, 2018-11-13:
This issue was my fault. I apologize for a false alarm @davidgyoung 
[Line 78 in this class was called every time the beacon started scanning and it made library perceive that the app was in background](https://pastebin.com/nq0f3U8D)

On an unrelated note, background scanning still doesn't take place on OnePlus 6 using the scheduled scans.

closed 2018-11-13 by chintan-mishra

davidgyoung, 2018-11-14:
Ok, @chintan-mishra glad you found it.

Do you want to open a new issue on the background scheduled scans not working?  It may very well be that OxygenOS is cancelling the jobs scheduled every 15 mins to do scans.    But if so, perhaps you could see if there is a log line confirming this.  If would be good to document the expected behavior.  Even if we cannot find a workaround, a new issue would serve to help document this for others.

chintan-mishra, 2018-11-14:
@davidgyoung 

I have added background scanning issue in my product's bug database. I will report my findings hopefully before Christmas. In the background, Xiaomi is scanning once every 30 mins or more.

I found [a relevant article(redirects to medium.com)](https://medium.com/mindorks/enable-background-services-in-chinese-roms-32e73dfba1a6) where the author somehow managed to overcome similar restrictions by Chinese ROMs including OxygenOS, MIUI, etc.

476rick, 2020-03-31:
I'm still having this issue, my OnePlus 6 won't find any beacons. In contrast, an app like Locate does find the beacons... Can someone help me with this?

I made sure that backgroundMode is false. 

```
2020-03-31 11:57:33.611 25259-25259/com.package D/CycledLeScanner: starting a new scan cycle
2020-03-31 11:57:33.611 25259-25259/com.package D/CycledLeScanner: We are already scanning and have been for 4564 millis
2020-03-31 11:57:33.612 25259-25259/com.package D/CycledLeScanner: Waiting to stop scan cycle for another 1100 milliseconds
2020-03-31 11:57:33.612 25259-25259/com.package D/CycledLeScanner: Scan started
2020-03-31 11:57:34.624 25259-25259/com.packageD/CycledLeScanner: Waiting to stop scan cycle for another 88 milliseconds
2020-03-31 11:57:34.752 25259-25259/com.package D/CycledLeScanner: Done with scan cycle
2020-03-31 11:57:34.752 25259-25259/com.package D/ScanHelper: Beacon simulator not enabled
2020-03-31 11:57:34.757 25259-25259/com.package D/CycledLeScanner: Not stopping scan because this is Android N and we keep scanning for a minimum of 6 seconds at a time. We will stop in 292 millisconds.
2020-03-31 11:57:34.757 25259-25259/com.package D/CycledLeScanner: starting a new scan cycle
2020-03-31 11:57:34.757 25259-25259/com.packageD/CycledLeScanner: We are already scanning and have been for 5710 millis
2020-03-31 11:57:34.757 25259-25259/com.package D/CycledLeScanner: Waiting to stop scan cycle for another 1100 milliseconds
2020-03-31 11:57:34.757 25259-25259/com.package D/CycledLeScanner: Scan started
```

davidgyoung, 2020-03-31:
@476rick, this is most certainly an issue with setup code in the test app as the Locate app uses this same library.  Please review this question on StackOverflow.com https://stackoverflow.com/questions/60908058/android-beacon-library-not-start-detecting-beacons/60915235#60915235  If that does not help, please open your own question on StackOverflow.com showing your setup code.