AltBeacon/android-beacon-library #773: Can't resrtart scan

NickRimmer, 2018-11-13:
### Expected behavior
Get beacons after restart Activity

### Actual behavior
Can't find any beacon on different devices, until restart Application

### Steps to reproduce this behavior
1. Start app, press "Start"
2. In openned activity (with current beacons info in textview), press "Stop" (second activity will close)
3. Again press "Start"
4. Nothing happening )=

### Mobile device model and OS version
1. Xiaomi Mi Mix 2 (Android 8)
2. Honor 9 model STF-L09 (Android 8)

### Android Beacon Library version
implementation 'org.altbeacon:android-beacon-library:2.15.2'

### Log
https://gist.github.com/NickRimmer/e041fe484192c846981fe72288d2ffaa

**'I/Second activity: ##### BEGIN #####'** -  when second activity opened,
**'I/Second activity: ----- END -----'**  - when activity pause or destroy.
**'E/Second activity: WGX_eddystone'** - when found beacon

NickRimmer, 2018-11-13:
Project sources:
[AltBeaconTest.zip](https://github.com/AltBeacon/android-beacon-library/files/2575281/AltBeaconTest.zip)

NickRimmer, 2018-11-13:
I think this is my problem #767

davidgyoung, 2018-11-13:
I suspect your issue is  lthe same as #756 as your app us are using a ScanJob not a foreground service.   This has been fixed and will be in a binary release planned for next week.  If you want to try sooner, you may compile from source the fix from the branch in the pull request #765.

davidgyoung, 2019-01-09:
closing due to inactivity.  please reopen with an update if you are still having trouble.

closed 2019-01-09 by davidgyoung