AltBeacon/android-beacon-library #775: didRangeBeaconsInRegion not triggered until 10 minutes after startRangingBeaconsInRegion (Android O)

tnortman, 2018-11-14:
### Expected behavior
A `didRangeBeaconsInRegion` callback should occur shortly after `startRangingBeaconsInRegion`, after calling such in `didDetermineStateForRegion` callback while app is in background.

### Actual behavior
After receiving a `didDetermineStateForRegion` callback in the background, I call `startRangingBeaconsInRegion` on the region. I don't ever receive a `didRangeBeaconsInRegion` until ~10 minutes later.

I noticed that there are two other issues similar to the cases I described: #603 and #741 

From what I gathered from both these cases, it seems after the app is killed and the background scan occurs the `didDetermineStateForRegion` callback is triggered. Calling `startRangingBeaconsInRegion` in `didDetermineStateForRegion` does NOT trigger another scan, so no immediate callback to `didRangeBeaconsInRegion` occurs. I think the ending solution with both these cases was to `startRangingBeaconsInRegion` at app launch. I tried this myself and noticed that when I did the aforementioned, I got immediate `didRangeBeaconsInRegion` callbacks after `didDetermineStateForRegion`.

My team is willing to always be ranging but are concerned about battery consumption and having our app always be running in the background. If you could also shed some light on what exactly is the difference between always monitoring (which I believe occurs if you are using BootstrapNotifier) and ranging that would be much appreciated. Mainly, do both use the same scan cycle (i.e. the scan cycle occurs every 10 minutes, if you are monitoring you get monitoring callbacks, if you are ranging, you get ranging callbacks). If that is the case it would seem there is no concern for battery consumption difference for always ranging vs. only ranging when in the region. 

All in all, the chief concern here is that the `didRangeBeaconsInRegion` callback is not triggered until long after `startRangingBeaconsInRegion` while in the background. My team would like to only range when we are inside the region.

### Steps to reproduce this behavior
0. Make sure beacons are off
1. Open app
2. Put app into background
3. Turn on beacon
4. After `didDetermineStateForRegion` callback occurs, it will be approx. 10 minutes before `didRangeBeaconsInRegion` callback is triggered for the first time

Here is a gist of my MainApplication (simple demo app)
[https://gist.github.com/tnortman/94c40b54cf23a664005f2adb3ca0285c](url)

Here is a log I have gathered for my application with the above steps:
Part 1 (begins with app launch) [https://gist.github.com/tnortman/107d0d52aeb6fa0010d34f32625b8aa2](url)
Part 2: [https://gist.github.com/tnortman/97a70c4ee7cf549187c6b1f6a48d3173](url)
I had to split up the logs into two files, so there is a bit of an overlap

Searching for `didDetermineStateForRegion`, you'll notice it determines it is outside the region right away after app launch (10:39:22.205). After putting the app into the background (~10:39:30), I turn on the beacon, and almost immediately get a `didDetermineStateForRegion` (10:39:35.171) This begins startRangingBeaconsInRegion, but no `didRangeBeaconsInRegion` callback occurs until ~10 minutes later (10:51:16.390).

### Mobile device model and OS version
SAMSUNG-SM-G930A - 8.0.0

### Android Beacon Library version
2.15.2

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2018-11-15:
@tnortman thanks for the through writeuonof this issue.  Indeed, when ranging is started during the middle of a background ScanJob, it does not take effect until the next ScanJob.  Fixing this requires cancelling and restarting the ScanJob when ranging/monitoring regions  change.

The alternative of ranging all the time will have negligible impact on battery.  You just get an extra 30 callbacks to your app every 15 minutes.  

Given the simple alternative, putting in a fix is not a super high priority.  If you are concerned and would like to work on a PR to resolve this, I'll gladly give you some guidance as the changes that would be needed.  Code changes should be small -- it is the testing that is time consuming.

tnortman, 2018-11-15:
@davidgyoung Thank you for the info. By confirming that ranging and monitoring callbacks occur on the same ScanJob and that the impact on battery should be negligible, I think my team will be OK that our app is always ranging.

davidgyoung, 2018-11-15:
Yep, makes sense.  I'll leave this ooen since it really does deserve a proper fix, and edit the title so it makes the core issue clearer.