AltBeacon/android-beacon-library #777: setBluetoothName is not working

pjhskcc, 2018-11-15:
### Expected behavior
Local Name filed of scan response is set as specified.

### Actual behavior
No scan response is generated.

### Steps to reproduce this behavior
    public Beacon build() {

        if(uuid == null) return null;
        if(major == null) return null;
        if(minor == null) return null;

        Beacon.Builder builder = new Beacon.Builder();
        builder.setId1(uuid);
        builder.setId2(major);
        builder.setId3(minor);
        builder.setManufacturer(MANUFACTURER_CODE);
        builder.setTxPower(txPower);

        // remember : x0000000b : sos field  | 0xxxxxxxb : battery level
        Long extraField = sos ? Long.valueOf(batteryLevel + HEX_0X80) : Long.valueOf(batteryLevel);
        builder.setDataFields(Arrays.asList(new Long[] {extraField}));

        builder.setBluetoothName(bluetoothName);
        builder.setBluetoothAddress(bluetoothAddress);

        Beacon retBeaconObj = builder.build();
        return retBeaconObj;
    }

### Mobile device model and OS version
Android 8.0 (SM-N950N)
Android 5.1 (Z10)

### Android Beacon Library version

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2018-11-15:
Are you trying to get the BeaconTransmitter to send a scan response?

pjhskcc, 2018-11-15:
> Are you trying to get the BeaconTransmitter to send a scan response?

Yes, right.

davidgyoung, 2018-11-15:
The BeaconTransmitter just doesn't work that way.  It only triggers advertising of the beacon frame.  The beacon.bluetoothName field can be used to read the name from scanned beacons, but it has no affect on transmitted beacons.  Perhaps the JavaDocs on this field can make this explicit.

I believe there are built-in Android APIs to configure the scan response, but this Library simply foes not do this.

closed 2018-11-15 by pjhskcc