AltBeacon/android-beacon-library #781: Background monitoring causing crashes when outside region (& ScanHelper thread pools are not shutdown when ScanJob is finished)

fiv, 2018-11-21:
Not sure if this should be two different issues or named differently. Thread pools are not properly shutdown for every ScanHelper (so not related to background monitoring). And not sure if the background monitoring is working as it should.

### Expected behavior

App is monitoring for predefined beacon region. When app is outside the beacon region and is put to the background, monitoring should continue normally without crashes.

### Actual behavior

When outside beacon region and app is put to the background, it will eventually run out of resources / OS will kill the app. One reason for this seems to be that the ScanHelpers thread pools will stay running forever. Every ScanJob will create a new ScanHelper with new thread pool (pool size seems to be CPU cores + 1). So after running the app for a few minutes in the background, it will end up having hundreds of threads, and eventually thousands.

Used settings: 
```
beaconManager.setForegroundScanPeriod(15 * DateUtils.SECOND_IN_MILLIS);
beaconManager.setForegroundBetweenScanPeriod(3 * 60 * DateUtils.SECOND_IN_MILLIS);
beaconManager.setBackgroundScanPeriod(15 * DateUtils.SECOND_IN_MILLIS);
beaconManager.setBackgroundBetweenScanPeriod(21 * 60 * DateUtils.SECOND_IN_MILLIS);
```

On background and outside the beacon region, ScanJob runs for 15 seconds and then expires. After that ScanJob will call startAndroidOBackgroundScan (scheduleNextScan->startPassiveScanifNeeded->startAndroidOBackgroundScan-StartupBroadcastReceiver->ScanJobScheduler). So basically new ScanJob & new ScanHelper with new threadpool (with my Samsung SM-G935F, pool size is 9 threads) is created every 15 seconds.

Here is log from the app https://gist.github.com/fiv/91aaa7e4405b4572bbef1f12fe2943cb (HUGE)

Profiler screenshot showing increasing amount of threadpools: 
[image]

### Steps to reproduce this behavior

I managed to reproduce this using the reference app with the following BeaconManager settings: https://gist.github.com/fiv/20a63d7e7eda820ae8075c20a9840e7a

### Mobile device model and OS version

Samsung SM-G935F
Android 8.0.0

### Android Beacon Library version

2.15.2 and also master branch (commit 7df874d)

davidgyoung, 2018-11-21:
Thanks for this detailed analysis, @fiv.  I suspect this is the root cause of this issue: #647 

Based on the docs [here](https://docs.oracle.com/javase/7/docs/api/java/util/concurrent/Executors.html#newFixedThreadPool(int,%20java.util.concurrent.ThreadFactory)) it seems there simply needs to be a call to `mExecutor.shutdown();` before the ScanHelper is disposed.  

This could be put in a `finalize()` method on the class, or probably better, in a new custom `disable()` method that gets explicitly called by ScanJob or BeaconService before they exit.

davidgyoung, 2018-11-24:
@fiv, I've coded a fix in #781 and made an experimental release including that fix here: 
 https://github.com/AltBeacon/android-beacon-library/releases/tag/2.15.3-beta1

Would you be willing to re-run your test setup to see if you can confirm this resolves the issue?

fiv, 2018-11-28:
@davidgyoung Tested with the experimental release 2.15.3-beta1. My app has been running for ~20 hours and seems to be working correctly, so no thread leaks anymore.

davidgyoung, 2018-11-28:
Great, thanks for the assistance @fiv

closed 2018-11-28 by davidgyoung