AltBeacon/android-beacon-library #784: Backgroun and foreground together. Not starting after boot

YuraAAA, 2018-11-22:
### Expected behavior

Bind to consumer in foreground and work with RegionBootstrap in background

### Actual behavior

Background scan doesn't start

### Steps to reproduce this behavior

I want to combine background and foreground beacon detection. 

Class for foreground scan

```
class BeaconForegroundService(private val application: Application) : BeaconConsumer {

    private var beaconManager: BeaconManager? = null

    companion object {
        private const val REGION_UID = "region_uid"
    }

    override fun getApplicationContext(): Context {
        return application.applicationContext
    }

    override fun unbindService(p0: ServiceConnection) {
        application.unbindService(p0)
    }

    override fun bindService(p0: Intent, p1: ServiceConnection, p2: Int): Boolean {
        return application.bindService(p0, p1, p2)
    }

    override fun onBeaconServiceConnect() {

        beaconManager!!.beaconParsers.clear()
        beaconManager!!.beaconParsers.add(AltBeaconParser())
        beaconManager!!.beaconParsers.add(BeaconParser().setBeaconLayout(BeaconParser.URI_BEACON_LAYOUT))
        beaconManager!!.beaconParsers.add(BeaconParser().setBeaconLayout(BeaconParser.EDDYSTONE_TLM_LAYOUT))
        beaconManager!!.beaconParsers.add(BeaconParser().setBeaconLayout(BeaconParser.EDDYSTONE_UID_LAYOUT))
        beaconManager!!.beaconParsers.add(BeaconParser().setBeaconLayout(BeaconParser.EDDYSTONE_URL_LAYOUT))
        beaconManager!!.beaconParsers.add(BeaconParser().setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25"))
        beaconManager!!.backgroundMode = false

        beaconManager!!.addMonitorNotifier(object : MonitorNotifier {
            override fun didDetermineStateForRegion(p0: Int, p1: Region?) {
                Timber.d("FGBeacon: didDetermineStateForRegion INSIDE = ${p0 == MonitorNotifier.INSIDE}")
            }

            override fun didEnterRegion(p0: Region?) {
                Timber.d("FGBeacon: didEnterRegion")
            }

            override fun didExitRegion(p0: Region?) {
                Timber.d("FGBeacon: didExitRegion")
            }
        })
        beaconManager!!.addRangeNotifier { mutableCollection: MutableCollection<Beacon>, region: Region ->
            Timber.d("FGBeacon $mutableCollection")
        }
        beaconManager!!.startMonitoringBeaconsInRegion(Region(REGION_UID, null, null, null))
        beaconManager!!.startRangingBeaconsInRegion(Region(REGION_UID, null, null, null))
    }

    fun startScan() {
        beaconManager = BeaconManager.getInstanceForApplication(application)
        beaconManager!!.bind(this)
    }
[... 140 lines cut ...]
Processing pdu type FF: 0201061aff4c000215b5b182c7eab14988aa99b5c1517008d90001df6dc50d0961626561636f6e5f36444446000000000000000000000000000000000000 with startIndex: 5, endIndex: 29
D/BeaconParser: This is not a matching Beacon advertisement. Was expecting aa fe at offset 5 and 20 at offset 7.  The bytes I see are: 0201061aff4c000215b5b182c7eab14988aa99b5c1517008d90001df6dc50d0961626561636f6e5f36444446000000000000000000000000000000000000
 D/BeaconParser: Ignoring pdu type 01
...
 D/BeaconParser: Processing pdu type FF: 0201061aff4c000215b5b182c7eab14988aa99b5c1517008d90001df6dc50d0961626561636f6e5f36444446000000000000000000000000000000000000 with startIndex: 5, endIndex: 29
D/BeaconParser: This is a recognized beacon advertisement -- 02 15 seen
D/BeaconParser: Bytes are: 0201061aff4c000215b5b182c7eab14988aa99b5c1517008d90001df6dc50d0961626561636f6e5f36444446000000000000000000000000000000000000
D/ScanHelper: Beacon packet detected for: id1: b5b182c7-eab1-4988-aa99-b5c1517008d9 id2: 1 id3: 57197 with rssi -46
D/RangeState: adding id1: b5b182c7-eab1-4988-aa99-b5c1517008d9 id2: 1 id3: 57197 to existing range for: org.altbeacon.beacon.service.RangedBeacon@332c859
```

_After reboot:_

```
I/BeaconManager: BeaconManager started up on pid 7407 named 'com.myapp' for application package 'com.myapp'.  isMainProcess=true
D/BeaconParser: Parsing beacon layout: m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25
```
And thats all. No detections...

So, my question:
1. Is it bug of library/device or i doing something wrong?
2. Can we use background scan without notification?
3. Whan we start application again (to foreground) notification refreshed. How can i avoid this?

Thanks

davidgyoung, 2018-11-22:
@YuraAAA, this question is really out of scope for this forum since it is about debugging a custom app.  This forum is reserved for feature requests and reproducible bugs with the library itself.

If you repost your question on StackOverflow.com, I am happy to look, but please try to make it as comcise as possible.

In general, for background detection to work, you must trigger Region bootstrap in the onCreate method of your custon Application class.  Please verify you do that and show clearly how that is done.

closed 2018-11-22 by davidgyoung

YuraAAA, 2018-11-22:
> @YuraAAA, this question is really out of scope for this forum since it is about debugging a custom app. This forum is reserved for feature requests and reproducible bugs with the library itself.
>
> If [...]

Sure, sorry for this issue) Can you please look at my question?[https://stackoverflow.com/questions/53435116/android-altbeacon-background-and-foreground-together](link)