AltBeacon/android-beacon-library #791: This is not a matching Beacon advertisement

breakline87, 2018-12-04:
Hello,

I have a bunch of Beacons here from an older project. Now when I'm using the reference application, I see a lot of things like:

```
12-04 14:53:08.276 27968-28108/org.altbeacon.beaconreference D/BeaconParser: This is not a matching Beacon advertisement. Was expecting aa fe at offset 2 and 00 at offset 4.  The bytes I see are: 1eff060001092000c681e225456aa27d40d2d09712dc329a4b27293c61de9b00000000000000000000000000000000000000000000000000000000000000
    Processing pdu type FF: 1eff060001092000c681e225456aa27d40d2d09712dc329a4b27293c61de9b00000000000000000000000000000000000000000000000000000000000000 with startIndex: 2, endIndex: 30
```

I'm kind of new to this and don't understand what this all means. I have various beacons like Blue cats, Estimote etc. and not sure how to scan them. 

Thanks

davidgyoung, 2018-12-04:
@breakline87, this line simply means that the library read a Bluetooth LE advertisement that does not match any beacon layout it knows about.  The log line displays the bytes it saw.  Based on those bytes:

```
 This is not a matching Beacon advertisement. Was expecting aa fe at offset 2 and 00 at offset 4.  The bytes I see are: 1eff060001092000c681e225456aa27d40d2d09712dc329a4b27293c61de9b00000000000000000000000000000000000000000000000000000000000000
```

It is a manufacturer advertisement of a device with manufacturer code 0x0006 = Microsoft

This is most likely not a beacon advertisement at all, but instead a bluetooth packet emitted by your Windows laptop (or XBox or other Windows device).  Devices often emit such advertising packets for proprietary purposes that have nothing to do with beaconing.  It is normal to see such packets.

When you have a bunch of beacons and you don't know what they advertise, it is very difficult to find them with logs like this, because you never know which packets are sent by the beacons and which are sent by other non-beacon devices.  This library is not designed to be a tool to reverse-engineer such advertisements.  But if you want to use such debug lines to do this, you should probably:

1. Capture a baseline measurement of all the distinct BLE packets you scan when none of these beacons are in the vicinity
2. Introduce *one* beacon at a time to the vicinity (within ~50m)
3. Find what *new* BLE packets show up when the above beacon is around.

If you follow this procedure you might post the results in a question on StackOverflow.com.  I'd be glad to help there, but this effort is certainly out of scope for this forum.   Good luck!

closed 2018-12-04 by davidgyoung

breakline87, 2018-12-05:
Thanks, that makes sense.