AltBeacon/android-beacon-library #794: Compatibility with Evernote's Android-Job

dmelo, 2018-12-06:
We are using Android-Job for a few other services that our app run and it seems to be conflicting with AltBeacon own way of scheduling background tasks.

Is there a known way of making AltBean background work with Android-Job? Or, more generically, how can I manage background scanning on my own?

davidgyoung, 2018-12-06:
I have not used Evernote Android-Job, but based on my understanding of it, it should work fine with the library.  My suggested usage would be to:

1. Make your job class implement the BeaconConsumer interface.  (That interface is generally designed for Activity or Application, but you can use it in any class that has access to an Android context.  Simply chain the interface's method stubs to the equivalent context method.)

2. At the start of your job, instantiate a BeaconManager, register any BeaconParser instances as needed, then call bind(this) on the BeaconManager.

3. When you get a `onBeaconServiceConnected()` callback, you can start ranging or monitoring for beacons.

4. If you want to keep scanning for a fixed time interval, in your main job method you can call Thread.sleep(interval).

5. When stopping the job call unbind(this) on the BeaconManager.

If you run into trouble, feel free to open questions on StackOverflow.com, which I'm glad to answer, but I'm closing this issue as it is reserved for feature requests and bug reports with the library itself.

closed 2018-12-06 by davidgyoung

dmelo, 2018-12-11:
It worked. Thank you.