AltBeacon/android-beacon-library #797: ANR  EXCEPTION

TracyEminem, 2018-12-09:
### Expected behavior
no anr
### Actual behavior
each time i launch my app,it shows anr exception 
### Steps to reproduce this behavior

when i add  following code my app will show anr exception,if i delete the following code the anr will not show,even when i put the following code in  an asynchronous thread，it still will show anr exception.but when i use the android-beacon-library-reference to add the code,it will not show anr exception,

  Region region = new Region("backgroundRegion",
                    Identifier.parse("EEC56DB5-DFFB-48D2-B060-D0F5A71096EE"), null, null);
            regionBootstrap = new RegionBootstrap(this, region);

here is the code of my application ,is there anything wrong with my code?

`public class App extends Application implements BootstrapNotifier , BeaconConsumer, RangeNotifier {

    private static final String TAG = "BeaconReferenceApp";
    private RegionBootstrap regionBootstrap;
//    private BackgroundPowerSaver backgroundPowerSaver;
    private boolean haveDetectedBeaconsSinceBoot = false;
    private HomeFragmentActivity monitoringActivity = null;
    ECallModule tempModule;

    List<ECallModule> eCallModuleList;
    private List<ElevatorData> mBuildList;
    Gson gson;
    private ECallService myCall;
    Disposable mDisposable;
    private boolean isEnabled;

    private static App INSTANCE;
    // 用于存放倒计时时间
    public static Map<String, Long> map;

    public static App getINSTANCE() {return INSTANCE;} ;

    public static Context getCONTEXT(){
        return INSTANCE;
    }

    SharedPreferenceUtil mPerference;

    public String getAccessTokenAuth(){
        if(mPerference == null)
            mPerference = new SharedPreferenceUtil(this);
        return mPerference.getToken();
    }

    @Override
    public void onCreate() {
        super.onCreate();

        gson = new Gson();

        BeaconManager beaconManager = org.altbeacon.beacon.BeaconManager.getInstanceForApplication(this);

        beaconManager.getBeaconParsers().clear();
        beaconManager.getBeaconParsers().add(new BeaconParser().
                setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24"));

        Notification.Builder builder = new Notification.Builder(this);
        builder.setSmallIcon(R.mipmap.ic_launcher);
        builder.setContentTitle(getResources().getString(R.string.we_are_working));
        builder.setAutoCancel(true);
        builder.setContentText(this.getString(R.string.app_name));
        builder.setDefaults(Notification.DEFAULT_LIGHTS);

        Intent intent = new Intent(this, HomeFragmentActivity.class);
        PendingIntent pendingIntent = PendingIntent.getActivity(
                this, 0, intent, PendingIntent.FLAG_UPDATE_CURRENT
        );
        builder.setContentIntent(pendingIntent);
        beaconManager.enableForegroundServiceScanning(builder.build(), 456);

            beaconManager.setEnableScheduledScanJobs(false);
            beaconManager.setBackgroundBetweenScanPeriod(1100);
            beaconManager.setBackgroundScanPeriod(5000);

            Region region = new Region("backgroundRegion",
                    Identifier.parse("EEC56DB5-DFFB-48D2-B060-D0F5A71096EE"), null, null);
            regionBootstrap = new RegionBootstrap(this, region);

        TCAgent.LOG_ON = true;
       
        mPerference = new SharedPreferenceUtil(this);

        if (LeakCanary.isInAnalyzerProcess(this)) {
            // This process is dedicated to LeakCanary for heap analysis.
            // You should not init your app in this process.
            return;
        }

        //initTalkingData
        TCAgent.init(this);

        INSTANCE = this;

        Intent bindIntent = new Intent(this, ECallService.class);
        this.bindService(bindIntent, mFhrSCon, Context.BIND_AUTO_CREATE);

        /*开启网络广播监听*/
        NetStateReceiver.registerNetworkStateReceiver(this);
        FormatStrategy   formatStrategy = PrettyFormatStrategy.newBuilder()
                .tag("E-CALL")   // (Optional) Global tag for every log. Default PRETTY_LOGGER
                .build();

        Logger.addLogAdapter(new AndroidLogAdapter(formatStrategy));

        LeakCanary.install(this);

    }

    private ServiceConnection mFhrSCon = new ServiceConnection() {
        @Override
        public void onServiceConnected(ComponentName name, IBinder service) {
            myCall = ((ECallService.ECallServiceBinder) service).getService();
//            myCall.setEcallCallback(callback);
        }

        @Override
        public void onServiceDisconnected(ComponentName name) {
            myCall = null;
        }
    };

    @Override
    public void onLowMemory() {
        super.onLowMemory();
        NetStateReceiver.unRegisterNetworkStateReceiver(this);
        android.os.Process.killProcess(android.os.Process.myPid());
    }

    @Override
    public void didEnterRegion(Region region) {
        //进入区域
        if(!haveDetectedBeaconsSinceBoot){
            //第一次打开app的操作
            haveDetectedBeaconsSinceBoot = true;
        }
        if(!isEnabled) {
            new Handler().postDelayed(() -> {
                myCall.eCallDisable();
                myCall.eCallEnable();
                isEnabled = true;
            },4000);
        }
        mPerference.setTime(String.valueOf(System.currentTimeMillis()));
//        else {
//
//        }
    }

    @Override
    public void didExitRegion(Region region) {
        myCall.eCallDisable();
        isEnabled = false;
    }

    @Override
    public void didDetermineStateForRegion(int i, Region region) {

    }

    public void setMonitoringActivity(HomeFragmentActivity activity) {
        this.monitoringActivity = activity;
    }

    @Override
    public void onBeaconServiceConnect() {
        
    }

    @Override
    public void didRangeBeaconsInRegion(Collection<Beacon> collection, Region region) {

    }
}`

### Mobile device model and OS version

huawei honor android 8.0,huawei mate20 android 9.0 redmi 4x

### Android Beacon Library version
android-beacon-library-2.15.3-beta1
IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2018-12-09:
It nay be that you have to construct a RegionBootstrap on the main thread.  The code shown does not include constructing this on an asynchrounous thread so I cannot tell if this is already done.  If not, you could try wrapping your asynchronius thread constructor in a block that executes it on the main thread.

TracyEminem, 2018-12-11:
when i construct a RegionBootstrap on an aysnchrounous thread 

 ((Runnable) () -> {
            Region region = new Region("backgroundRegion",
                    Identifier.parse("EEC56DB5-DFFB-48D2-B060-D0F5A71096EE"), null, null);
            regionBootstrap = new RegionBootstrap(this, region);

        }).run();

it still show anr..It doesn't appear in the reference library..

TracyEminem, 2018-12-14:
i have fixed it,i put the code in application class but i call beaconmanager.bind in activity.....when i call beaconmanager.bind in application class.it won't show anr.

closed 2018-12-14 by TracyEminem