AltBeacon/android-beacon-library #800: Issue #790 not included on release 2.15.3

dmelo, 2018-12-11:
Release notes of 2.15.3 says it fixes #790 but commit 9dcd2f6 does not seem to be  included on tag 2.15.3 . Moreover, I have the 2.15.3 version installed and I'm getting the error:

```
java.lang.RuntimeException: Unable to stop service org.altbeacon.beacon.service.BeaconService@bea0e2f: java.lang.NullPointerException: Attempt to invoke virtual method 'void org.altbeacon.bluetooth.BluetoothCrashResolver.stop()' on a null object reference
\tat android.app.ActivityThread.handleStopService(ActivityThread.java:3494)
\tat android.app.ActivityThread.-wrap27(ActivityThread.java)
\tat android.app.ActivityThread$H.handleMessage(ActivityThread.java:1667)
\tat android.os.Handler.dispatchMessage(Handler.java:110)
\tat android.os.Looper.loop(Looper.java:203)
\tat android.app.ActivityThread.main(ActivityThread.java:6339)
\tat java.lang.reflect.Method.invoke(Native Method)
\tat com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1084)
\tat com.android.internal.os.ZygoteInit.main(ZygoteInit.java:945)
Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'void org.altbeacon.bluetooth.BluetoothCrashResolver.stop()' on a null object reference
\tat org.altbeacon.beacon.service.BeaconService.onDestroy(BeaconService.java:333)
\tat android.app.ActivityThread.handleStopService(ActivityThread.java:3477)
\t... 8 more
java.lang.NullPointerException: Attempt to invoke virtual method 'void org.altbeacon.bluetooth.BluetoothCrashResolver.stop()' on a null object reference
\tat org.altbeacon.beacon.service.BeaconService.onDestroy(BeaconService.java:333)
\tat android.app.ActivityThread.handleStopService(ActivityThread.java:3477)
\tat android.app.ActivityThread.-wrap27(ActivityThread.java)
\tat android.app.ActivityThread$H.handleMessage(ActivityThread.java:1667)
\tat android.os.Handler.dispatchMessage(Handler.java:110)
\tat android.os.Looper.loop(Looper.java:203)
\tat android.app.ActivityThread.main(ActivityThread.java:6339)
\tat java.lang.reflect.Method.invoke(Native Method)
\tat com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1084)
\tat com.android.internal.os.ZygoteInit.main(ZygoteInit.java:945)
```

davidgyoung, 2018-12-11:
Indeed.  The PR for #790 was not made against the master branch but another working branch, so I did not realize it was not merged to master when I merged the PR.  I have merged it now, and it will be going into 2.15.4 release which will appear imminently.  I have also corrected the CHANGELOG

davidgyoung, 2018-12-11:
The release is published on JCenter and here: https://github.com/AltBeacon/android-beacon-library/releases/tag/2.15.4

closed 2018-12-11 by davidgyoung