AltBeacon/android-beacon-library #806: OnePlus foreground service killed after a while.

orhaneee, 2018-12-30:
### Expected behavior
Foreground service is not being killed by operating system.

### Actual behavior
After a while, foreground service gets killed and when grepping logcat, could not find any logical reason. I think it may be related to doze mode in OnePlus.

### Steps to reproduce this behavior
Try sample code with foreground service activated, then wait 2 hours.

### Mobile device model and OS version
OnePlus 5T, Android 8.1

### Android Beacon Library version
Latest version as indicated in installation `2+`

orhaneee, 2018-12-30:
wow, thanks for your quick response. I understand it but the problem is, OS does not restart this service.

davidgyoung, 2018-12-30:
Can you compare your code to that posted by @AntonHellbe to see if you can find any differences that might affect the restart?  He alsonpoated LogCat excerpts which may be helpful in comparing to your case.

orhaneee, 2018-12-30:
I checked it, seems similar nothing different except I instantiated `BackgroundPowerSaver`. I thought it was a power issue..

davidgyoung, 2019-04-04:
Actually if you read @AntonHelbe's post again, he clearly says:

> When that happens a "High battery usage" notification appears telling that MyApp has high battery usage. The "force stop" requires you to manually start the app from the launcher again

So the fact that it will not restart without manual intervention is expected on OnePlus.  

My understanding is there is no way around this, other than putting the app into OnePlus' background whitelist.  You can give your users instructions on how to do this, but I know of no way to do it programmatically.

closed 2019-04-04 by davidgyoung

zain-noman, 2023-01-09:
On my OnePlus Nord CE. I Long Pressed the app's icon ->App Info->BatteryUsage->Allow Background activity.
This solved the issue for me even though im only using a foreground service.
PS I'm not using this library, I just had trouble with foreground services in one plus.