AltBeacon/android-beacon-library #809: Distance calculation for Android Gateway Box

amaadraza, 2019-01-07:
I am using Eddystone Beacons and connecting them with the Android Gateway Box.

The problem is the distance returned by the gateway box is not that proper. Even though the beacon is within a meter from the gateway box, it gives the distance between 3 to 5 meters. So I thought of customizing the config file (JSON file of resource directory ) used for distance calculation. But looking at the SDK code I found that the distance calculation model is actually being downloaded by the web API. I want to add data specific to the gateway box to the distance calculation model and want to get that data in the response of the API that gets called for distance calculation model So I need help on how can I make it possible.

Thanks.

davidgyoung, 2019-01-07:
Two options:

1. You can call BeaconManager.setDistanceModelUpdateUrl(url); to set the URL of your own JSON file containing custom definitions.

2. You can build your own version of the library from source and embed your own JSON file, then call  BeaconManager.setDistanceModelUpdateUrl(null); to disable the remove update so it won't overwrite your embedded changes.

closed 2019-01-07 by davidgyoung

amaadraza, 2019-02-01:
Hello David,

Thank you very much for your helpful response.

I choose the first option that you described and did the calibration and formula constant calculation process. I am using this https://docs.google.com/spreadsheets/d/1ymREowDj40tYuA5CXd4IfC4WYPXxlx5hq1x8tQcWWCI/edit#gid=0 sheet for reference. However, I am stuck in Step 3: Run the Regression.

To calculate the constant A & B I used this http://www.xuru.org/rt/powr.asp link as described in doc and for testing purpose, I used the data of the above sheet. The result I found of A is different then what is in the sheet however B is the same.
In sheet A is 1.203420305 and in my results A is 4.295400587.
[image]

I want to know if there is any different process to calculate the constant A or I am doing anything wrong.

Thank you.

davidgyoung, 2019-02-01:
it is not surprising that different regression tools five different constants.  what is important is that you get sufficiently low error rates in your table.  this tells you of the formula will providw a good estimate.