AltBeacon/android-beacon-library #810: Incremental hits on didEnterRegion when app closed opened

biswas123, 2019-01-10:
### Expected behavior
Open app monitoring started, should detect a beacon and didEnterRegion should hit once. 
### Actual behavior
Open app monitoring starts, didEnterRegion hits 1 time on first run but after unbind and destroyed activity then open app again didEnterRegion hits 2 times. (repeat same process then hits goes on increasing from 1, 2 , 3 ,4 , ...... )

### Steps to reproduce this behavior
- Open app monitoring starts, didEnterRegion hits 1 time on first run 
- Close the app (unbind and destroyed activity)
- Open app again didEnterRegion now hits 2 times. 
(repeat same process then hits goes on increasing from 1, 2 , 3 ,4 , ...... )

### Mobile device model and OS version
- android 9.0 (Oneplus 6)

### Android Beacon Library version
- 12.15.4

My code is attached and below are onCreate() and onDestroy() code snippets:
```
    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        Log.i(TAG, "OnCreate() called.");
        setContentView(R.layout.activity_main);
        beaconManager = BeaconManager.getInstanceForApplication(this);
        beaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24"));

 
        if (!beaconManager.isBound(this)) {
            Notification.Builder builder = new Notification.Builder(MainActivity.this);
            builder.setSmallIcon(R.mipmap.ic_launcher);
            builder.setContentTitle("Scanning for Beacons");
            if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
                NotificationChannel channel = new NotificationChannel(TAG,
                        "BeaconPOC", NotificationManager.IMPORTANCE_HIGH);
                channel.setDescription("BeaconPOC");
                NotificationManager notificationManager = (NotificationManager) getSystemService(
                        Context.NOTIFICATION_SERVICE);
                notificationManager.createNotificationChannel(channel);

                PendingIntent contentIntent = PendingIntent.getActivity(this, 0,
                        new Intent(this, MainActivity.class), PendingIntent.FLAG_UPDATE_CURRENT);

                builder.setContentIntent(contentIntent);
                builder.setChannelId(channel.getId());
            }
            beaconManager.enableForegroundServiceScanning(builder.build(), FOREGROUND_NOTIFICATION_ID);
            beaconManager.setEnableScheduledScanJobs(false);
            beaconManager.setRegionStatePersistenceEnabled(false);
            beaconManager.bind(this);
            Log.i(TAG, "Beacon service just got bound");
            backgroundPowerSaver = new BackgroundPowerSaver(this);
        }
  }

  @Override
    protected void onDestroy() {
        super.onDestroy();
        stopRangingAndMonitoring();
        beaconManager.unbind(this);
        NotificationManager notificationManager = (NotificationManager) getSystemService(Context.NOTIFICATION_SERVICE);
        notificationManager.cancel(10001);
        beaconManager.disableForegroundServiceScanning();
    }
```

davidgyoung, 2019-01-10:
Can you please show `stopRangingAndMonitoring` and where you start ranging and monitoring?

biswas123, 2019-01-10:
@davidgyoung  Updated code and attached activity file.

biswas123, 2019-01-10:
> Can you please show `stopRangingAndMonitoring` and where you start ranging and monitoring?
@davidgyoung Updated code and attached activity file.

davidgyoung, 2019-01-10:
I suspect the issue is in this line;

      beaconManager.addRangeNotifier(new RangeNotifier() 

Each time that is executed a new RangeNotifier instance is constructed and added to the range notifier callback list.  Because none of these are ever removed, you get a call back to each one 

The addRangeNotifier call is smart enough keep a distinct list of notifiers, so if you add the same object twice,it only retains one.  But in this case, each RangeNotifier object is different, so you end up with multiple callbacks.

To fix this, construct a single RangeNotifier object as a member variable in your class, or make your class implement the RangeNotifier interface.  Either solution will ensure you only have one object at a time and therefore get only one callback.

biswas123, 2019-01-10:
> I suspect the issue is in this line;
>
> ```
>   beaconManager.addRangeNotifier(new RangeNotifier()
> ```
> Each time that is executed a new RangeNotifier instance is constructed and added to the range [...]

Thanks, I will try implementing as a member variable and see if solves the problem.
(Although the examples on alt beacon library documentation : https://altbeacon.github.io/android-beacon-library/documentation.html) have code that follow the same way as I do.

davidgyoung, 2019-01-10:
Good point about the reference code.  Those were originally written with the now deprecated `addRangeNotifier()` and `addMonitorNotifier()` variants so this wasn't a problem.  The simplest change to the samples is to add a call to `removeAllRangeNotifiers()` before the add.  I'll make that change.

davidgyoung, 2019-01-10:
Docs updated

biswas123, 2019-01-12:
@davidgyoung Hi David, I was not able to fix the problem calling         `**beaconManager.removeAllMonitorNotifiers();**`
before starting monitoring / ranging but was able to solve it by adding this line on onDestroy() method.

```
@Override
    protected void onDestroy() {
        super.onDestroy();
        stopRangingAndMonitoring();
        beaconManager.removeAllMonitorNotifiers();
        beaconManager.removeAllRangeNotifiers();
        beaconManager.unbind(this);
        NotificationManager notificationManager = (NotificationManager) getSystemService(Context.NOTIFICATION_SERVICE);
        notificationManager.cancel(FOREGROUND_NOTIFICATION_ID);
        beaconManager.disableForegroundServiceScanning();
    }
```

davidgyoung, 2019-01-14:
Glad you found a fix that worked for you.  

The rangeNotifiers and monitorNotifiers on BeaconManager are contained in a Set, so they will not allow duplicates.  However, if you end up with two instances of your MainActivity (quite possible in Android depending on how you set up your AndroidManifest.xml or use Intents to start your activity), then you might end up with multiple notifiers.  Clearing the notifiers is always the safest bet to avoid duplicates.

closed 2019-01-14 by davidgyoung