AltBeacon/android-beacon-library #812: Beacon Manager ignore custom Beacon Parser and use inside only AltBeaconParser

digital-prime, 2019-01-16:
I'm creating beaconManager and do work like this:
```this.altBeaconManager = org.altbeacon.beacon.BeaconManager.getInstanceForApplication(this.appContext);

final BeaconParser beaconParser = new BeaconParser("bia_beacon");
        beaconParser.setBeaconLayout(BEACON_LAYOUT)
                .setHardwareAssistManufacturerCodes(MANUFACTURERS);
        altBeaconManager.getBeaconParsers().clear();
        altBeaconManager.getBeaconParsers().add(beaconParser);
```

But correct setting of this property work only when I run code from android studio. When I swipe-and-close app in device and run App again - always inside altBeaconManager parsing are processing with default AltBeaconParser! 
In logs it's printing that my custom beacon parser is set, but if I attach debug to process I see, that function `fromScanData` is calling from AltBeaconParser instance. What's happening and how can I use my custom parsers?

digital-prime, 2019-01-16:
@TonyTangAndroid or @davidgyoung - can you help with this problem?

davidgyoung, 2019-01-16:
where do you put that setup code?  It needs to be triggered at app startup, typically in the onCreate method of a custom Application class.

digital-prime, 2019-01-17:
> where do you put that setup code? It needs to be triggered at app startup, typically in the onCreate method of a custom Application class.

I put it in OnCreate of my single MainActivity. 
Seems like setting of AltBeacon parser happens when I close \ swipe application and after that calling my Custom Parser in OnCreate doesn't affect on Beacon Manager

davidgyoung, 2019-01-17:
For this to work, you must implement an Android Application class and put your monitoring initialization code in the onCreate method of this class (this includes any BeaconParsers).  See [here](https://altbeacon.github.io/android-beacon-library/background_launching.html) for instructions.

closed 2019-01-17 by davidgyoung