AltBeacon/android-beacon-library #816: Context.startForegroundService() did not then call Service.startForeground

TFreudi1, 2019-02-01:
I want to restart the service + bluetooth stack

After a few try's I got a panic and the error, see subject

I simplify and change your refrence App and add a timer that restart everything every 60 seconds. Sometimes the error comes up after 1 try sometimes it needs 10 times. In use version 2.15.4 of your lib as you can see in the gradle file that is included in sample project that I send.

[android-beacon-library-reference-master.zip](https://github.com/AltBeacon/android-beacon-library/files/2821410/android-beacon-library-reference-master.zip)

TFreudi1, 2019-02-01:
Hmm, thats a bit more, I remove the monitor and ranging activity and change the Application to an AppCompatActivity. I put the code that starts the lib + foreground service in an fuction called startBluetooth.
The code is theorignal code + a row that binds the service.
I added new code that stops everything and call the function that start it again.

TFreudi1, 2019-02-01:
Look at BeaconReferenceApplication.java it is not that much code.
As I said I took the "Lib-Starting Code" into a function startbeacon and add
beaconManager.bind(this);
The main Application changed into
public class BeaconReferenceApplication extends AppCompatActivity implements BeaconConsumer,BootstrapNotifier 
And completly new is:
 public class RestartTimerTask extends TimerTask {
        @Override
        public void run() {
            runOnUiThread(new Runnable() {
                @Override
                public void run() {
                    tvRestartCnt.setText(String.valueOf(nRestartCnt++));
                }});

            AsyncTask.execute(new Runnable() {
                    @Override
                    public void run() {
                        restartCompleteBluetooth();
                    }});
        }
    }
    public void restartCompleteBluetooth()
    {
        try {
            //stop Bluetooth
            try {
                    if (regionBootstrap != null) {
                        regionBootstrap.disable();
                        regionBootstrap = null;

                    }
            } catch (Exception e) {
                Log.d(TAG,"stopBluetoothLib sregionbootstrap " + e.getLocalizedMessage());
            }

            try {
                beaconManager.removeAllRangeNotifiers();
                beaconManager.removeAllMonitorNotifiers();

            } catch (Exception e) {
                Log.d(TAG,"stopBluetoothLib stop Monitoring/Ranging" + e.getLocalizedMessage());
            }

            try {
                beaconManager.unbind(this);
            } catch (Exception e) {
                Log.d(TAG,"stopBluetoothLib unbind service" + e.getLocalizedMessage());
            }

            try {
                    beaconManager.disableForegroundServiceScanning();
            } catch (Exception e) {
            Log.d(TAG,"stopBluetoothLib disable Foregroundservice" + e.getLocalizedMessage());
            }

            beaconManager = null;
            /*
            BluetoothAdapter mBluetoothAdapter = BluetoothAdapter.getDefaultAdapter();
            if (mBluetoothAdapter.isEnabled())
                mBluetoothAdapter.disable();
            try {
                Thread.sleep(1000);
                } catch (InterruptedException e) {
                    e.printStackTrace();
            }

            // Start Bluetooth
            if (!mBluetoothAdapter.isEnabled())
                 mBluetoothAdapter.enable();
*/
            try {
                Thread.sleep(10000);
            } catch (InterruptedException e) {
                e.printStackTrace();
            }

            startBluetooth();

        } catch (Exception e) {
            Log.d(TAG,e.getLocalizedMessage());
        } finally {

        }
    }``

davidgyoung, 2019-02-01:
Sorry, @Tfreudi1,this forum is reserved for reproducible bug reports or feature requests and this is certainly bleeding into debugging of a custom implementation. 
This puts it out of scope for this forum.

Perhaps you can make a smaller change to the reference app to reproduce the problem.   If you can do that, then please reopen this issue with a snippet of the change needed to reproduce.  Alternatively,. you might consider StackOverflow.com as a forum for helping with a custom implementation, buy there, too, you probably need to pare down the changes to. bite sized issue to realistically get help.

TFreudi1, 2019-02-01:
A few lines are only possible if you extend your reference application with code that stops the foregroundservice again. I saw many people asking for it. I also did this because I have a applications based on your lib and at this point only about 10 people that testing it and it happens a lot that ranging/ monitoring stops although are still sending. My function restart helps with that and the lib is working good after that. So in my eyes it is an issue with your lib and I gave you a simple project to reproduce it.
BTW. BluetoothMedic did'nt help in such cases.

TFreudi1, 2019-02-04:
"I cant get to work" means I cant wrote it as code. It has nothing to do with the function of the code inside the app. 
I remove all tabs from the code, check al new lines but it is always missinterpreted by the github software :-(

davidgyoung, 2019-02-04:
Thanks @TFreudi1, this example is much easier to work with.  Can you please set `beaconManager.setDebug(true)` and attach a LogCat capture of the sequence that results in the exception?  The log should capture the time from the start of the `restartCompleteBluetooth` method (it may be helpful to add a debug line to the top of that so it is easier to identify in the log) to the time you get the exception.

TFreudi1, 2019-02-04:
Well I try, it is much harder to reproduce with this code, than with my simplified, but I still get it, the worst thing is that it is an ANR, so I can't catch it with an exception. As I relized that it happens again, 1 hour ago,  all the logcat output related to it is gone, I must look at my phone/Androidstudio all the time to be in time and still have the logcat things you need.

TFreudi1, 2019-02-06:
Ok, I have a log :-) but because of Murphys rule and nothing happens yesterday, today I also let it run and than press start monitoring and than the ANR happens.
[altbeaconlog.zip](https://github.com/AltBeacon/android-beacon-library/files/2836142/altbeaconlog.zip)

davidgyoung, 2019-02-10:
I think I might see the problem. An excerpt of  the interesting log lines looks like this:

``` 
02-06 10:09:52.635 26368 27327 D BeaconManager: This consumer is not bound.  Binding now: org.altbeacon.beacon.startup.RegionBootstrap$InternalBeaconConsumer@5e0ff25
02-06 10:09:52.636 26368 27327 D BeaconManager: Binding to service
02-06 10:09:52.636 26368 27327 I BeaconManager: Starting foreground beacon scanning service.
02-06 10:09:52.649 26368 26368 I BeaconService: beaconService version 2.15.4 is starting up on the main process
02-06 10:09:52.685 26368 26368 I BeaconService: starting with intent Intent { cmp=org.altbeacon.beaconreference/org.altbeacon.beacon.service.BeaconService }
02-06 10:09:52.686 26368 26368 I BeaconService: starting with intent Intent { cmp=org.altbeacon.beaconreference/org.altbeacon.beacon.service.BeaconService }
02-06 10:09:52.686 26368 26368 I BeaconService: binding
02-06 10:09:52.687 26368 27327 D BeaconManager: This consumer is not bound.  Binding now: org.altbeacon.beaconreference.BeaconReferenceApplication@8c29dc
02-06 10:09:52.687 26368 27327 D BeaconManager: Binding to service
02-06 10:09:52.687 26368 27327 I BeaconManager: Starting foreground beacon scanning service.
```

What I see is that the foreground service is started as a foreground service with `mContext.startForegroundService(intent);` at 10:09:52:646, and it starts up normally, and quickly calls `            this.startForeground(notificationId, notification);` as required to set the notification for the foreground service.   

So far so good.

But then a second method calls `beaconManager.bind(...)` at `02-06 10:09:52.686`.  This second beacon consumer agains starts the process of requesting it as a foreground service with a call to `mContext.startForegroundService(intent);`, but this second time, the foreground service is already started, so it does not again call `mContext.startForegroundService(intent);` within a few seconds causing Android to throw that exception and crash the app 13 seconds later:

```
02-06 10:10:05.152  3796  3815 E ActivityManager: ANR in org.altbeacon.beaconreference
02-06 10:10:05.152  3796  3815 E ActivityManager: PID: 26368
02-06 10:10:05.152  3796  3815 E ActivityManager: Reason: Context.startForegroundService() did not then call Service.startForeground(): ServiceRecord{17ccdfc u0 org.altbeacon.beaconreference/org.altbeacon.beacon.service.BeaconService}
```

This surprises me, but it looks like it can easily be solved.  The immediate fix is not call `beaconManager.bind(...)` twice if you are using a foreground service.  And since using `RegionBootstrap` counts as one  bind, if you are using it, you simply cannot call `beaconManager.bind(...)` after setting up `RegionBootsrap`.

I believe the above restriction can be fixed in the library by modifying the following code in BeaconService.java:

```
                    LogManager.d(TAG, "Binding to service");
                    Intent intent = new Intent(consumer.getApplicationContext(), BeaconService.class);
                    if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O &&
                            this.getForegroundServiceNotification() != null) {
                        LogManager.i(TAG, "Starting foreground beacon scanning service.");
                        mContext.startForegroundService(intent);
                    }
                    else {
                    }
                    consumer.bindService(intent, newConsumerInfo.beaconServiceConnection, Context.BIND_AUTO_CREATE);
```

The problem in the above code is that the call to ` mContext.startForegroundService(intent);`must not be called if another consumer has already bound.  So adding a clause to the if statement should fi it:

```
                    if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O &&
                            this.getForegroundServiceNotification() != null && !isAnyConsumerBound()) {
                        LogManager.i(TAG, "Starting foreground beacon scanning service.");
                        mContext.startForegroundService(intent);
                    }
```
This will prevent it from getting called a second time in the case where multiple consumers are used with a foreground service.

davidgyoung, 2019-02-10:
closed via #828

closed 2019-02-10 by davidgyoung