AltBeacon/android-beacon-library #817: RegionBootstrap constructor parameters

tekinalper, 2019-02-04:
https://github.com/AltBeacon/android-beacon-library/blob/b9876b45acd6c72c84b8d0325062259020eb89ba/lib/src/main/java/org/altbeacon/beacon/startup/RegionBootstrap.java#L49

As an SDK developer, I get Application context from App and implement BootstrapNotifier in the SDK. With the current constructors, I'm not able to set the two different parameter in one. 
I found a solution by creating a custom RegionBootstrap. In case of someone needs such a solution like I did, you may add new constructors or I can send PR if it makes sense for you.  

`public RegionBootstrap(BootstrapNotifier bootstrapNotifier, Application application, Region region) {`

`public RegionBootstrap(BootstrapNotifier bootstrapNotifier, Application application, List<Region> regions) {`

davidgyoung, 2019-02-04:
@tekinalper, I agree it would be useful to more easily separate the context from the notifier.  Is there a way to make this backward-compatible?

It seems such a solution would have to redefine BootstrapNotifier so it no longer has a getApplicatonContext method.  

Thoughts?

tekinalper, 2019-02-04:
Yes, it was the another issue! Currently, I don't initialize RegionBootstrap if the application context is null. But also, setting application context (assuming nonnull) to getApplicationContext method makes me feel not so comfortable. 

It seems it won't be really backward-compatible because of the interface. Making getApplicatonContext optional is not possible. 

How about removing it? :)

davidgyoung, 2019-02-04:
How about making the method signature look like this:

`public RegionBootstrap(Context context, MonitorNotifier monitorNotifier, Region region)`

Then internally you could create a new RegionBootstrap by constructing a new BootstrapNotifier as an anonymous inner class made up of the passed monitorNotifier and context with chained methods.  Something like:

```
mContext = context;
mMonitorNotifier = monitorNotifier;
mBootstrapNotifier = new BootstrapNotifier() {
  public Context getApplicationContext() {
    return mContext.getApplicationContext();
  }
  public void didEnterRegion(Region region) {
    return mMonitorNotifier.didEnterRegion(region);
  }
  ...
  
}
```

tekinalper, 2019-02-04:
Looks much better 👍

davidgyoung, 2019-02-04:
OK, cool.  Do you maybe want to play with an implementation like that in your code, make sure it works OK, then issue a PR?

davidgyoung, 2019-02-10:
Fixed in #821, released in 2.16.1

closed 2019-02-10 by davidgyoung