AltBeacon/android-beacon-library #818: Single RangeNotifier result for multiple regions

tekinalper, 2019-02-04:
Well, I reconsider Beacon detection mechanism in our SDK. So, I'm going to tell what I do first. 

What I need is, ranging beacons in both foreground and background(without foreground service) with different scan and betweenscan periods. I have 9 UUIDs to range if any beacon seen within them. So, is it a good approach to monitor beacons and start ranging if the state is "INSIDE" in didDetermineStateForRegion and stop ranging if it is "OUTSIDE"? 

I tried to do something like: a list of becaonRegions and start ranging once if the size greater than 0 and stop ranging if becaonRegions is empty as all the beacons disappeared from the regions.

- I set multiple regions with the specific UUIDs in RegionBootstrap and also in startRangingBeaconsInRegion method.

Consequently, I would like to have only one result with the beacons in multiple regions. How does it sounds like? 

Currently, I have a global list and adding beacons after each scan callback received from 9 region. But it causes more transactions because of multiple regions during evaluating scan results.

By the way, I know that the other option is also creating only one region with a unique ID and checking the beacons' UUID in the scan result. This one seems more costly to me. What do you think?

davidgyoung, 2019-02-04:
The API is designed as it is (with one ranging callback per region) because it mirrors iOS CoreLocation APIs.  Having APIs work in fundamentally similar ways across Android and iOS is very useful for developers writing apps on both platforms as well as for developers of cross-platform plugins using this library (e.g. Cordova or ReactNative).

You could certainly write some kind of adapter class that does what you want -- it could combine all the beacon collections for each region for each didRangeBeaconsInRegion callback into a single collection, and then make its own callback as soon as they have all come in (you could tell this with a last updated timestamp for each region, verifying you have gotten a recent update for all regions in getRangedRegions(), and once you have, make the adapter callback with the combined list.

The alternative of simply doing a wildcard region, however, is probably more efficient.  The library has to parse each beacon to know if it matches the regions, so unless you are operating in an environment with lots of beacons you don't care about, there is little extra overhead in just getting all beacon detections and filtering yourself for the identifiers you care about.

tekinalper, 2019-02-05:
Thank you for your answer David. I'll consider your advices.

closed 2019-02-05 by tekinalper