AltBeacon/android-beacon-library #819: Why we need to create custom App to enableForegroundServiceScanning

TayfunCesur, 2019-02-04:
As I mentioned on title, I want to know why we need to initialize `beaconManager.enableForegroundServiceScanning(notification, 457)`
on our custom App that extends BeaconConsumer ?

I thought, I can do it for any Activity lifecycle because we just start a foreground service. But when I tried, I didn't work on Activity. It just worked on my custom App.

Is it expected behavior (if it is, why ?) or am I doing sth wrong ?

Thanks. Best regards

davidgyoung, 2019-02-04:
The whole idea of a foreground service is to have the scanning service run independently of what activities are in the foreground.  So it often makes sense to do this outside the lifecycle of an Activity, and make an Application class the BeaconConsumer.

The library reference app shows how to do this in the onCreate method of an Application class for the case where you are using RegionBootstrap to start beacon detections in the background -- before any Activity is launched.  But I see no reason you can't set this up to start with an Activity, just make the Application class continue to be the BeaconConsumer so that when the Activity gets disposed the scanning service doesn't lost its consumer.

TayfunCesur, 2019-02-04:
Thanks for your fast reply David.

Actually I'm not using RegionBootstrap. I'm using rangeNotifier like this;

```
class MainActivity : AppCompatActivity(),BeaconConsumer {
        
        override fun onCreate(savedInstanceState: Bundle?) {
        //...
        beaconManager.enableForegroundServiceScanning(notification, 457) 
        beaconManager.bind(this)

        override fun onBeaconServiceConnect() {
        //Adding range notifier
        }
```

My manifest is here;

```
<service android:name="org.altbeacon.beacon.service.BeaconService" android:enabled="true"
                 android:exported="false"
                 android:isolatedProcess="false"
                 tools:node="replace">
            <meta-data android:name="longScanForcingEnabled" android:value="true"/>
        </service>
```

- I'm doing this way, and my foreground service is stopping when I kill my app. 
My API Level Android 8.1 (Samsung S8)
and Library version 2.15.4

davidgyoung, 2019-02-04:
Right, killing your app will stop your foreground service.  It also kills it if you follow the example in the reference app.

But the difference there is that the fact that the foreground setup happens in application.onCreate causes it to be able to resurrect itself.

The scanning service has the START_STICKY flag set, which restarts the service if the app is killed.  This calls application.onCreate first (just because that is how the Android app lifecycle works) which sets it up again as a foreground service.

In your case it does not have that code in application.onCreate, so when the service tries to restart it gets killed by the OS because nobody made it a foreground service fast enough.

I suppose it would be possible to set this up automatically based on a persisted group of settings, but this effectively means executing that setup code in application.onCreate anyway.  It seems it is just easier to do that yourself, no?

TayfunCesur, 2019-02-05:
Ok David. I am familiar with foreground services and did some projects at the past. I created some example like this and I will be so happy if you point the differences between them.

MainActivity
```
@Override
    protected void onCreate(Bundle savedInstanceState) {
          Intent serviceIntent = new Intent(this, ExampleService.class);
          ContextCompat.startForegroundService(this, serviceIntent);
         //... 
}
```
ExampleService
```
public class ExampleService extends Service {

@Override
    public int onStartCommand(Intent intent, int flags, int startId) {
    //...
     startForeground(1, notification);
     return START_NOT_STICKY;
    }
}
```
AndroidManifest
```
<service android:name=".ExampleService" />
```

This sample works like charm any api level especially Oreo. You are also starting foreground service in your BeaconService but the behaviors is different from my sample.
What do you think?

Thanks a lot.

davidgyoung, 2019-02-05:
What do you mean that your example above "works like a charm"?  Are you saying that if you swipe the app off the screen the foreground service continues running?

TayfunCesur, 2019-02-05:
Yes. Just like your BeaconService. If you want to look inside, I can push my sample on github?