AltBeacon/android-beacon-library #820: Why doesn't the altbeacon library work when the phone has the monitor turned off?

lorenzomalferrari, 2019-02-05:
I'm building an android service with the Altbeacon library. Following the examples found on the documentation and with the help of Stackoverflow, I was able to develop the application.

The problem now is that after some time, when the app is not in the foreground or when the phone is off the monitor, the application stops searching for the beacon.

I'm testing the app on the following phones:

- Huawei 9 lite + (Model TRT-LX1 with Android 7.0)
- Alcatel (Model 5034D with Andorid 8.1.0)
- Alcatel (Model 5033D with Andorid 8.1.0)
- BlackView (Model A20Pro with Andorid 8.1.0)
- Sony Xperia XZ1 (Model G8342 with Android 9)

And everyone has the same problem, there are those who last longer but after a few hours the service stops looking for the Beacons.

To recap: The user who owns the phone with the app installed, must only go around with the phone in his pocket and with the phone locked. The phone will never be used by the person, so in theory, the only processes that run are Android and my App. During the tour the phone will search for the beacons and save the data in a local DB.

Has anyone managed to make the library work for hours in the background? It's possible to do it? Do you need high-performance phones?

All comments and any GitHub repositories will be useful.

Thank you

P.S.

In the gradel I'm using the latest version

`implementation 'org.altbeacon:android-beacon-library:2.+'`

lorenzomalferrari, 2019-02-06:
In these hours I continued to look for a possible solution to the block. In the tests carried out I noticed that if I keep the phone always active the application does not stop.

So I think the blame is more of Android than the Altbeacon library. Does the library allow me to exceed this limit? Is it possible to create a new feature to solve this problem?

davidgyoung, 2019-02-06:
I suspect the problem you are seeing  may have a different cause depending on the device.  Huawei with Android 7 is very different than other devices with Android 8 and 9.

Are you creating a foreground service or use the default job scheduler scanning method used on 8+ devices?

Can you please try to reproduce this problem with the reference app?  I have not personally witnessed  these issues with or without a foreground service with the reference app and it is important to be able to reproduce with a simple set of common reference code to solve this issue.

lorenzomalferrari, 2019-02-06:
This is the piece of the android manifest.xml with my service and the services that are needed for the operation of Altbeacon.

[image]

This is the simplified class of service

`public class MyServiceBeacon extends Service implements BeaconConsumer {

	public static BeaconManager mBeaconManager;
	private BackgroundPowerSaver backgroundPowerSaver;

	@Override
    public void onCreate() {
        super.onCreate();
        
		int secondi = 1000*30;

		mBeaconManager = BeaconManager.getInstanceForApplication(this);

        mBeaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout("m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25"));
        mBeaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout("s:0-1=feaa,m:2-2=00,p:3-3:-41,i:4-13,i:14-19"));
        mBeaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout("x,s:0-1=feaa,m:2-2=20,d:3-3,d:4-5,d:6-7,d:8-11,d:12-15"));
        mBeaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout("s:0-1=feaa,m:2-2=10,p:3-3:-41,i:4-20v"));
        mBeaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24"));
        mBeaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout("m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25"));
        mBeaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout("m:0-3=4c000215,i:4-19,i:20-21,i:22-23,p:24-24"));

        Notification.Builder builder = new Notification.Builder(this);
        builder.setSmallIcon(R.mipmap.ic_launcher);
        builder.setContentText(getString(R.string.fixedNotification));
        Intent intent = new Intent(this, MyServiceBeacon.class);
        PendingIntent pendingIntent = PendingIntent.getActivity(this, 0, intent, PendingIntent.FLAG_UPDATE_CURRENT);
        builder.setContentIntent(pendingIntent);

        if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
            NotificationChannel channel = new NotificationChannel("My Notification Channel ID",
                    "My Notification Name", NotificationManager.IMPORTANCE_DEFAULT);
            channel.setDescription("My Notification Channel Description");
            channel.setImportance(NotificationManager.IMPORTANCE_DEFAULT);
            NotificationManager notificationManager = (NotificationManager) getSystemService(
                    Context.NOTIFICATION_SERVICE);
            notificationManager.getImportance();
            notificationManager.createNotificationChannel(channel);
            builder.setChannelId(channel.getId());
        }

        mBeaconManager.enableForegroundServiceScanning(builder.build(), 456);

        mBeaconManager.bind(this);

        mBeaconManager.setEnableScheduledScanJobs(false);
        mBeaconManager.setBackgroundBetweenScanPeriod(0);
        mBeaconManager.setBackgroundScanPeriod(secondi);

        mBeaconManager.setForegroundScanPeriod(secondi);
        mBeaconManager.setForegroundBetweenScanPeriod(secondi);

        backgroundPowerSaver = new BackgroundPowerSaver(this);

        bluetoothCrashResolver = new BluetoothCrashResolver(this);
        bluetoothCrashResolver.start();
        
        medic = BluetoothMedic.getInstance();
        medic.enablePowerCycleOnFailures(this);
        medic.enablePeriodicTests(this, BluetoothMedic.SCAN_TEST | BluetoothMedic.TRANSMIT_TEST);

    }

	@Override
    public void onBeaconServiceConnect() {
        mBeaconManager.addMonitorNotifier(new MonitorNotifier() {
            @Override
            public void didEnterRegion(Region region) {
                Log.i(TAG, "I just saw an beacon for the first time!");
            }

            @Override
            public void didExitRegion(Region region) {
                Log.i(TAG, "I no longer see an beacon");
            }

            @Override
            public void didDetermineStateForRegion(int state, Region region) {
                Log.i(TAG, "I have just switched from seeing/not seeing beacons: "+state);
            }
        });

        try {mBeaconManager.startMonitoringBeaconsInRegion(new Region("myMonitoringUniqueId", null, null, null));}
        catch (RemoteException e) { Log.e(TAG, "startMonitoringBeaconsInRegion: "+e);  }

        mBeaconManager.addRangeNotifier(new RangeNotifier() {
            @SuppressLint("InvalidWakeLockTag")
            @Override
            public void didRangeBeaconsInRegion(final Collection<Beacon> rangedBeacons, Region region) {
                Intent RTReturn = new Intent(MainActivity.REFRESH_IBEACON_INTENT);
                RTReturn.putExtra("rangedBeacons", (Serializable) rangedBeacons);
                sendBroadcast(RTReturn);
                
				primaryCycle(rangedBeacons);
                
            }
        });

        try {
            mBeaconManager.startRangingBeaconsInRegion(new Region("myRangingUniqueId", null, null, null));
           // mBeaconManager.startMonitoringBeaconsInRegion(new Region("myRegion", null, null, null));
        } catch (RemoteException e) { Log.e(TAG, "startRangingBeaconsInRegion: "+e);  }

    }
	
	private void primaryCycle(Collection<Beacon> rangedBeacons){
		
		//Inserisco Beacons in SQLite DB
	
    }

}`

davidgyoung, 2019-02-06:
@lorenzomalferrari, I'd be glad to help, but I really need to start with a problem that can be reproduced with a lightly modified version of the reference application [here.](https://github.com/AltBeacon/android-beacon-library-reference)  I see in the example shown above that the code creates a new service of its own (in addition to the scanning service , which opens up a whole new set of lifecycle complications.  This is certainly a legitimate way of using the library, but it might be the cause of the issues you are seeing.   Please understand that as a volunteer effort, this project must focus on fixing issues in the library itself vs. troubleshooting  the specific implementations that  use it.

I suspect you may find that if you use the reference app with a foreground service enabled, you won't see these same issues.  The reason may be that the library's service automatically restarts if closed by the OS due to low memory, or other conditions.  You might experiment with the START_STICKY flag.  

Again, my suggestion is that you experiment with the reference app as-is, and see if you can reproduce.   If you can, we can certainly help here.    If you cannot reproduce, then you might try modifying the reference app (one step at a time, testing after each change) until it sets up scanning like in your custom app, and see exactly which change stopped it from working.

lorenzomalferrari, 2019-02-06:
Thanks for the reply.
We will get back to our app as soon as possible to start from yours.

lorenzomalferrari, 2019-03-06:
Good morning,

- We downloaded the application
- Gradle updated;
- Bring the minSdkVersion to 21;
- Deleted the TimedBeaconSimulator class;
- Changed the method of displaying data in the logToDisplay method;
- Set the BeaconManager with the following data:
```
beaconManager.getBeaconParsers().clear();
        beaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout("m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25"));
        beaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout("s:0-1=feaa,m:2-2=00,p:3-3:-41,i:4-13,i:14-19"));
        beaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout("x,s:0-1=feaa,m:2-2=20,d:3-3,d:4-5,d:6-7,d:8-11,d:12-15"));
        beaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout("s:0-1=feaa,m:2-2=10,p:3-3:-41,i:4-20v"));
        beaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24"));
        beaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout("m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25"));
        beaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout("m:0-3=4c000215,i:4-19,i:20-21,i:22-23,p:24-24"));
```
For the scan the code is set as follows:
```
 int secondi = 30 * 1000;

beaconManager.setEnableScheduledScanJobs(false);

beaconManager.setBackgroundBetweenScanPeriod(0);
beaconManager.setBackgroundScanPeriod(secondi);

beaconManager.setForegroundBetweenScanPeriod(0);
beaconManager.setForegroundScanPeriod(secondi);
```

The application during these test days did not give particular problems.

How do we make it a service that starts immediately without the need to click the Start Ranging button?

davidgyoung, 2019-03-06:
You can start ranging automatically in the didDetermineState callback of the Application class by adding a call to startRangingBeaconsInRegion there.  You will also need to add a RangeNotifier in that same class.  This  will eliminate the need to push the button.  

When using the BackgroundPowerSaver class, the library will auto shift to background mode and use background scan periods.  On Android 8+ this is limited to scanning once every 15 min by the operating system.  You can avoid this by using a foreground service and disabling BackgroundPowerSaver.

Read more about Android 8 limits here: http://www.davidgyoungtech.com/2017/08/07/beacon-detection-with-android-8

lorenzomalferrari, 2019-03-11:
BackgroundPowerSaver has been commented.
About foreground service, as described in the documentation page, in the BeaconReferenceApplication class in the onCreate () method I set the required methods.

I have arranged the code in such a way that it starts automatically but after some time, the cycle stops reading the beacon around.

Why? Where do we get lost?

davidgyoung, 2019-03-11:
The answer is different for the different device models you mention in your question.  

On stock Android 8+ if you start a foreground service and disable scan jobs (`mBeaconManager.setEnableScheduledScanJobs(false);`) , it will *not* automatically restart when the app is killed.  If you want auto-restart behavior you must leave scan jobs enabled and not use a foreground service.  This is an OS limitation.  You must choose one or the other.

Huawei devices contain custom code that blocks apps from running more than an hour or so in the background unless they are added to the background white list.  Again, this is a Huawei-specific customization, although OnePlus works similarly.

lorenzomalferrari, 2019-03-12:
At this point we have a doubt.
With the use of this library it is possible to create a service that even without Activity in foreground scans the Beacons every 30 seconds and that every 5 minutes the data is sent somewhere?

The app would be used on Android 7 and 8 devices.

davidgyoung, 2019-03-12:
No, on Android 8 you can only run in the background every ~15 minutes using the job scheduler.  The only alternative is a foreground service.

lorenzomalferrari, 2019-03-12:
And if we had a phone on purpose, where to keep the app open but with the screen off? Being able to keep the service active forever?

davidgyoung, 2019-03-12:
I am not sure I understand the quedstion.  If no foreground service is active and the screen is off (or another app is in the foreground) , Android considers your app in the background.   On Android 8+, backgrounded apps are generally terminated after being in the background ~10 minutes.

lorenzomalferrari, 2019-03-12:
So summarizing.
If we wanted a service that works constantly forever or we develop an app in the background (monitor off or other app on the first floor) that scans at a minimum interval of 15 minutes or a foreground app that scans every 30 seconds.
Have we understood correctly?

davidgyoung, 2019-03-12:
Yes, that is correct on Android 8+, the 15 minute background rate is the
most frequent allowed.  The actual period is not exactly 15 minutes, but
varies +/- 5 minutes by the operating system (so in practice every 10-20
minutes).  The 15 minute jobs are also suspended in Doze mode if the phone
is motionless for several minutes at a time with the screen off when not
charging.

Also, on Android 4.3-7.x you can scan every 30 seconds in the background.

lorenzomalferrari, 2019-03-20:
We have created the application with its recommendations.
We have seen that after several hours the service continues to start but stops reading beacons. We did not notice any errors through the analysis of logcats. Why do you think he stops reading beacons? Blame the bluetooth? Does it come out of the region?

davidgyoung, 2019-03-20:
Which specific device model and operating system do you see detections stop "after several hours"?

lorenzomalferrari, 2019-03-28:
Continuing the analysis of the phones, these days, we noticed this:
We found some phones with Bluetooth turned off. Is it possible that Android automatically turns off Bluetooth? Do you have any suggestions for solving this event?

davidgyoung, 2019-03-28:
I am not aware of any open source Android features that cause bluetooth to be automatically turned off.  However, it is always possible that a proprietary, closed-source battery saving feature of a specific manufacturer's ROM does this.  And if bluetooth is turned off, detections will certainly stop.

One strategy for this is to periodically check if bluetooth is turned off (say every hour or so using the Job Scheduler) You can then send a notification to your user to ask the user to turn bluetooth on.  It is also possible to programmatically turn it on, although you should think hard before doing this whether this is a good idea without the user's permission.

davidgyoung, 2019-05-29:
Closing due to inactivity.

closed 2019-05-29 by davidgyoung