AltBeacon/android-beacon-library #822: PlaceInferenceEngine: No beacon scan available - ignoring candidates.

garyburch, 2019-02-06:
How did you @mspapant  resolve the issue? I am getting same issue with Android 8.1. Please help.   
Below are the  logs : 
01-28 12:59:32.279  4106 26329 I PlaceInferenceEngine: No beacon scan available - ignoring candidates.
01-28 12:59:32.299  4106 26329 I PlaceInferenceEngine: No beacon scan available - ignoring candidates.
01-28 12:59:47.308  4106 22923 I PlaceInferenceEngine: No beacon scan available - ignoring candidates.
01-28 12:59:47.317  4106 22923 I PlaceInferenceEngine: No beacon scan available - ignoring candidates.

_Originally posted by @garyburch in https://github.com/AltBeacon/android-beacon-library/issues/733#issuecomment-458025342_

closed 2019-05-29 by davidgyoung

renaro, 2019-06-07:
was this fixed?

davidgyoung, 2019-06-07:
The log line shown is not part of this library.  I believe it comes from Google Nearby inside the Google Play Services SDK.  This issue is therefore out if scope for this forum.