AltBeacon/android-beacon-library #823: Android O background ranging

tekinalper, 2019-02-06:
### Expected behavior
Ranging beacons in the background
### Actual behavior
Not ranging beacons in the background
### Steps to reproduce this behavior
Kill the app and wait for 15 mins 
### Mobile device model and OS version
Samsung Galaxy S8 - Android 8
### Android Beacon Library version
2.15.4

Although I see it scans after 15 mins, it does not range my beacons. Should I see my beacon parser log(**BeaconParser: Parsing beacon layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25**) as well in the log below? I'm not sure but somehow it seems not including my layout when the app is killed? What may cause this?

No problem in the foreground, I see both parser logs.

_**Logs when app status is killed**:_ 
> 2019-02-06 18:31:06.603 I/vndksupport: sphal namespace is not configured for this process. Loading /vendor/lib64/egl/libGLES_mali.so from the current namespace instead.
2019-02-06 18:31:06.683 D/libEGL: loaded /vendor/lib64/egl/libGLES_mali.so
2019-02-06 18:31:06.726 I/BeaconManager: BeaconManager started up on pid 14765 named 'com.blesh.demo' for application package 'com.blesh.demo'.  isMainProcess=true
**2019-02-06 18:31:06.736 D/BeaconParser: Parsing beacon layout: m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25**
2019-02-06 18:31:07.017 I/CycledLeScanner: Using Android O scanner
2019-02-06 18:31:07.024 I/ScanJob: Using immediateScanJobId from manifest: 208352939
2019-02-06 18:31:07.024 I/ScanJob: Running periodic scan job: instance is org.altbeacon.beacon.service.ScanJob@3f75d50
2019-02-06 18:31:07.027 I/ScanJob: scanJob version 2.15.4 is starting up on the main process
2019-02-06 18:31:07.045 W/ModelSpecificDistanceCalculator: Cannot find match for this device.  Using default
2019-02-06 18:31:07.046 W/ModelSpecificDistanceCalculator: Cannot find match for this device.  Using default
2019-02-06 18:31:07.065 D/BluetoothAdapter: STATE_ON
2019-02-06 18:31:07.067 D/BluetoothAdapter: STATE_ON
2019-02-06 18:31:07.076 D/BluetoothAdapter: STATE_ON
2019-02-06 18:31:07.078 D/BluetoothAdapter: STATE_ON
2019-02-06 18:31:07.080 D/BluetoothLeScanner: Stop Scan with callback intent
2019-02-06 18:31:07.114 D/BluetoothAdapter: STATE_ON
2019-02-06 18:31:07.116 D/BluetoothAdapter: STATE_ON
2019-02-06 18:31:07.124 D/BluetoothAdapter: STATE_ON
2019-02-06 18:31:07.126 D/BluetoothAdapter: STATE_ON
2019-02-06 18:31:07.130 I/ScanJob: Scan job running for 10000 millis
2019-02-06 18:31:07.132 D/BluetoothAdapter: STATE_ON
2019-02-06 18:31:07.135 D/BluetoothAdapter: STATE_ON
2019-02-06 18:31:07.138 D/BluetoothLeScanner: Start Scan with callback
2019-02-06 18:31:07.145 D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=9 mScannerId=0
2019-02-06 18:31:08.680 I/ScanHelper: Non-distinct packets detected in a single scan.  Restarting scans unecessary.
2019-02-06 18:31:10.360 I/zygote64: Do partial code cache collection, code=28KB, data=29KB
2019-02-06 18:31:10.405 I/zygote64: After code cache collection, code=28KB, data=29KB
2019-02-06 18:31:10.405 I/zygote64: Increasing code cache capacity to 128KB
2019-02-06 18:31:13.427 I/zygote64: Do partial code cache collection, code=61KB, data=46KB
2019-02-06 18:31:13.458 I/zygote64: After code cache collection, code=61KB, data=46KB
2019-02-06 18:31:13.458 I/zygote64: Increasing code cache capacity to 256KB
2019-02-06 18:31:17.130 I/ScanJob: Scan job runtime expired: org.altbeacon.beacon.service.ScanJob@3f75d50
2019-02-06 18:31:17.135 D/BluetoothAdapter: STATE_ON
2019-02-06 18:31:17.141 D/BluetoothAdapter: STATE_ON
2019-02-06 18:31:17.148 D/BluetoothAdapter: STATE_ON
2019-02-06 18:31:17.149 D/BluetoothAdapter: STATE_ON
2019-02-06 18:31:17.151 D/BluetoothLeScanner: Stop Scan with callback intent
2019-02-06 18:31:17.163 D/BluetoothAdapter: STATE_ON
2019-02-06 18:31:17.165 D/BluetoothAdapter: STATE_ON
2019-02-06 18:31:17.165 D/BluetoothLeScanner: Stop Scan with callback
2019-02-06 18:31:17.499 I/ScanJob: We are inside a beacon region.  We will not scan between cycles.

chintan-mishra, 2019-02-08:
Can you confirm if a job is scheduled on the device even after the app is killed?

This issue persists in many devices. Since background execution limits have been imposed this issue is much more persistent. I have implemented a [`SyncManager`](https://developer.android.com/training/sync-adapters/) in my [app](https://8hoot.com). I have noticed some improvement in the background scanning.

davidgyoung, 2019-02-08:
Yes, a 15 min persistent job, which the library uses by default to schedule scans on Android 8+, fires even after an app is killed.  This is true on AOSP and other devices I have tested.

TayfunCesur, 2019-02-09:
My personal device is also Samsung S8. Since the android O have much more background execution limits, I've decided to use foreground service and it works as its expected behavior(I tested many cases such as power save mode, doze mode, reboot phone vs.) but if you don't want to show notification which is pinning on the top of screen, your job is so hard. Because there can't be exists a job which is scanning on background in Oreo and above versions.

davidgyoung, 2019-02-09:
@tekinalper, for your specific issue, you should be able to get this working.  If your parser is not showing up in the logs, this is likely the issue. Can you show the code where you set it up in context? This should be in the onCreate method of a custom Application class.

tekinalper, 2019-02-13:
Finally, I solved the issue. I had changed the flow in my case. So, just noticed that I was initializing BeaconManager in another method called from Activity instead of the the one that called from onCreate of Application class. False alarm! :)

closed 2019-02-13 by tekinalper