AltBeacon/android-beacon-library #825: New logo/icon proposal

mansya, 2019-02-10:
Good day sir. I am a graphic designer and i am interested in designing a logo for your good project. I will be doing it as a gift for free. I just need your permission first before I begin my design. Hoping for your positive feedback. Thanks

davidgyoung, 2019-02-10:
Hi @mansya, yes contributions are always welcome -- no permission needed!  While the library itself doesn't have an app icon, the reference app does, and there is a (pretty outdated) beacon icon on the project web page [here](https://altbeacon.github.io/android-beacon-library/).  

Graphic designs are always a bit subjective, so maybe you could sketch a couple of quick ideas and link them to this issue.  We could then solicit feedback from folks who use the library and maybe vote on our favorite before you finalize anything.

mansya, 2019-05-26:
HI, sir, I have designed it. How ? do you like it ? if yes please tell me which one you like.
[image]

mansya, 2019-07-04:
hi sir. how about this offer. don't you want to continue?

closed 2019-09-18 by mansya