AltBeacon/android-beacon-library #826: Cannot build library as subproject

davidgyoung, 2019-02-10:
### Expected behavior

Library should work to be included as a subproject of another project.

### Actual behavior

When you try to compile the library as a subproject with the steps below, you get the error:
```
Could not determine the dependencies of task ':app:compileReleaseJavaWithJavac'.
> Could not resolve all task dependencies for configuration ':app:releaseCompileClasspath'.
   > Could not resolve project :android-beacon-library.
     Required by:
         project :app
      > Unable to find a matching configuration of project :android-beacon-library: None of the consumable configurations have attributes.
```

### Steps to reproduce this behavior
1. git clone https://github.com/AltBeacon/android-beacon-library.git
2. git clone https://github.com/AltBeacon/android-beacon-library-reference.git
3. make s symbolic link from an android-beacon-library subfolder of the reference app to the library repo above (or copy the folder)
4. edit the reference app's settings.gradle to uncomment the line: `//include ':android-beacon-library'`
5. edit the reference app's app/build.gradle to uncomment `compile project(':android-beacon-library')` and comment out `compile 'org.altbeacon:android-beacon-library:2+'`
6. run `./gradlew build`

### Mobile device model and OS version
N/A - build issue

### Android Beacon Library version
code tagged as 2.16 (but still unreleased)

@TonyTangAndroid, I believe this problem came up as a result of the reorganization of the library so it has a top-level gradle file and one under the /lib folder.  I tried a few settings changes to try and get this to work again, but to no avail.  Can you please take a look and let me know if you see anything that might help?

I know that you had suggested reorganizing the reference app so it is included in this app, but even if we do that, I think the library should support being used as a subproject dependency as before.

TonyTangAndroid, 2019-02-11:
Cool. I will look into it and get you back in 24 hours. I will keep you posted if any road block comes up.

davidgyoung, 2019-02-11:
Thanks, @TonyTangAndroid.  FYI, I did make a change in #829 that named the lib folder project name as android-beacon-library, which fixed the artifact naming, but did not resolve this problem.  I am not sure why.

TonyTangAndroid, 2019-02-11:
@davidgyoung If the whole purpose of this issue is to provide the capability of referring beacon library source code directly rather than gradle dependencies from maven, then I have managed to resolve it from reference app. Please refer to this [pull request](https://github.com/AltBeacon/android-beacon-library-reference/pull/52) for the resolution, which is actually quite simple. Let me know this works for you or not.

PS : I have been intended to write more easy and simple beacon library samples and provide more utility tools to track the log of beacon event. And I believe I have made good progress. Be hold that I will make more pull requests from reference app repository.

closed 2019-04-04 by davidgyoung