AltBeacon/android-beacon-library #836: didEnterRegion sometimes not firing

stephenruda, 2019-02-22:
### Expected behavior
didEnterRegion to call any time I enter the region from an exited state.

### Actual behavior
didEnterRegion is sometimes not getting called (seemingly randomly).

### Steps to reproduce this behavior
Unplug my Beacon.  Wait for about ~10 minutes until I have gotten a didExitRegion event.  Plug the Beacon back in and my didEnterRegion only fires sometimes.

### Mobile device model and OS version
Samsung Galaxy S7 on Android 8.0

### Android Beacon Library version
v 2.16.1

Here is some debug logs from around the time I plugged the Beacon back in...
https://pastebin.com/raw/JtERQ1N8

Specifically, I had unplugged the Beacon at 14:00:00, a few minutes before the logs start.  I plugged the Beacon back in around 14:07:58.
Interestingly, you can see that it is trying to notify me that we have entered, I just never get the entry event...

"2019-02-22 14:07:58.219 5114-24731/com.simpleinout.android D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.monitor_notification"

The Beacon remains plugged in for the remainder of the logs.

If I unplug the Beacon I get the didExitRegion event (despite didEnterRegion never getting called).  I have never had any issues with didExitRegion firing, only didEnterRegion.

Let me know if you need any implementation details.

Thanks!

davidgyoung, 2019-02-22:
This may be an issue of how the notifiers are set up not working well with the Android lifecycle.

Can you please:
1. Try with the library reference app and see if you see the same
2. Show your monitoring setup and notifier assignment (in context)

stephenruda, 2019-02-22:
I will try to reproduce with the reference library app.

Some notes on my implementation...
I am using the RegionBootstrap / BootstrapNotifier combined with a foreground service.  It is my own custom foreground service as I do multiple things within the service (not just Beacon scanning).

On application create, I start my foreground service.  Here is the relevant code for my custom foreground service...  The foreground service runs 24/7.
```
if(beaconHelper == null) { 
     beaconHelper = new BeaconHelper();
}
beaconHelper.setForegroundScanPeriods();
beaconHelper.beaconManager.setBackgroundMode(false);
```

So my foreground service instantiates the BeaconHelper, here is the primary chunk of the BeaconHelper class.
```
public class BeaconHelper implements BootstrapNotifier {

    public BeaconManager beaconManager;
    private RegionBootstrap regionBootstrap;
    protected Context context;

    public BeaconHelper(final Context context) {
        this.context = context;
        this.beaconManager = BeaconManager.getInstanceForApplication(context);
        this.beaconManager.setAndroidLScanningDisabled(true);
        setForegroundScanPeriods();
        try {
             this.beaconManager.getBeaconParsers().add(new 
             BeaconParser().setBeaconLayout("*REDACTED*"));
        }
        catch(Exception e) {
             e.printStackTrace
        }

        List<Region> regions = new ArrayList<>();
        regions.add(new Region(myBeacon.id, Identifier.parse(myBeacon.uuid), null, null));

        regionBootstrap = new RegionBootstrap(this, regions);
    }

    public void setForegroundScanPeriods() {
            this.beaconManager.setRegionExitPeriod(90000l);
            this.beaconManager.setForegroundBetweenScanPeriod(40100);
            this.beaconManager.setForegroundScanPeriod(4100l);
    }
}
```

I know my implementation is a little different than most peoples - but I hadn't experienced any issues until the last few months.  My production build is still using version 2.15.1 of the library and I haven't experienced any issues with it there.  It wasn't until I upgraded to 2.15.4 that I started to have issues.  My guess is that something must have changed between 2.15.1 and 2.15.4 that started causing the problem.

Hopefully this helps.

TayfunCesur, 2019-02-27:
I also faced this issue nowadays. While some ble scanner apps is showing my beacon, my app does not detect the beacon sometimes. I've no idea why it happens. My version is also 2.15.4. Now I downgrade to 2.15.1 and keep watching for this issue

stephenruda, 2019-02-27:
Just a quick update...  When I started noticing this problem, I had guessed it was because of an update to the Beacon Library I had taken.  I also had updated `targetSdkVersion` from 27 to 28.  So this started happening due to an update within the Beacon Library or due to targeting the latest SDK version.  Not sure which one would be the cause.  I will be testing both scenarios.

davidgyoung, 2019-02-27:
This issue might be caused by a library change that added non-empty scan filters for background scans on Samsung devices.  This was needed because Samsung devices with 8+ started blocking any background scan with the scan filter not set.  The scan filter is based on your beacon layout bytes plus the manufacturer code.   That manufacturer code defaults to 0x004c (Apple) for iBeacon and 0x0215 (Radius Networks) for AltBeacon.  If you have a beacon that is using a different company code, this will fail.  You can see if this is true by reviewing a foreground scan log that shows the raw bytes of your beacon advertisement when it does match.  

If this is indeed the problem, then there is a way in the library to set a custom company code of your beacon layout so this will work again.

TayfunCesur, 2019-02-28:
Thank you for you great explanation David. I'm not sure this is not happening only Samsung devices. I'm facing this issue also Xiaomi devices also. I'll be watching the logs. 

Can you give us an idea about the way you said which is set custom company code?

Thank you

davidgyoung, 2019-02-28:
See this method on BeaconParser here:

https://altbeacon.github.io/android-beacon-library/javadoc/org/altbeacon/beacon/BeaconParser.html#setHardwareAssistManufacturerCodes-int:A-

The default list of manufacturer codes is set to 0x004c (apple) for all beacon formats except AltBeacon.  This is generally what you want, unless you find that your particular iBeacon is using a non-standard manufacturer code.

Major caveat:  This is just a guess as to one thing that may be happening.  It is difficult to impossible to help more proactively with this issue because of the extensive customizations in the way the library is used.  (the special scanning service, the non-standard scan intervals, the custom range exit period, etc.)   

I suggest that you try to find out what specifically is triggering this by starting with the reference app and verifying it works as expected.  Then modify the reference app to apply each of your customizations one by one until you find the one that makes the unexpected behavior appear.  If you can narrow it down in this fashion it will be much easier to find and solve the problem.  What we need is a way to reproduce with a lightly modified version of the reference app vs. a custom application.

TayfunCesur, 2019-02-28:
Thank you again David. I checked it now. My beacon parser is below;

`val beaconParser = BeaconParser().setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24")`

After this, I looked default HardwareAssistManufacturers via

`beaconParser.getHardwareAssistManufacturers()` and it returned only one data which is 76.
Then, I searched 76 value in here (https://www.bluetooth.com/specifications/assigned-numbers/company-identifiers) and I found 0x0076  = Creative Technology Ltd.

Is this means that you are supporting only this manufacturer and If I use some different company's beacon, I may face this issue ? 

Please correct me, if I'm wrong.

Thank you

davidgyoung, 2019-02-28:
Nah, 76 is UST the decimal representation of the hex number 0x4c.  Both indicate the apple manufacturer code.

so that just confirms the default library setup.  what you need to check is what manufacturer code your beacon is 
actually transmitting.  for that you need to get raw logs of a detection's hex bytes on android

TayfunCesur, 2019-02-28:
The how can I get the raw logs that you mentioned ? Is it default logging when debug mode ?

davidgyoung, 2019-02-28:
yes it is by default in debug mode

stephenruda, 2019-02-28:
I don't think that it has to do with the Samsung scan filters because I have been using Samsung on Android 8 for a long time and wasn't having issues until recently.

Specifically, I have two versions of my app in the store...

1) Production - Uses version v2.15.1 of the Beacon Library and has targetSdkVersion 27.  Beacons scan 24/7 with a foreground service in foreground mode.

2) Beta - Uses v2.16.1 of the Beacon Library and has targetSdkVersion 28.  Beacons scan 24/7 with a foreground service in foreground mode.

I didn't change any other Beacon Scanning related code between production and beta.  So my initial guess is that either the Beacon Library update is causing the problem or the targetSdkVersion is causing the problem.  I suppose it is possible that my phone took a new update in the last few months and I didn't notice the problem until recently.

You say that Samsung 8+ devices started blocking background scans - would this apply to foreground scans if we are using a foreground service and the Beacon Library in foreground mode?

What really confuses me is that it looks like everything is working and it detects the Beacon - the callback just isn't working.
You can see in the logs that I attached that I get... 
`D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.monitor_notification` but that broadcast never gets sent.
A few milliseconds after the above log, I get...
`ScanState: Scan state restore regions: monitored=1 ranged=0`
`BeaconLocalBroadcastProcessor: Register calls: global=115 instance=115` which makes me think that the monitor_notification is fired before the receiver is registered.

The scan is clearly working and detecting the Beacon, the callback just isn't working because it is seemingly unregistered.

I am trying to isolate if this issue is related to the latest version of the Beacon Library or if this is related to my updated targetSdkVersion.  Still working on that.

davidgyoung, 2019-02-28:
Yes, it could be a subtle timing issue of the BeaconLocalBroadcastProcessor starting up later.  It's unclear, though, what triggers that -- I certainly do not see this with Android 8+ devices using a foreground service with the reference app.  

You might check to see how long it takes for the call to `regionBootstrap = new RegionBootstrap(this, regions);` from the time the app starts up.  This really needs to be very fast otherwise you will see problems like you describe. In the reference app, we trigger this in the onCreate method of the Application class.  This way if there is a background detection that auto-launches the app, that callback will be called and the listeners will be set up to get the callbacks from ranging and monitoring before the service starts up and delivers them.   If you have logic that causes some kind of delay that would be problematic.

The Samsung blocking background scans without a scan filter applies to any app when the screen is off, even if it is a foreground service.    So it is the screen illumination status not the service type that determines this blocking.

stephenruda, 2019-02-28:
Okay, I will try to test all the things discussed.  It is tricky because it happens randomly.  I have noticed that this problem generally occurs when the screen is off and I haven't been using the device.  I have never experienced the problem while the phone is plugged in and charging (so when debugging, I have to make sure to keep it unplugged).

I am calling `regionBootstrap = new RegionBootstrap(this, regions);` in the onCreate of my application class.  I do a few other initializations before it, but nothing that should slow it down significantly.  I could try moving this around but none of this logic has changed in over a year.

I don't doubt that the sample app works as expected.  It could be an implementation detail of mine that is causing it to work unexpectedly.  I think my foreground service works a little differently than the one provided by the library.  I am just trying to figure out why my foreground service stopped working and if there is a way I can get it working again without having to use the foreground service in the library.  As I said before, my foreground service does a lot of stuff unrelated to Beacon scanning.  If I used the foreground service built into the beacon library, I would essentially need to run two foreground services (1 for beacon scanning and 1 for everything else).

TayfunCesur, 2019-03-04:
Hello @davidgyoung. I found a log when beacon found like this.
`D/BtGatt.GattService: onScanResult to scannerId: 6- event_type=0x1b, address_type=0, address=78:0, primary_phy=1, secondary_phy=0, advertising_sid=0xff, tx_power=127, rssi=-88, periodic_adv_int=0x0`

which one of these parameter is bluetooth manufecturer id that you mentioned before ?

Thanks

davidgyoung, 2019-03-04:
Sorry, no manufacturer id in that log line.  Look for a log line like one of those below -- logged when the library detects your beacon with beaconManager.setDebug(true) -- which show the manufacturer id in bold:

D/BeaconParser: Bytes are: 0201061aff**4c00**0215df96db9f46cb4c1dab92ef169f30ea183f4f14f9b6040945535403030f180e160a184f3ff9140fd3b64f3ff9140000000000000000

TayfunCesur, 2019-03-04:
There is only 
`D/BeaconParser: Parsing beacon layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24`
line about BeaconParser David. My library version is 2.15.4 and as I see 

`beaconManager.setDebug(true)` is deprecated. Also I put a breakpoint to your libraries setDebug method and it didn't triggered.

My sample code is below

```        beaconManager = BeaconManager.getInstanceForApplication(this)
        beaconManager.beaconParsers.add(BeaconParser().setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24"))

beaconManager.enableForegroundServiceScanning(notification, 457)

```

TayfunCesur, 2019-03-05:
> Okay, I will try to test all the things discussed. It is tricky because it happens randomly. I have noticed that this problem generally occurs when the screen is off and I haven't been using the [...]

Dear friend, I am facing also this issue and I use range notifier. I saw the ranging notifier not triggerred when my phone is near to beacon. So, I think this is not about your implementation. 

If I saw the notifier triggers and other process stuck, I would aggree you.

EnesKy, 2019-03-08:
Hi, i'm struggling with this problem too. 
I tried it on Xiaomi Mi 5 and Samsung Galaxy A5 2016.
I am using the latest version of the library.
I never seen didEnterRegion or didExitRegion fired.

Did anyone resolved this issue?

davidgyoung, 2019-03-08:
@EnesKy, I do not think the problem you are having is related to this issue.  You say you never get a didEntry callback.  That certainly suggests a setup problem. Please try the reference app and see if it works for you.  if you need help, please open a question on StackOverflow.com.    Please leave this issue focussed on the intermittent problem @stephenruda mentioned above.

stephenruda, 2019-03-08:
@davidgyoung I have been working on creating a sample project that reproduces this problem.  Should we be using `beaconManager.setBackgroundMode(true);` or `beaconManager.setBackgroundMode(false);` when using the Foreground Service?

davidgyoung, 2019-03-08:
Either are legal and it generally only affects whether foreground or background scan periods are used.   For clarity I recommend beaconManager.setBackgroundMode(false);  and using foreground scan intervals with a foreground service

stephenruda, 2019-03-08:
@davidgyoung I forked the reference application and [built an application that reproduces the problem.](https://github.com/stephenruda/android-beacon-library-reference/tree/reproduce-problem)
It took me about an hour or two in order to encounter the problem, follow these steps...
1. Build project and install app onto phone.  I am using Samsung Galaxy S7 on Android 8.0.0
2. Once app is running, press button to start the foreground service.  This should present a notification.  You should also get a notification saying you entered a Beacon if you have one plugged in.
3. Unplug phone from computer and close app.  Foreground service should still be running.
4. Unplug Beacon, you should get a notification that you have exited the Beacon region within a minute or two.
5. Wait ~15 minutes and plug the back in.  Some times the didEnterRegion event will be fired, some times it does not.  If you do get the didEnterRegion event just repeat steps 4 and 5.  I believe it took me about 7 attempts before I encountered the problem.  It is seemingly random.

There are some dramatic differences between the way I am using Foreground Services in this sample application versus the way you are using them in the initial reference application.  This most prevalent thing is that I never call `enableForegroundServiceScanning(builder.build(), 123);` or `setEnableScheduledScanJobs(false);`.  I just build my own Foreground Service because in my actual production application that service is responsible for multiple things - not just Beacon scanning.  I believe that I might just have to bite the bullet and run two foreground services, 1 for the Beacon Library and 1 for all of my other actions unrelated to Beacons.  (I use the single Foreground Service for other things like WIFI scanning and Location scanning).

davidgyoung, 2019-03-08:
OK, thank you for sharing the code.  Do you think you can get a log capture of the failure with this setup?

(connect via USB, then issue `adb tcpip 5555`), then disconnect USB type `adb connect x.x.x.x` where x.x.x.x is your phone's IP address on the same subnet as your workstation.

You'll need to capture all logs to a file with:

adb logcat > logfile.txt

Then repeat the steps to reproduce.  Once you do, note the date and time, and pull a log excerpt from the file for the previous 60 seconds (at least a few seconds before you turned on the beacon.)

stephenruda, 2019-03-08:
@davidgyoung 
I have done as you asked.  [Please take a look.](https://raw.githubusercontent.com/stephenruda/android-beacon-library-reference/8e240b55631d33df5f2a8c89916275e97c08dee7/logfile.txt)

Specifically, the Beacon was unplugged at 17:00:00.  I left the phone idle with the screen off on my desk and it was checked OUT of the Beacon.  About 15 minutes later I plugged the Beacon back in.  You can see in the logs exactly when I plugged it in. `17:16:19.847 22672 25160 D Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.monitor_notification`.  I have captured the logs a few minutes before plugging the Beacon in and a few minutes after plugging it in.  If you need more logs, let me know.  I trimmed some off the beginning and end of the file just so I could get the logs around the time I plugged the Beacon in.  I did not get a didEnterRegion event.

davidgyoung, 2019-03-09:
@stephenruda, a few observations below after reviewing your setup and the log.

There are a few unusual things going on but nothing that I can see causes the callback failure you are seeing.   I suspect for some reason that the handler that converts local broadcast intents to MonitorNotifier callbacks is simply not registered early enough in your case.

I  have made a change that sets this up earlier to see if it fixes your problem.  Please give this beta release a try based on #842

https://github.com/AltBeacon/android-beacon-library/releases/tag/2.16.2-beta1

Below are the other observations just for my notes.

* The app is configured to scan for about 4 seconds every 44 seconds regardless of background mode
* The app is using the JobScheduler to accomplish the above, and since Android limits background jobs to once eery 15 minutes (even with a foreground service) the app is effectively scanning for 4 seconds every 15 minutes.
* Each time a ScanJob starts up (about every 15 minutes) there will be four seconds of active scanning and 40 seconds of low power scanning (the exact boundaries of these are normalized on the clock boundary of even scan cycles)
* A low power scan cycle existed from 17:12:40.489 to  17:13:20.524 
* A high power scan cycle existed from 17:13:20.524  17:13:24.663 
* Cycling like above continued on a 4/40 cycle
* The phone entered doze mode at 17:14:03.870, cancelling the ScanJob and stopping scanning
* The ScanJob set up an intent-based background scan
* The scanning failed to stop cleanly for some reason -- it resumed the cycled scanning at 03-08 17:14:04.706, keeping the ScanJob running beyond when the OS told it to stop.
* An intent-based background scan was delivered matching the beacon at 03-08 17:16:19.838. This causes an immediate scan job to be scheduled
* The overdue running ScanJob also detects that beacon.  
* A region entry event is attempted to be delivered via intent at 03-08 17:16:19.847
* The immediate scan job is scheduled at 03-08 17:16:19.880

stephenruda, 2019-03-11:
Hi David, I will try this out later this week.  Thanks a lot!

stephenruda, 2019-03-15:
@davidgyoung I tried following your steps to use 2.16.2-beta1 but I got this error:
`Supplied String module notation 'android-beacon-library-2.16.2-beta1' is invalid.`
Then instead I tired using `implementation 'android-beacon-library:2.16.2-beta1@aar'` and the error I got was `Failed to resolve: android-beacon-library:2.16.2-beta1:`.  Any idea what I might be doing wrong?  I have added the flatDir lines and the put the aar file in the libs folder.

stephenruda, 2019-03-15:
@davidgyoung [You can see the changes I made here.](https://github.com/stephenruda/android-beacon-library-reference/commit/13cc83ef0c2fb33f67b4f098369c5110a420521a)  If you inspect that branch you can view the file structure.

davidgyoung, 2019-03-15:
@stephenruda, sorry for the trouble.  Gradle is pretty picky about the syntax here, and newer versions keep changing the rules.  I also noticed I had a - in a place where I had a colon.  I have updated the instructions on the release to what just worked for me to try with the reference app:

* Edit your outermost build.gradle file to add an libs entry like so:

```
     allprojects {
         repositories {
             google()
             jcenter()
             flatDir {
                 dirs 'libs'
             }
         }
     }
```

* Edit your innermost build.gradle file to add the beta library AAR as a dependency like so:

```
dependencies {
    implementation 'org.altbeacon:android-beacon-library:2.16.2-beta1@aar'
}
```

* Create a /libs folder next to the innermost build.gradle above, then download and copy the aar file attached to this release into this folder.

There are some subtle changes there (note the : in the implementation line and the location of the flatDirs folder.)

stephenruda, 2019-03-15:
@davidgyoung I have spent several hours testing the new version (just sitting here unplugging the beacon and plugging it back in).  So far so good!  I will continue to test it over the weekend.  If I still haven't experienced the problem by Monday I believe it can be marked as fixed.
As a side note, I have gone ahead and started implementing the Foreground Service within the Beacon Library rather than my own work-around foreground service.  I am able to use one notification that displays for my own foreground service and the Beacon library foreground service so the user should never notice a change.

davidgyoung, 2019-03-16:
Good to hear @stephenruda.  If you have some instructions you can share about setting up a combined notification for your foreground service and the library's, please do.  I can then add this to the foreground service docs for the library.

stephenruda, 2019-03-18:
Hi David, all is good.  Feel free to close this issue.
Setting up one Notification between the two services was actually pretty easy.  [I followed this answer for updating the foreground notification on the fly.](https://stackoverflow.com/a/20142620/6754511)
Essentially, all you have to do is update the NotificationManager with a new notification and the same notification id and it will overwrite the previous foreground notification.  This way I can use one notification for two services and update the text of the notification at any time.  Thanks!

davidgyoung, 2019-03-18:
Fixed in #842

closed 2019-03-18 by davidgyoung