AltBeacon/android-beacon-library #837: java.lang.NullPointerException:  at android.os.Parcel.readException (Parcel.java:1546) on Android 5.0

lucky1412, 2019-02-26:
java.lang.NullPointerException: 
at android.os.Parcel.readException (Parcel.java:1546)
at android.os.Parcel.readException (Parcel.java:1493)
at android.app.job.IJobCallback$Stub$Proxy.jobFinished (IJobCallback.java:167)
at android.app.job.JobService$JobHandler.handleMessage (JobService.java:147)
at android.os.Handler.dispatchMessage (Handler.java:102)
at android.os.Looper.loop (Looper.java:145)
at android.app.ActivityThread.main (ActivityThread.java:5942)
at java.lang.reflect.Method.invoke (Native Method)
at java.lang.reflect.Method.invoke (Method.java:372)
at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run (ZygoteInit.java:1400)
at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:1195)
Android 5.1
115
53.0%
Android 5.0
60
27.6%
Android 6.0
42
19.4%

lucky1412, 2019-02-26:
Please any budy help me for this i cant understand why its happening

davidgyoung, 2019-03-15:
This is an internal Android exception.  I see no references to any library code in the stack trace.  Why do you think it has something to do with this library?

lucky1412, 2019-03-15:
I don't understand wt u want to say.. please give me solution.
thanks
Santosh

davidgyoung, 2019-03-15:
Sorry, @Santos, please understand this project is a volunteer effort.  We need users to assist in resolving these kinds of issues. 

Based in the info provided I cannot help because the crash is not related to the library as far as I can tell.  If you find evidence to the contrary, please share.

closed 2019-03-15 by davidgyoung

lucky1412, 2019-03-15:
31 minutes ago on app version 12
Samsung Galaxy Note3 (ha3g), Android 5.0
Report 1
java.lang.NullPointerException: 
at android.os.Parcel.readException (Parcel.java:1546)
at android.os.Parcel.readException (Parcel.java:1493)
at android.app.job.IJobCallback$Stub$Proxy.jobFinished (IJobCallback.java:167)
at android.app.job.JobService$JobHandler.handleMessage (JobService.java:147)
at android.os.Handler.dispatchMessage (Handler.java:102)
at android.os.Looper.loop (Looper.java:145)
at android.app.ActivityThread.main (ActivityThread.java:5942)
at java.lang.reflect.Method.invoke (Native Method)
at java.lang.reflect.Method.invoke (Method.java:372)
at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run (ZygoteInit.java:1400)
at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:1195)