AltBeacon/android-beacon-library #838: getDistance() returns too large value

mittay, 2019-02-28:
### Expected behavior
Expected  to see result of Beacon.getDistance() is 0,1-0,5 meters (beacon is very close to device) 
### Actual behavior
Method Beacon.getDistance() return too large value like this: 1.891023809148152E7  (**18910238 meters**)
Here is log:
02-27 17:24:34.820 D/CurveFittedDistanceCalculator( 6704): calculating distance based on mRssi of -37.89473684210526 and txPower of -3
02-27 17:24:34.820 D/CurveFittedDistanceCalculator( 6704): avg mRssi: -37.89473684210526 distance: 1.891023809148152E7

### Steps to reproduce this behavior
android-beacon-library-reference app, Galaxy S7 and FeasyBeacon. 
Monitoring view of app will show values like: "The first beacon id1: fda50693-a4e2-4fb1-afcf-c6eb07647825 id2: 10065 id3: 26049 is about **1.891023809148152E7**  meters away"
### Mobile device model and OS version
Galaxy S7
Beacon : https://www.amazon.com/programmable-Battery-Bluetooth-eddystone-Technology/dp/B078N2B7RD/ref=pd_lpo_sbs_147_t_1?_encoding=UTF8&psc=1&refRID=QJA2N9HX7CR4WFYM6FKX
Enabled in iBeacon format .TxPower set to -3

### Android Beacon Library version
 'org.altbeacon:android-beacon-library:2.16.1'

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2019-02-28:
The log line:  `
calculating distance based on mRssi of -37.89473684210526 and txPower of -3` indicates that your beacon is not calibrated correctly.  You must set the txPower or measuredPower value inside your beacon with the expected RSSI value at 1 meter.  In your case this value is -3, which is certainly not correct, and is causing your distance estimates to be very wrong.  If you are 1 meter away, a typical value would be -59. 

Please consult your beacon manufacturer's documentation on how to calibrate your beacon and set this value.

closed 2019-02-28 by davidgyoung

mittay, 2019-03-06:
Thank you, fixed! Also thank you for the lib!