AltBeacon/android-beacon-library #840: Keeping scan even if I stopped service

TayfunCesur, 2019-03-04:
### Expected behavior
Stopping scan and not detection

### Actual behavior
After I call that lines
```
beaconManager.removeAllRangeNotifiers()
beaconManager.disableForegroundServiceScanning()
```

I still see that logs
```
D/BeaconParser: Processing pdu type FF: 0201***** with startIndex: 5, endIndex: 29
D/BeaconParser: This is a recognized beacon advertisement -- 02 15 seen
D/ScanHelper: Beacon packet detected for: id1: 7427**** id2: 10 id3: 7 with rssi -82
D/BeaconParser: Bytes are: 0201*****
D/ScanHelper: beacon detected : id1: 7427**** id2: 10 id3: 7
D/ScanHelper: looking for ranging region matches for this beacon
D/ScanHelper: matches ranging region: id1: null id2: null id3: null
D/RangeState: adding id1: 7427**** id2: 10 id3: 7 to existing range for: org.altbeacon.beacon.service.RangedBeacon@1fd3a03
D/ScanHelper: Beacon packet detected for: id1: 7427**** id2: 10 id3: 7 with rssi -82
D/ScanHelper: beacon detected : id1: 7427****: 10 id3: 7
```

### Steps to reproduce this behavior
I'm definitely sure that services is not running after I stopped. I checked my working services under Developer options and I saw my services is not running. So, where does this logs come from?

### Mobile device model and OS version
Samsung  Galaxy J5 Prime- OS 8.0.0

### Android Beacon Library version
2.15.4

davidgyoung, 2019-03-04:
you also need to call beaconManager.unbind(...)  or regionBootstrap.disable() depending on which you are using.

TayfunCesur, 2019-03-04:
I prepared some example code snipped based on your reference app. 

It started when app launches and after stop and start service, range notifier triggers twice in 1100 second and also logs goes double. My sample is below

**App.kt**

```
class App : Application(), BeaconConsumer {

    lateinit var beaconManager: BeaconManager

    override fun onBeaconServiceConnect() {
        beaconManager.addRangeNotifier { p0, p1 ->
            if (p0?.isNotEmpty()!!) {
                showNotification("Found Beacon")
            }
        }
        beaconManager.startRangingBeaconsInRegion(Region("asddsa-sds-sdsadasd-afcf-gasdaj", null, null, null))
    }

    fun stop() {
        beaconManager.unbind(this)
        beaconManager.removeAllRangeNotifiers()
        beaconManager.disableForegroundServiceScanning()
    }

    fun start() {
        beaconManager = BeaconManager.getInstanceForApplication(this)
        BeaconManager.setDebug(true)
        beaconManager.beaconParsers.add(BeaconParser().setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24"))
        val builder = Notification.Builder(this)
        /*
        * ...
        * */
        beaconManager.enableForegroundServiceScanning(builder.build(), 457)

        beaconManager.backgroundBetweenScanPeriod = 0
        beaconManager.backgroundScanPeriod = 1100

        beaconManager.bind(this)
    }

    override fun onCreate() {
        super.onCreate()
        start()
    }

```

**MainActivity.kt**

```
class MainActivity : AppCompatActivity() {

    private val PERMISSION_REQUEST_COARSE_LOCATION = 1

    override fun onCreate(savedInstanceState: Bundle?) {
        super.onCreate(savedInstanceState)
        setContentView(R.layout.activity_main)

        start.setOnClickListener {
            (application as App).start()
        }

        stop.setOnClickListener {
            (application as App).stop()
        }

        if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.M) {
            // Android M Permission check
            if (this.checkSelfPermission(Manifest.permission.ACCESS_COARSE_LOCATION) != PackageManager.PERMISSION_GRANTED) {
                val builder = AlertDialog.Builder(this)
                builder.setTitle("Location permission")
                builder.setMessage("Location permission")
                builder.setPositiveButton(android.R.string.ok, null)
                builder.setOnDismissListener {
                    requestPermissions(
                        arrayOf(Manifest.permission.ACCESS_COARSE_LOCATION),
                        PERMISSION_REQUEST_COARSE_LOCATION
                    )
                }
                builder.show()
            }
        }
    }

    override fun onRequestPermissionsResult(
        requestCode: Int,
        permissions: Array<String>, grantResults: IntArray
    ) {
        when (requestCode) {
            PERMISSION_REQUEST_COARSE_LOCATION -> {
                if (grantResults[0] == PackageManager.PERMISSION_GRANTED) {
                } else {
                    val builder = AlertDialog.Builder(this)
                    builder.setTitle("Functionality limited")
                    builder.setMessage("Since location access has not been granted, this app will not be able to discover beacons when in the background.")
                    builder.setPositiveButton(android.R.string.ok, null)
                    builder.setOnDismissListener { }
                    builder.show()
                }
                return
            }
        }
    }

}
```

**activity_main.xml**

```
<?xml version="1.0" encoding="utf-8"?>
<LinearLayout
        xmlns:android="http://schemas.android.com/apk/res/android"
        xmlns:tools="http://schemas.android.com/tools"
        android:layout_width="match_parent"
        android:layout_height="match_parent"
        tools:context=".MainActivity">

    <android.support.v7.widget.AppCompatButton
            android:layout_width="wrap_content"
            android:layout_height="wrap_content"
            android:text="Start"
            android:id="@+id/start"/>

    <android.support.v7.widget.AppCompatButton
            android:layout_width="wrap_content"
            android:layout_height="wrap_content"
            android:text="Stop"
            android:id="@+id/stop"/>
</LinearLayout>

```

davidgyoung, 2019-03-04:
Can you please verify with 2.16.1?  There was a similar problem reported that was fixed in #804: Fix problem on service shutdown that leaked threads and left scanning on

TayfunCesur, 2019-03-04:
Fixed it! Actually I didn't know the library has a new version. I trusted the Gradle to warn me for a latest versions but it didn't :( . Upgraded 2.16.1 and fixed!.

Thank you again David

closed 2019-03-04 by TayfunCesur

paulpv, 2019-03-22:
@TayfunCesur FYI, I have noticed that if you decorate your dependencies implementation lines with closures then gradle will not warn you that there is a new version available. If you comment out any decorating closure then gradle should warn you that there is a new version available. Seems like a weak sauce bug in Gradle. :/

TayfunCesur, 2019-03-23:
Waow. Didn't think that. Thank you so much @paulpv