AltBeacon/android-beacon-library #841: java.lang.IllegalStateException: Not allowed to start service Intent { cmp=***.***.***/org.altbeacon.beacon.service.BeaconService }

pawelwichniewicz, 2019-03-06:
### Expected behavior
Application should not crash on Android 9.0+ devices while tries to start ranging from background.

### Actual behavior
Application crash on Android 9.0+ devices while tries to start ranging from background.

### Steps to reproduce this behavior
Stacktrace:
`Caused by java.lang.IllegalStateException: Not allowed to start service Intent { cmp=***.***.***./org.altbeacon.beacon.service.BeaconService }: app is in background uid UidRecord{ba840a7 u0a211 CEM  idle change:idle|cached procs:1 proclist:17865, seq(0,0,0)}
       at android.app.ContextImpl.startServiceCommon(ContextImpl.java:1666)
       at android.app.ContextImpl.startService(ContextImpl.java:1611)
       at android.content.ContextWrapper.startService(ContextWrapper.java:677)
       at org.altbeacon.beacon.startup.RegionBootstrap$InternalBeaconConsumer.bindService(SourceFile:171)
       at org.altbeacon.beacon.BeaconManager.bind(SourceFile:437)
       at org.altbeacon.beacon.startup.RegionBootstrap.(SourceFile:77)
       at ***.***.***.bluetooth.beacon.manager.AppBeaconManager.start(SourceFile:150)
       at ***.***.***.Application.onCreate(SourceFile:60)`

Code which binds:
```
    public void start() {
        if (mRegionBootstrap != null) return;

        mBeaconManager.bind(this);
        mRegionBootstrap = new RegionBootstrap(this, REGIONS);
    }
```

Code which inits beacons manager (it is invoked **before** binding):
```
 mBeaconManager = BeaconManager.getInstanceForApplication(mAppContext);

        mBeaconManager.setForegroundBetweenScanPeriod(FOREGROUND_BETWEEN_SCAN_PERIOD);
        mBeaconManager.setForegroundScanPeriod(FOREGROUND_SCAN_PERIOD);
        mBeaconManager.setRegionStatePersistenceEnabled(false);
        Beacon.setHardwareEqualityEnforced(true);
        mBeaconTimeout = FOREGROUND_TIMEOUT;

        if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
            enableScanningInForeground();
        }

        mBeaconManager.getBeaconParsers().clear();
        mBeaconManager.getBeaconParsers()
                .add(new BeaconParser().setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25"));

        mBeaconManager.addRangeNotifier(this);
```

Code which enables foreground scanning on Android 8.0+ devices:
```
    private void enableScanningInForeground() {
        mBeaconManager.enableForegroundServiceScanning(
                NotificationHelper.Companion.createScanningForBeaconsNotification(mAppContext),
                NotificationHelper.SCANNING_FOR_PETS);
        mBeaconManager.setEnableScheduledScanJobs(false);
        mBeaconManager.setBackgroundBetweenScanPeriod(BACKGROUND_BETWEEN_SCAN_PERIOD);
    }
```

### Mobile device model and OS version
Android 9.0, 434 crashes for 119 users, 88% of devices are Samsung (Samsung Galaxy S8, S9, S9+, Note9), 5% Google (Pixel XL, Pixel 3XL). 100% of crashes where when application was in background.

### Android Beacon Library version
2.15.4 - 2.16.1

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2019-03-06:
@pawelwichniewicz, can you reproduce this yourself or have you only seen it in crash reports?

I wonder if this just happens if you have a foreground service whenever you reboot your phone.  

It may be that it is not allowed by Android 8/9 to start a foreground service if the app is launched in the background by beacon detection or other event.  Normally if the user has launched the app the foreground service gets started.  But if the user reboots the phone, the foreground service will not automatically restart.  In this case, perhaps a beacon detection triggers a background launch which tries and fails to start the foreground service per the crash above.

This is all just conjecture.  Any help testing to determine if this is i deed the root cause would be appreciated.

pawelwichniewicz, 2019-03-08:
@davidgyoung I've tried very hard to reproduce this issue, but without success. It's weird cause I have Galaxy Note 9 with Android Pie and couldn't reproduce this. Also I've tried on other phones with Android Pie and couldn't reproduce it either. It must be some special case - but from crashes - it happens all the time, but only on Android 9.

Actually, there was an issue (same crash) regarding launching foreground service from background after phone restart on Android 8 devices but version **2.15.2** fixed it and there were no crashes until it started to pop up on Android 9 devices.

pawelwichniewicz, 2019-03-26:
@davidgyoung any luck about this issue?

davidgyoung, 2019-03-26:
No work yet. I'd appreciate helping test my theory above to help speed thing along.  You can use the library reference app (uncommenting out the foreground service setup lines) to make it use a foreground service, then:

1. Start up the app
2. Reboot the phone
3. Turn on a beacon that should trigger the app 
4. Check if you get a crash

Then try:
1. Reboot the phone
2. Manually launch the app
3. Kill the app from the task switcher
4. Trigger a beacon that should launch the app
5. See if you get the crash

davidgyoung, 2019-05-29:
I tried all of the above steps on a Samsung Galaxy S9 / 9.0 but could not reproduce using the reference app.  

I suspect the problem simply has to do with the way the foreground service is attempted to be started when the phone is in some state that the OS does not like.  But I don't know what that is.

@pawelwichniewicz, are you 100% sure there is no way for the enableScanningInForeground() can fail to be called on Android 8+?  Or perhaps called only *after* the service is started through another path.  This would certainly cause the problem.

csgulley, 2019-08-07:
I am seeing this issue as well. It is the same call stack as the original poster and 100% of the cases are on Android 9 phones. I see the crash reports in Google's Play Console but have not been able to reproduce it myself.

alekskuzmin, 2019-09-16:
I'm experiencing the same problem and seems like there is a workaround for it (it is a known Android bug): https://stackoverflow.com/questions/52013545/android-9-0-not-allowed-to-start-service-app-is-in-background-after-onresume/55376015#55376015

davidgyoung, 2019-09-16:
@csgulley, thanks for the info.  If that is indeed the case described in this issue, I don't understand why nobody can reproduce it simply by booting a phone with a beacon visible.  I'd be glad to try applying the workaround, but I really need a way to reproduce otherwise we won't know if the change fixes anything or not.

turbo-lab, 2019-09-25:
Hi all.
I also have such issue on my Android vitals. It appears only on Samsung devices. I also can't reproduce it on my Xiaomi Redmi Note 5 with Androd 9.

I tried to play with this issue. I have no new reports for a week.

`diff --git a/lib/src/main/java/org/altbeacon/beacon/startup/RegionBootstrap.java b/lib/src/main/java/org/altbeacon/beacon/startup/RegionBootstrap.java
index 4931a08..63fa81f 100644
--- a/lib/src/main/java/org/altbeacon/beacon/startup/RegionBootstrap.java
+++ b/lib/src/main/java/org/altbeacon/beacon/startup/RegionBootstrap.java
@@ -216,7 +216,7 @@ public class RegionBootstrap {
         @Override
         public boolean bindService(Intent intent, ServiceConnection conn, int arg2) {
             this.serviceIntent = intent;
-            context.startService(intent);
+            context.startForegroundService(intent);
             return context.bindService(intent, conn, arg2);
 
         }
`

pawelwichniewicz, 2019-10-15:
@turbo-lab what is current status of crashes when you introduced your fix?

turbo-lab, 2019-10-16:
@pawelwichniewicz 
This method does not work at all. I'm so sorry.

csgulley, 2019-10-24:
Here is some additional information. We are using version 2.16.3 of the library. I have not been able to reproduce this personally but over 90 days we've recorded 2700 crashes from 160 users. 100% of the crashes have been on Android 9.

Here is the call stack and exception:

`Fatal Exception: java.lang.IllegalStateException: Not allowed to start service Intent { cmp=xxx/org.altbeacon.beacon.service.BeaconService }: app is in background uid UidRecord{4218edc u0a276 CEM  idle change:cached procs:1 proclist:29742, seq(0,0,0)}
       at android.app.ContextImpl.startServiceCommon + 1666(ContextImpl.java:1666)
       at android.app.ContextImpl.startService + 1611(ContextImpl.java:1611)
       at android.content.ContextWrapper.startService + 677(ContextWrapper.java:677)
       at org.altbeacon.beacon.startup.RegionBootstrap$InternalBeaconConsumer.bindService + 219(RegionBootstrap.java:219)
       at org.altbeacon.beacon.BeaconManager.bind + 443(BeaconManager.java:443)
       at org.altbeacon.beacon.startup.RegionBootstrap.<init> + 125(RegionBootstrap.java:125)
       [app frames deleted]
       at android.os.Handler.handleCallback + 873(Handler.java:873)
       at android.os.Handler.dispatchMessage + 99(Handler.java:99)
       at android.os.Looper.loop + 216(Looper.java:216)
       at android.app.ActivityThread.main + 7263(ActivityThread.java:7263)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run + 494(RuntimeInit.java:494)
       at com.android.internal.os.ZygoteInit.main + 975(ZygoteInit.java:975)`

Here is the calling code:
```
if (regionBootstrap == null) {
    manager = BeaconManager.getInstanceForApplication(application);
    final Notification notification = buildNotification(context);
    manager.enableForegroundServiceScanning(notification, VelApplication.NOTIFICATION_ID_BEACON);
    manager.setEnableScheduledScanJobs(false);
    manager.setBackgroundScanPeriod(BEACON_BACKGROUND_SCAN_PERIOD);
   
 manager.setBackgroundBetweenScanPeriod(BEACON_BACKGROUND_BETWEEN_SCAN_PERIOD);

    manager.getBeaconParsers().clear();
    manager.getBeaconParsers().add(new BeaconParser().setBeaconLayout("m:0-3=4c000215,i:4-19,i:20-21,i:22-23,p:24-24"));
    
    regionBootstrap = new RegionBootstrap(notifier, getRegions());
}
```

As a potential work around, I added code that would catch the IllegalStateException and then try to enable the beacon monitoring again 5 seconds later. I did this thinking maybe the condition that was causing the exception was transitory. However, the work around did not help. When we tried to enable the monitoring 5 seconds later IllegalStateException was thrown again.

Let me know if there is any more information I can provide.

davidgyoung, 2019-10-24:
I did a Google search on this and came across a similar report in a different library.  In that case, they found that the root cause was a different component registering 100 jobs, and then the exception being thrown by the library when it tried to schedule just one job, which pushed the system over the limit and caused the exception.

To check if this is happening to your app, add some debug code that prints out the number of currently scheduled jobs using this method:

https://developer.android.com/reference/android/app/job/JobScheduler.html#getAllPendingJobs()

You can even inspect the jobs to get details like the jobid, job class, etc.  You can use this to figure out which component stated all those jobs.

Even if you cannot reproduce the crash, if you see more than two jobs registered then you know that another component must also be registering jobs (this library only registers two).

If you can share info about a list of scheduled jobs > 2 I can help confirm how many originated from this library.  This will then tell us where to look next.

turbo-lab, 2019-11-26:
Hi, 
I placed such code to my app as you mentioned.

```
        try {
            regionBootstrap = new RegionBootstrap(this, regions);
        } catch (IllegalStateException e) {
            if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
                    JobScheduler jobScheduler =
                        (JobScheduler) context.getSystemService(Context.JOB_SCHEDULER_SERVICE);
                List<JobInfo> pendingJobs = jobScheduler.getAllPendingJobs();

                StringBuilder message = new StringBuilder("pendingJobsList");

                if (pendingJobs.isEmpty()) {
                    message.append(" is empty");
                } else {
                    message.append(" contains ").append(pendingJobs.size()).append(" entries:");
                    for (JobInfo job : pendingJobs) {
                        message.append("\n").append(job.toString());
                    }
                }

                throw new IllegalStateException(message.toString());
            } else {
                e.printStackTrace();
            }
        }
```

Unfortunately I didn't know that google cuts off all messages from crash reports since 2018. I got poor stacktrace :
```
java.lang.RuntimeException: 
  at android.app.ActivityThread.handleBindApplication (ActivityThread.java:6167)
  at android.app.ActivityThread.access$1200 (ActivityThread.java:235)
  at android.app.ActivityThread$H.handleMessage (ActivityThread.java:1768)
  at android.os.Handler.dispatchMessage (Handler.java:106)
  at android.os.Looper.loop (Looper.java:214)
  at android.app.ActivityThread.main (ActivityThread.java:6990)
  at java.lang.reflect.Method.invoke (Method.java)
  at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run (RuntimeInit.java:494)
  at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:1445)
Caused by: java.lang.IllegalStateException: < message has to be here >
  at org.turbo.beaconmqtt.beaconFactory.BeaconFactory.startBleMonitoring (BeaconFactory.java:533)
  at org.turbo.beaconmqtt.beaconFactory.BeaconFactory.<init> (BeaconFactory.java:61)
  at org.turbo.beaconmqtt.BeaconApplication.onCreate (BeaconApplication.java:32)
  at android.app.Instrumentation.callApplicationOnCreate (Instrumentation.java:1154)
  at android.app.ActivityThread.handleBindApplication (ActivityThread.java:6162)
  at android.app.ActivityThread.access$1200 (ActivityThread.java:235)
  at android.app.ActivityThread$H.handleMessage (ActivityThread.java:1768)
  at android.os.Handler.dispatchMessage (Handler.java:106)
  at android.os.Looper.loop (Looper.java:214)
  at android.app.ActivityThread.main (ActivityThread.java:6990)
  at java.lang.reflect.Method.invoke (Method.java)
  at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run (RuntimeInit.java:494)
  at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:1445)
```

It looks like I have to have custom crash collector. Make sense? Any ideas?

davidgyoung, 2019-11-27:
Ugh, that is unfortunate.  I did not realize Google started stripping the message.   You might also simply catch the exception without rethrowing it and send the message string to a server you control.  It could be as simple as making it a URL parameter on a simple Get request to a web server, then use a

turbo-lab, 2019-12-02:
I have faced with the same problem on paho-mqtt library.

```
java.lang.RuntimeException: 
  at android.app.ActivityThread.handleBindApplication (ActivityThread.java:6167)
  at android.app.ActivityThread.access$1200 (ActivityThread.java:235)
  at android.app.ActivityThread$H.handleMessage (ActivityThread.java:1768)
  at android.os.Handler.dispatchMessage (Handler.java:106)
  at android.os.Looper.loop (Looper.java:214)
  at android.app.ActivityThread.main (ActivityThread.java:6990)
  at java.lang.reflect.Method.invoke (Method.java)
  at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run (RuntimeInit.java:494)
  at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:1445)
Caused by: java.lang.IllegalStateException: 
  at android.app.ContextImpl.startServiceCommon (ContextImpl.java:1666)
  at android.app.ContextImpl.startService (ContextImpl.java:1611)
  at android.content.ContextWrapper.startService (ContextWrapper.java:677)
  at org.eclipse.paho.android.service.MqttAndroidClient.a (MqttAndroidClient.java:27)
  at org.eclipse.paho.android.service.MqttAndroidClient.a (MqttAndroidClient.java:1)
  at org.turbo.beaconmqtt.broadcaster.MqttBroadcaster.a (MqttBroadcaster.java:80)
  at org.turbo.beaconmqtt.broadcaster.MqttBroadcaster.<init> (MqttBroadcaster.java:56)
  at org.turbo.beaconmqtt.broadcaster.Broadcaster.<init> (Broadcaster.java:21)
  at org.turbo.beaconmqtt.BeaconApplication.onCreate (BeaconApplication.java:46)
  at android.app.Instrumentation.callApplicationOnCreate (Instrumentation.java:1154)
  at android.app.ActivityThread.handleBindApplication (ActivityThread.java:6162)
  at android.app.ActivityThread.access$1200 (ActivityThread.java:235)
  at android.app.ActivityThread$H.handleMessage (ActivityThread.java:1768)
  at android.os.Handler.dispatchMessage (Handler.java:106)
  at android.os.Looper.loop (Looper.java:214)
  at android.app.ActivityThread.main (ActivityThread.java:6990)
  at java.lang.reflect.Method.invoke (Method.java)
  at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run (RuntimeInit.java:494)
  at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:1445)
```

So, I found corresponding issue. Please have a look: https://github.com/eclipse/paho.mqtt.android/pull/358/commits

turbo-lab, 2019-12-26:
Hello! It's me again :) Today I got the first report for that issue.

```

DATE | Wed Dec 25 19:44:37 GMT+09:00 2019

CUSTOM DATA | pendingJobsList is empty
APP VERSION CODE | 0.3 build 11
ANDROID VERSION | 9
PACKAGE NAME | org.turbo.beaconmqtt
APP VERSION NAME | Beacon MQTT
PHONE MODEL | SM-N976N
STACK TRACE | java.lang.IllegalStateException: Not allowed to start service Intent { cmp=org.turbo.beaconmqtt/org.altbeacon.beacon.service.BeaconService }: app is in background uid UidRecord{ea7df43 u0a293 CEM  idle change:idle\|cached procs:1 proclist:26482, seq(0,0,0)} 	at android.app.ContextImpl.startServiceCommon(ContextImpl.java:1666) 	at android.app.ContextImpl.startService(ContextImpl.java:1611) 	at android.content.ContextWrapper.startService(ContextWrapper.java:677) 	at org.altbeacon.beacon.startup.b$b.bindService(Unknown Source:8) 	at org.altbeacon.beacon.f.a(Unknown Source:139) 	at org.altbeacon.beacon.startup.b.(Unknown Source:39) 	at org.turbo.beaconmqtt.c.a.r(Unknown Source:440) 	at org.turbo.beaconmqtt.c.a.(Unknown Source:61) 	at org.turbo.beaconmqtt.BeaconApplication.onCreate(Unknown Source:32) 	at android.app.Instrumentation.callApplicationOnCreate(Instrumentation.java:1158) 	at android.app.ActivityThread.handleBindApplication(ActivityThread.java:6396) 	at android.app.ActivityThread.access$1300(ActivityThread.java:248) 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1831) 	at android.os.Handler.dispatchMessage(Handler.java:106) 	at android.os.Looper.loop(Looper.java:216) 	at android.app.ActivityThread.main(ActivityThread.java:7263) 	at java.lang.reflect.Method.invoke(Native Method) 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:494) 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:975)
```

my code is:
```
        try {
            regionBootstrap = new RegionBootstrap(this, regions);
        } catch (IllegalStateException e) {
            if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
                    JobScheduler jobScheduler =
                        (JobScheduler) context.getSystemService(Context.JOB_SCHEDULER_SERVICE);
                List<JobInfo> pendingJobs = jobScheduler.getAllPendingJobs();

                StringBuilder message = new StringBuilder("pendingJobsList");

                if (pendingJobs.isEmpty()) {
                    message.append(" is empty");
                } else {
                    message.append(" contains ").append(pendingJobs.size()).append(" entries:");
                    for (JobInfo job : pendingJobs) {
                        message.append("\n").append(job.toString());
                    }
                }

                StringWriter stackTraceWriter = new StringWriter();
                e.printStackTrace(new PrintWriter(stackTraceWriter));
                new CrashReporter(context).execute(stackTraceWriter.toString(), message.toString());
            } else {
                e.printStackTrace();
            }
        }
```

davidgyoung, 2019-12-26:
Interesting, @turbo-lab.  So this confirms that your app (and the library) has not used up all the available jobs -- indeed it has not used up *any*.   So this disproves my theory from the comment on Oct. 24 above about a possible root cause.  

Unfortunately this puts us back to square one: I now have no working theories about what causes this crash.

Any thoughts welcome.

bpede-at-nalys, 2020-09-28:
Hello @davidgyoung. 

We're having this issue as well, and it occurs quite often in production (500 times for 100 different users in the last 90 days according to Crashlytics: [screenshot](https://nalys-my.sharepoint.com/personal/bpede_nalys-group_com/Documents/RegionBootstrap-IllegalStateException.png)). As you can see on the screenshot it only happened on Samsung Galaxy devices (A40, A6, A6+, S8, S9, S10+, S20+).

We bought a Samsung Galaxy A40, hoping I would find a reproduce scenario. So far I've only been able to reproduce it once, on the first day of testing. No idea if any of this is relevant but it happened when the app was in background. I pressed `|||` button and killed the app from the task switcher. The device battery was 78% and "Optimize battery usage" was still default (on). Since then I haven't been able to reproduce it anymore. I tried rebooting, killing the app from the task switcher, force stopping... all at different battery percentages.

I have the application logs when the crash happened. I compared it with a normal run on the foreground and don't see any differences before the crash.

Crash logs:

```
$ pidcat com.*.*.demoapp

                            Process com.*.*.demoapp created for mllaunch {com.*.*.demoapp/com.*.*.demoapp.SplashActivity}
                            PID: 29711   UID:    GIDs:

                 Zygote  E  isWhitelistProcess - Process is Whitelisted
                         E  accessInfo : 1
        com.*.*.demoapp  I  Late-enabling -Xcheck:jni
                         E  Unknown bits set in runtime_flags: 0x8000
         ActivityThread  D  setConscryptValidator
                         D  setConscryptValidator - put
         DynamiteModule  I  Considering local module com.google.android.gms.measurement.dynamite:15 and remote module com.google.android.gms.measurement.dynamite:0
                         I  Selected local version of com.google.android.gms.measurement.dynamite
        com.*.*.demoapp  W  Unsupported class loader
                         I  Waiting for a blocking GC ClassLinker
                         I  WaitForGcToComplete blocked ClassLinker on ClassLinker for 115.513ms
    FirebaseCrashlytics  I  Initializing Crashlytics 17.1.1
   FirebaseInitProvider  I  FirebaseApp initialization successful
CrashlyticsInitProvider  I  CrashlyticsInitProvider skipping initialization
        com.*.*.demoapp  W  Verification of void com.google.android.gms.measurement.internal.zzaq.<clinit>() took 299.885ms
                  MyApp  I  Application initialization
          BeaconManager  I  BeaconManager started up on pid 29711 named 'com.*.*.demoapp' for application package 'com.*.*.demoapp'.  isMainProcess=true
           BeaconParser  D  Parsing beacon layout: m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25
                         D  Parsing beacon layout: m:0-3=4c000215,i:4-19,i:20-21,i:22-23,p:24-24
                     FA  V  App measurement collection enabled
                         V  App measurement enabled for app package, google app id: com.*.*.demoapp, 1:84097332596:android:d80beeb48308f1356b6821
                         I  App measurement initialized, version: 31000
                         I  To enable debug logging run: adb shell setprop log.tag.FA VERBOSE
                         I  To enable faster debug mode event logging run:
                         I    adb shell setprop debug.firebase.analytics.app com.*.*.demoapp
                         D  Debug-level message logging enabled
          BeaconManager  W  Disabling ScanJobs on Android 8+ may disable delivery of beacon callbacks in the background unless a foreground service is active.
                ScanJob  I  Using immediateScanJobId from manifest: 208352939
                         I  Using periodicScanJobId from manifest: 208352940
          BeaconManager  W  Disabling ScanJobs on Android 8+ may disable delivery of beacon callbacks in the background unless a foreground service is active.
                ScanJob  I  Using immediateScanJobId from manifest: 208352939
                         I  Using periodicScanJobId from manifest: 208352940
          BeaconManager  D  Not synchronizing settings to service, as it has not started up yet
                         W  Setting a short backgroundBetweenScanPeriod has no effect on Android 8+, which is limited to scanning every ~15 minutes
                         D  updating background flag to true
                         D  updating scan period to 1500, 0
                         W  The BeaconManager is not bound to the service.  Call beaconManager.bind(BeaconConsumer consumer) and wait for a callback to onBeaconServiceConnect()
                         D  This consumer is not bound.  Binding now: org.altbeacon.beacon.startup.RegionBootstrap$InternalBeaconConsumer@163b493
                         D  Binding to service
                         I  Starting foreground beacon scanning service.
         AndroidRuntime  D  Shutting down VM
                         E  FATAL EXCEPTION: main
                         E  Process: com.*.*.demoapp, PID: 29711
                         E  java.lang.RuntimeException: Unable to create application com.*.*.demoapp.MyDebugDemoApplication: java.lang.IllegalStateException: Not allowed to start service Intent { cmp=com.*.*.demoapp/org.altbeacon.beacon.service.BeaconService }: app is in background uid UidRecord{50282b5 u0a229 CEM  idle change:idle|cached procs:1 seq(0,0,0)}
[... cut ...]
            backgroundBetweenScanPeriod = 0
            backgroundScanPeriod = 1500
            addRangeNotifier(this@MyBeaconScanner)
        }

        backgroundPowerSaver = BackgroundPowerSaver(applicationContext)
        startMonitoringRegion(REGION)
    }

    private fun startMonitoringRegion(regionToMonitor: Region) {
        regionBootstrap = RegionBootstrap(this, regionToMonitor)
        Timber.d("Monitoring region ${str(regionToMonitor)}")
    }
}
```

I want to help, so if you think of anything I can try just let me know.

davidgyoung, 2020-09-28:
@bpede-at-nalys, can you please share a non-password protected copy of your crashlytics screenshot  (or paste the erect content?) I am interested in seeing the Android versions of the Samsung devices that have seen this crash.

bpede-at-nalys, 2020-09-28:
Oops sorry, I corrected the link in my original post: https://nalys-my.sharepoint.com/:i:/g/personal/bpede_nalys-group_com/EdI9pc-qfpZAgHgMC0jA5pABkVbwR1de0AhQU1LC_wv3mg?e=hboNMX

Devices: 100% Samsung
- 79% Galaxy A40
- 17% Galaxy A6
- 1% Galaxy S20+
- 1% Galaxy S8
- 1% Galaxy S10+
- <1% Galaxy S9
- <1% Galaxy A6+

Operating systems:
- 97% Android 10
- 3% Android 9

davidgyoung, 2020-09-28:
Since this is happening on Android 10, this is clearly not a bug in Android 9.  This may be a Samsung-specific fork of Android that disallows starting a foreground service from the background (in some states at least) after an app has been killed.  If this conjecture is true, then there may be no automatic solution.   We could catch the exception, but the foreground service would not start.

A few possible (clunky) solutions:

1. Catch the exception, and send a notification to the user saying “Please tap here to allow XXX app to run in the background”.  If the notification is tapped, the app would be brought to the foreground allowing the foreground service to be started.

2. Fall back to scheduled job scanning every ~15 minutes if the foreground service cannot be started.  If the user ever brings the app to the foreground, restart foreground service scanning.

3. A combination of both 1 and 2