AltBeacon/android-beacon-library #843: can't get filtered scan to work, meaning android 8.1 and above will stop scanning

sboo, 2019-03-13:
I've tried to follow the instructions to keep the scan working after device screen is off. With the below code I'd expect to get a filtered scan which should be the way to go in Android 8.1 and above.

### Expected behavior
Log should show:
`D/CycledLeScannerForLollipop: starting filtered scan in SCAN_MODE_LOW_POWER`

### Actual behavior
Log showing:
`D/CycledLeScannerForLollipop: starting non-filtered scan in SCAN_MODE_LOW_LATENCY`

### Steps to reproduce this behavior
```
fun startBeaconService() {
        BeaconManager.setDebug(true)
        beaconManager = BeaconManager.getInstanceForApplication(this)
        beaconManager?.backgroundMode = false
        beaconManager?.foregroundScanPeriod = 10000L
        beaconManager?.foregroundBetweenScanPeriod = 50000L
        beaconManager?.beaconParsers?.add(BeaconParser().setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24"))
        val notificationManager = MyNotificationManager.getInstance()
        val notification = notificationManager.buildBeaconServiceNotification(this, "iBeacon test service", null)
        beaconManager?.enableForegroundServiceScanning(notification, 99)
        beaconManager?.setEnableScheduledScanJobs(false)
        beaconManager?.backgroundBetweenScanPeriod = 0
        beaconManager?.backgroundScanPeriod = 1100
        beaconManager?.bind(this)
}

override fun onBeaconServiceConnect() {
        beaconManager?.addRangeNotifier { beacons, _ ->
            Log.d("testbeacon", "RangeNotifier beacons = $beacons")
            //code to submit ranged beacons to API
        }
        beaconManager?.startRangingBeaconsInRegion(
            Region("com.beacon.test", Identifier.parse("f7826da6-4fa2-4e98-8024-bc5b71e0893e"), null, null)
        )
    }
```

### Mobile device model and OS version
Samsung Galaxy S9+, v9
Motorola android one, v8.1

### Android Beacon Library version
2.15.2

davidgyoung, 2019-03-13:
The library only uses filtered low-power scans when in the background mode and in a between scan period. 

Several issues here:

1.  The code shown turns background mode off, effectively disabling using low power scans.  You must set beaconManager?.backgroundMode = true or use BackgroundPowerSaver to toggle background mode automatically when no activities are visible.

2. The code shown sets the background between scan period to 0, also disabling this feature.  You must have a non-zero (and significantly large) background between scan period for you to do these low power scans.

If you need further help, please open a support question on StackOverflow.com as this forum is reserved for feature request and reproducible bug reports with the library itself.

closed 2019-03-13 by davidgyoung

sboo, 2019-03-14:
Thanks David, Thank you for getting me the right information. I figured I needed to make it a foreground service to keep it running, so I followed these instructions. Will make a separate service now and let it do background scanning instead.