AltBeacon/android-beacon-library #844: Scanning in separate process not working

bio007, 2019-03-15:
Using the library (**ranging notifier**) inside android service, running in a separate process (`android:process` tag in manifest) yields no results nor throwing any exception or error. Removing the `android:process` tag from the services makes the same code work.

Note: I'm not trying to run the service in one process and get the result in main process. Everything regarding AltBeacon is solely in the service (and service's process).

### Expected behavior - I can get results for nearby beacons or get an error if the configuration is wrong

### Actual behavior - getting no readings at all nor error that something is awry

### Steps to reproduce this behavior - create a service running in a separate process and implement beacon ranging (for example similar to Ranging Example Code here https://altbeacon.github.io/android-beacon-library/samples.html)

### Mobile device model and OS version Samsung Galagy S8 running Oreo

### Android Beacon Library version 2.16.1

davidgyoung, 2019-03-15:
The separate process functionality is not widely used.  It was last confirmed to be working in version

Two tips:

1. By default on Android 8+, scan jobs are used which do not support this separate p

bio007, 2019-03-25:
Hi, thanks for additional info. Actually I've made some progress.

It more or less works with `beaconManager.setEnableScheduledScanJobs(false)` and with 
```
        <receiver android:name="org.altbeacon.beacon.startup.StartupBroadcastReceiver"
                  android:process="process.name"
                  tools:replace="process" />
        <service
            android:name="org.altbeacon.beacon.service.BeaconService"
            android:process="process.name"
            tools:replace="process" />
        <service
            android:name="org.altbeacon.beacon.BeaconIntentProcessor"
            android:process="process.name"
            tools:replace="process" />
```
overridden in manifest.

Only funny thing I see is that even when the app and service is in foreground (running in foreground mode) I'm receiving scans every 5 minutes which is what I set for background scans ` beaconManager.setBackgroundBetweenScanPeriod(300000L)`. So it looks to me that it is somehow "operating in backround mode" all the time.

davidgyoung, 2019-03-25:
@bio007 glad you have it working.  Which library version are you seeing this working with?  Have you tried it with the latest version to see if it still works?

As far as the background mode goes, are you setting this manually with beaconManager.setBackgroundMode(...)?  Or are you using BackgroundPowerSaver?  You must use one or the other technique for this to work.

If you are using one of the above, and still see it not working, this may be a bug.  The library tracks visible activities to determine background mode, and when the scanner is in a separate process the number of open activities in that process is zero, hence it is always in background mode.  The library is supposed to communicate the foreground status between processes.  But this is not a widely used feature and not super well tested.

bio007, 2019-03-25:
I'm using 2.16.1 version.

Periods I set manually. To be precise I'm calling this:

```
beaconManager.setForegroundScanPeriod(FOREGROUND_SCAN_PERIOD);    //1100L
beaconManager.setForegroundBetweenScanPeriod(FOREGROUND_BETWEEN_SCAN_PERIOD);    //5000 L
beaconManager.setBackgroundScanPeriod(BACKGROUND_SCAN_PERIOD);    //8000 L
beaconManager.setBackgroundBetweenScanPeriod(BACKGROUND_BETWEEN_SCAN_PERIOD);    //300000 L
```

But I'm creating also BackgroundPowerSaver and RegionBootstrap:
````
regionBootstrap = new RegionBootstrap(this, REGION);
backgroundPowerSaver = new BackgroundPowerSaver(service.getApplicationContext());
```

bio007, 2019-03-25:
Trying to comment out the BackgroundPowerSaver and RegionBootstrap and it looks like RegionBootstrap is making the difference not BackgroundPowerSaver 🤔

davidgyoung, 2019-03-25:
Where do you create these two objects?  Is it in a custom application class onCreate method?  If so, do you have a guard condition to make sure you only create them in the main process instance (not the scanning process instance).  I believe there is an example for this guard condition in #479

bio007, 2019-03-26:
I don't have it in application class and I have it in scanning process indeed. The problem is that moving it to application (or main process in general) will create a second instance of `beaconManager`. Is the other process going to respect the background settings set to main process `beaconManager` instance? 

Also `RegionBootstrap` needs `BootstrapNotifier` so I need to forward the `didEnterRegion` / `didExitRegion` notifications to the service (and it's `beaconManager` instance). Is that what you had in mind?

davidgyoung, 2019-03-26:
When you have a service running in a separate process things get strange.  Because there are two processes, that means there are two copies of every singleton object.  That means two BeaconManager instances and two Application instances.  I had to build the library so that is OK.  The library detects if the scanning process is separate, and then knows whether the BeaconManager is the one in the main process or the scanning process.  Settings are marshalled between the the BeaconManager instance on the main. process

closed 2019-03-26 by davidgyoung