AltBeacon/android-beacon-library #845: Starting and Stopping Foreground Service Issues

stephenruda, 2019-03-18:
### Issue
Hi David, me again!  There are a couple of issues I noticed when implementing the Foreground Service.  This might be just be expected behavior but I couldn't find it documented anywhere.  It might be an issue too.

In my app, users can select specific Beacons and turn them OFF and ON for monitoring.  In my case, I don't run the regionBootstrap or start a foreground service until the first Beacon has been turned ON.  If all Beacons have been turned OFF I must disable the regionBootstrap and stop the foreground service.  Here are the issues I encountered:
1. Calling `beaconManager.updateScanPeriods();` before starting the foreground service will cause duplicate entry/exit events to be fired when monitoring a Beacon.  [I created a sample app that reproduces the problem on this branch.](https://github.com/stephenruda/android-beacon-library-reference/tree/duplicate-events)  You can just uncomment / comment the code in the BeaconReferenceApplication class to reproduce.
2. Calling `beaconManager.updateScanPeriods();` would fail unless I called it a little after constructing the `regionBootstrap`.  This makes sense because the `regionBootstrap` takes a little bit to bind and I can't call `updateScanPeriods()` unless bound.
3. I ran into some strange behavior when the last Beacon has been disabled and I no longer want to monitor any Beacons and I want to stop the foreground service.
I can't call `beaconManager.disableForegroundServiceScanning();` while the BeaconConsumer is still bound so I use `regionBootstrap.disable()` to unbind.
There is a `regionBootstrap.disable()` but no `regionBootstrap.enable()` method.  I have to construct an all new `regionBootstrap` object in order to start scanning again.
After the user has flipped a Beacon on again, constructing a new `regionBootstrap` object will cause me to get multiple events.  I have to call `beaconManager.removeAllMonitorNotifiers();` right after `regionBootstrap.disable()` to avoid duplicates.

In the end, here is what my code looks like to get it working.
```
public void lastBeaconDisabled() {
    // The last Beacon has been disabled so we want to unbind and stop the foreground service
    regionBootstrap.disable();
    beaconManager.removeAllMonitorNotifiers();
    beaconManager.disableForegroundServiceScanning();
}

public void enableFirstBeacon() {
    // The first Beacon has been enabled so we want to construct a regionBootstrap and start a foreground service
    beaconManager.enableForegroundServiceScanning();
    regionBootstrap = new RegionBootstrap(this, regions);
    while(!beaconManager.isAnyConsumerBound()) {
        // Wait until it is bound before updating scan periods
        beaconManager.updateScanPeriods();
    }
}
```
Some of this is likely intended behavior.  I don't know if any of it would be considered an issue.  I figured it is helpful information for other users that need to be able to scan for Beacons with a foreground service that the user can enable / disable at any time.

### Mobile device model and OS version
Samsung Galaxy S7 on Android 8.0.0

### Android Beacon Library version
2.16.1

davidgyoung, 2019-03-18:
Thanks for the report, @stephenruda.   A few thoughts:

1. You don't need to call `beaconManager.updateScanPeriods()` if you set up the scan periods you like before binding or using RegionBootstrap.  This is the recommended use.   It does sound like there is a bug with a dynamic call to `beaconManager.updateScanPeriods()` that causes a duplicate monitoring callback.  I suspect this is probably a general issue that had nothing to do with a foreground service.  If so, this should be an issue by itself.

2. Yes, it is expected that `beaconManager.updateScanPeriods()` will fail until binding is complete.  The API should throw a RemoteException in this case, so as long as you get that I'd say it works as designed.

3. It looks like RegionBootstrap disable() should internally call beaconManager.removeMonitorNotifier() on its notifier.  This should be another issue, too.  And I see no reason not to add an enable() method to go along with the disable() method so you don't have to construct a new one.  Might as well tack this change on, too.

In addition to the more-specific issues, I'd certainly welcome pull requests on items (1) and (2) if you'd like to work on the changes.

stephenruda, 2019-03-19:
@davidgyoung I would certainly be interested in working on a PR.  I will do that when I get some free time.  I have actually noticed that it is not solely related to `updateScanPeriods()`.  There are a lot of BeaconManager methods that will cause me to get duplicate events if I call them before starting the foreground service.  [Please see this class in my reproduction application.](https://github.com/stephenruda/android-beacon-library-reference/blob/duplicate-events/app/src/main/java/org/altbeacon/beaconreference/BeaconReferenceApplication.java)  Specifically, the `callPotentiallyBrokenMethods()` method will call a method on the BeaconManager that will cause a duplicate entry event to be fired.

I can call several methods before starting the foreground service and everything is fine...
`. setDebug(true);
`.setAndroidLScanningDisabled(true);`
`.getBeaconParsers().add()`
`. setBackgroundBetweenScanPeriod()`
`. setBackgroundScanPeriod()`
`. setForegroundBetweenScanPeriod()`
`. setForegroundScanPeriod()`
`. setRegionExitPeriod()`

But there are several methods that I can't call before starting the foreground service as it will cause me to get duplicate events when I do start the foreground service.
`updateScanPeriods()`
`setBackgroundMode()`
`startRangingBeaconsInRegion()`

Any thoughts on why calling some of these methods might be causing me to get duplicate events?  I have even tried calling `removeAllMonitorNotifiers()` before starting the foreground service but I still get two events.

My app only cares about monitoring Beacons in the background if the user flips the switch to monitor it.  This would create the foreground service.  However, the user has the option to range for Beacons while in the app and this can happen before ever monitoring the Beacon.  So they can potentially call `startRangingBeaconsInRegion()` and then a few minutes later they start monitoring the region (and then the foreground service is created and the regionBootstrap is constructed) and they get duplicate callbacks.

Thanks!

davidgyoung, 2019-03-20:
What those methods have in common is that they ultimately call `beaconManager.applyChangesToServices()` which eitherpersists changes to disk for the next ScheduledJob or sends an Intent to the running service to tell it to refresh this info.

I cannot think of why this would cause duplicates.
l

Sorry, you are a bit on the bleeding edge trying to use the library in this manner.  it certainly should support that usage, but it is an unusual usage so you are flushing out some bugs.

stephenruda, 2019-03-22:
@davidgyoung I have had some time to experiment with this today and here is what I have found...

I took a look at `applyChangesToServices()`.  I noticed that within this method you are calling `ScanJobScheduler.getInstance().applySettingsToScheduledJob(mContext, this);` if the library has `mScheduledScanJobsEnabled` set to true.

I started experimenting and realized that if I call `setEnableScheduledScanJobs(false);` before calling one of these 'problematic' methods that internally call `applyChangesToServices()` then it works as expected.

Looking at your `setEnableScheduledScanJobs(false)` method, you aren't cancelling the current scan jobs right when the user calls this method.  You are just setting the boolean for later use.

I changed this method to this...
```
   public void setEnableScheduledScanJobs(boolean enabled) {
        if (isAnyConsumerBound()) {
            LogManager.e(TAG, "ScanJob may not be configured because a consumer is" +
                    " already bound.");
            throw new IllegalStateException("Method must be called before calling bind()");
        }
        if (enabled && android.os.Build.VERSION.SDK_INT < Build.VERSION_CODES.LOLLIPOP) {
            LogManager.e(TAG, "ScanJob may not be configured because JobScheduler is not" +
                    " availble prior to Android 5.0");
            return;
        }
        if (!enabled && android.os.Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
            LogManager.w(TAG, "Disabling ScanJobs on Android 8+ may disable delivery of "+
                    "beacon callbacks in the background unless a foreground service is active.");
        }

        if(!enabled) {
            ScanJobScheduler.getInstance().cancelSchedule(mContext);
        }
        
        mScheduledScanJobsEnabled = enabled;
    }
```

and everything is working expected.  Now it doesn't matter if I call `setEnableScheduledScanJobs()` before or after calling one of these 'problematic' methods.

If you don't see any issues with me adding this simple check, I would be happy to add a pull request.  Is there a problem with calling `ScanJobScheduler.getInstance().cancelSchedule(mContext);` the moment the user calls `setEnableScheduledScanJobs(false)`?

Thanks!

davidgyoung, 2019-03-22:
Good catch.  That makes total sense!  Yes, please make a PR for this.

stephenruda, 2019-03-22:
@davidgyoung Okay, I have created the PR.  Just let me know if you need anything else.  Feel free to close this issue whenever the PR gets merged.  Thanks for the help!

davidgyoung, 2019-04-04:
Fixed in #847

closed 2019-04-04 by davidgyoung