AltBeacon/android-beacon-library #846: No method to set some scan settings

joew46167, 2019-03-21:
### Expected behavior

### Actual behavior

### Steps to reproduce this behavior

### Mobile device model and OS version

### Android Beacon Library version

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

joew46167, 2019-03-21:
I have cloned the library, checked out master, made a new branch, and then updated code to do the following:

BeaconMaster: Added variables and setter and getter methods for each of the following:

- Background Scan Mode
- Foreground Scan Mode
- Background Match Mode
- Background Match Mode
- Foreground Callback Type
- Background Callback 
- Foreground Number of Matches
- Background Number of Matches

ScanHelper:  If the above settings are set, use them when creating the scan settings to start a scan

CycledLeScannerForLollipop:   If the above settings are set, use them when creating the scan settings to start a scan

How can I submit these for inclusion in the library?

davidgyoung, 2019-03-22:
@joew46167, submitting a GitHub pull request on this project is the usual way to do this.  This is potentially a big change, so here are a few tips to make it possible to get the changes merged:

* The Beacon manager has already become cluttered with lots of scan settings, and I am reluctant to add even more accessor methods.  I suggest combining these in a ScanSettings class (perhaps with a ScanSettings.Builder) and setting them all at once on the BeaconManager.

* Look at how other settings are currently propagated (e.g
 backgroundScanPeriod) to the BeaconService and ScanJob.  You will need to make the new code follow the existing mechanisms.

* You will need to test this on actual devices with Android 8+ with scheduled scan jobs both enabled and disabled.

* Unit tests for showing all the settings get persisted when set are critical.