AltBeacon/android-beacon-library #848: [Feature Request] Allow 128-bit service UUID for scan filters

mannodermaus, 2019-03-28:
### Android Beacon Library version
2.16.1

---

Currently, [`ScanFilterUtils#createScanFiltersForBeaconParsers()`](https://github.com/AltBeacon/android-beacon-library/blob/b9876b45acd6c72c84b8d0325062259020eb89ba/lib/src/main/java/org/altbeacon/beacon/service/scanner/ScanFilterUtils.java#L79-L90) expects a `BeaconParser`'s service UUID to be 16-bit - 128-bit UUID aren't supported. Our custom parser is being neglected on Samsung 8.1+ devices because of the non-empty scan filters applied on these devices (working around #769, also ref. #691). There's also the option to include manufacturer data in the scan filter (the `else` branch of the linked code above), however I wanted to explore the service UUID route first.

Is there a specific reason to only allow 16-bit UUIDs for scan filters, or would it be possible to have custom `BeaconParser` implementations expose a 128-bit service UUID as well? I realize this is a niche use case, and we might be able to work with the manufacturer data as well - however, UUID would be preferable in our case.

davidgyoung, 2019-03-28:
A few points:

The scan filter is only supposed to be used on Samsung devices if the library is in background mode.  There is no reason to modify the library for foreground usage.  You can just call beaconManager.setBackgroundMode(false); or use BackgroundPowerSaver to have the library toggle foreground/background mode automatically based on visible activities.

The library was designed for common beacons including AltBeacon/iBeacon (using manufacturer data) and Eddystone (using 16 bit service UUIDs).  Beacons using 128 bit service UUIDs are very uncommon because few bytes are left in the advertisement for other identifiers.  So filter support has for 128 bit UUIDs has just never been added.  This is the first such usage I have heard about.

I am open to adding this in a pull request if you would like to put one together.

mannodermaus, 2019-03-29:
Thanks for your response. We actually do use `beaconManager.setBackgroundMode(false)` already at configuration time, to express our intent to only ever use it while in foreground. `CycledLeScannerForLollipop#startScan()` will be called regardless of the mode and hit this logic gate, no?

davidgyoung, 2019-03-29:
@mannodermaus, right, I hadn't really thought about that.  The scan filters are used on Samsung even in foreground mode (just in case the screen is off) and the app developer has configured foreground mode all the time.  So this does make it totally unworkable for your use case on Samsung devices.

An alternative may be to detect if the screen is off and only use the scan filter in that case, or to make it possible to disable these scan filters.

mannodermaus, 2019-04-02:
> An alternative may be to detect if the screen is off and only use the scan filter in that case, or to make it possible to disable these scan filters.

Is this something that the client side may control, or do we need some extensions to AltBeacon itself first?

davidgyoung, 2019-04-04:
All of these possible solutions we've discussed would require changes to the library.  For screen processing, the library could simply check before starting the scan if the screen is on using:

```
PowerManager powerManager = (PowerManager) getSystemService(POWER_SERVICE);
boolean isScreenOn = powerManager.isInteractive();
```
Then not use the filter on Samsung if the screen is on.  This would at least make it work when the screen is on.

This isn't a super high priority for this project given that you have the only project I've heard of that has run into this problem, but If you'd like to work on the change to the library and test it, I'd welcome a pull request with the change.  I'd gladly help you with any questions along the way.

mannodermaus, 2019-04-04:
Thanks, David! I'll see what I can do this weekend.

davidgyoung, 2019-04-06:
@mannodermaus, since it does appear possible to add this feature for 8+ devices, if you would like to work on a pull request to add this feature, I can guide you on the changes that would be needed.  Please let me know.

mannodermaus, 2019-04-06:
I'm slightly torn about which route to take here. If we simply added a conditional to check the `PowerManager` and only apply the filter when the screen is off, that might throw off other custom configurations which want to change this behavior for their use case. Instead, we could make this a setting to be configured all the way up on `BeaconManager`. The current behavior (i.e. "apply it always") would remain the default, but you can change it using an enum value. Is it really that front-facing to bubble up as a public setting, though? I wouldn't want to convolute the API surface unnecessarily.

davidgyoung, 2019-04-07:
Adding configurations is actually harder than it might seem, because the configuration setting must be propagated from the BeaconManager to the scanner that my be in a separate process and also may be running as part of a JobScheduler ScanJob or a BeaconService.  There is a serialization process that goes with it.  I am also reluctant to keep throwing more configuration settings into the already cluttered BeaconManager.  I really want to refactor this to have a standalone class for all the scan settings.  See #846 for a discussion about that.

Right now, using a non-empty scan filter on Samsung during the active scan period is already a Samsung-specific kludge to get around a  Samsung-specific block of scans with the screen off without a scan filter.  So I don't think there is much wrong with targeting this kludge more surgically by making sure the screen is off before applying it.

That said, it does deserve testing with standard beacon types.  One thing I want to make sure is that this does not break things if the scan is started without the filter when the screen is off, and then the screen goes of while the scan is still active.  Will such a scenario block further scan results on Samsung?  Or does Samsung only block the scan results if the scan was *initiated* without a non-empty scan filter with the screen off.

mannodermaus, 2019-04-07:
I understand that the conditional around the Samsung fix will be enhanced with an additional check for `!PowerManager#isInteractive()`. I could assist with testing for iBeacon advertisements on an affected device (given that we have a sample app at hand for the scenario you mentioned), but I don't own any of the other standard beacon types. I can prepare the PR for this change!

davidgyoung, 2019-04-07:
It's actually quite easy to transmit a standard beacon from any smart phone using the LocateBeacon app or similar.  Regardless, I will absolutely regression test that before merging any PR

mannodermaus, 2019-04-07:
Thanks for the insight. I'll try that out when testing the fix! I just submitted the change as #855.

closed 2019-05-23 by davidgyoung