AltBeacon/android-beacon-library #849: BootstrapNotifier ...application not started on  Android 8 / 9

manandharsabir, 2019-03-28:
### Expected behavior
After I start the application  with BootstrapNotifier in Application class, after entering the region, the activity should have started even if it was killed.

### Actual behavior
Activity is not started

### Steps to reproduce this behavior
Create beacon region with UUID in MyApplication class
Start the app for first time and swipe kill the app.. 
Enter beacon region with assigned UUID. 

### Mobile device model and OS version
Samsung Galaxy s8

### Android Beacon Library version
android-beacon-library-2.16.1

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2019-03-28:
@manandharsabir, we'll need more specifics to make this actionable:

1.  Please try to reproduce this with the Reference Application to eliminate any custom application issues.
2. On Android 8+ you will have to wait up to 20 minutes for detection after the app is killed, as it takes this long for Android to schedule its ~15 minute schedule job that restarts scanning.  Please verify you have waited this long and indicate exactly what you see that is different than what you expect. (What are you not seeing that you expect to see?  How do you know you don't see it?  Looking at logs?  Looking at the screen?)

davidgyoung, 2019-05-29:
I cannot reproduce this issue.

Testing the reference app with a foreground service active, it appears to auto launch on BOOT_COMPLETED:

Samsung Galaxy S9 / Android 9.0:

```
05-29 14:05:44.501  1431  1497 I ActivityManager: Start proc 9275:org.altbeacon.beaconreference/u0a249 for broadcast org.altbeacon.beaconreference/org.altbeacon.beacon.startup.StartupBroadcastReceiver
05-29 14:05:44.503  9275  9275 E Zygote  : accessInfo : 1
05-29 14:05:44.505  9275  9275 I SELinux : SELinux: seapp_context_lookup: seinfo=default, level=s0:c249,c256,c512,c768, pkgname=org.altbeacon.beaconreference 
05-29 14:05:44.519  9275  9275 I beaconreferenc: Late-enabling -Xcheck:jni
05-29 14:05:44.565  9275  9275 D TimaKeyStoreProvider: TimaKeyStore is not enabled: cannot add TimaSignature Service and generateKeyPair Service
05-29 14:05:44.624  9275  9275 D ConnectivityManager_URSP: Ursp sIsUrsp=false, sIsCheckUrsp=false, uid=10249
05-29 14:05:44.627  9275  9275 D Proxy   : urspP is null: 10249
05-29 14:05:44.770  9275  9275 W beaconreferenc: JIT profile information will not be recorded: profile file does not exits.
05-29 14:05:44.773  9275  9275 I chatty  : uid=10249(org.altbeacon.beaconreference) identical 10 lines
05-29 14:05:44.773  9275  9275 W beaconreferenc: JIT profile information will not be recorded: profile file does not exits.
05-29 14:05:44.821  9275  9275 I InstantRun: starting instant run server: is main process
05-29 14:05:44.841  9275  9275 I BeaconManager: BeaconManager started up on pid 9275 named 'org.altbeacon.beaconreference' for application package 'org.altbeacon.beaconreference'.  isMainProcess=true
05-29 14:05:44.860  9275  9275 D BeaconParser: Parsing beacon layout: m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25
05-29 14:05:44.870  9275  9275 W BeaconManager: Disabling ScanJobs on Android 8+ may disable delivery of beacon callbacks in the background unless a foreground service is active.
05-29 14:05:44.871  9275  9275 D BeaconReferenceApp: setting up background monitoring for beacons and power saving
05-29 14:05:44.875  9275  9275 D BeaconManager: This consumer is not bound.  Binding now: org.altbeacon.beacon.startup.RegionBootstrap$InternalBeaconConsumer@436be16
05-29 14:05:44.875  9275  9275 D BeaconManager: Binding to service
05-29 14:05:44.876  9275  9275 I BeaconManager: Starting foreground beacon scanning service.
05-29 14:05:44.895  9275  9275 D BeaconManager: consumer count is now: 1
05-29 14:05:44.895  9275  9275 D AppStarter: Waiting for BeaconService connection
05-29 14:05:44.900  9275  9275 D StartupBroadcastReceiver: onReceive called in startup broadcast receiver
05-29 14:05:44.900  9275  9275 D StartupBroadcastReceiver: No consumers are bound.  Ignoring broadcast receiver.
05-29 14:05:44.901  1431  2134 I ActivityManager: Finished processing BOOT_COMPLETED for u0
05-29 14:05:44.909  9275  9275 I CycledLeScanner: Using Android O scanner
05-29 14:05:44.915  9275  9275 I BeaconService: beaconService version 2.16.1 is starting up on the main process
```

Google Pixel 3a / Android 9.0:

```
05-29 14:09:34.149  9487  9487 D BeaconReferenceApp: setting up background monitoring for beacons and power saving
05-29 14:09:34.456  9487  9487 D IntentHandler: Calling monitoring notifier: org.altbeacon.beaconreference.BeaconReferenceApplication@aa67dae
05-29 14:09:34.456  9487  9487 D BeaconReferenceApp: did enter region.
05-29 14:09:34.456  9487  9487 D BeaconReferenceApp: auto launching MainActivity
05-29 14:09:36.592  1151  3684 I ActivityManager: Finished processing BOOT_COMPLETED for u0
```

closed 2019-05-29 by davidgyoung