AltBeacon/android-beacon-library #850: Beacon monitoring stops when running as foreground service on Android 8 / 9

manandharsabir, 2019-03-28:
### Expected behavior
After application is started with beacon monitoring service as foreground service and application in background.... even after hours it should respond to beacon enter and exit events and also allow to range for beacons
### Actual behavior
It runs for some hours and after that stops responding to beacon enter/exit events... after application is brought to front, it starts responding to beacon monitoring events. 

### Steps to reproduce this behavior
Start application with foreground service with beacon monitoring mode..keep it running for few hours with app screen turned off ... it stops responding to beacon monitoring events. 
 
### Mobile device model and OS version
Samsung s8 ..Android 9
### Android Beacon Library version
android-beacon-library-2.16.1

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2019-04-04:
I don't think this is a universal issue as I have tested apps with a foreground service running for days on a Galaxy S9 with Android 8 (albeit not exactly the same combination you are describing.)  It is certainly true that the phone will go into Doze mode if left motionless for that period and detections will stop, but if you keep the phone in motion periodically this should not happen.

I'd suggest capturing a LogCat excerpt over the course of these hours to see what the last beacon detection event is logged by your app, and look for any system level messages afterward.  Could the app be crashing?  Could the OS be killing it for some reason?  Logs might give a clue.

Having a USB cable attached might affect your results, so if you don't see it stop after several hours while capturing logs over USB, you might consider setting up ADB over WiFi so you can capture a log from your workstation without a USB cable attached.  

One other tip:  don't use Android Studio to capture your log, as the log buffer will certainly wrap over this long period of time.  Instead, use a shell command to send the logcat output to a file, e.g. `adb logcat > big_logfile.txt`.  This way you can capture 8 hours, 12 hours or more as needed, then search thorough the file for the last detection event to see what happened right after.

joviden, 2019-05-06:
I have the same problem. Using foreground service for scanning. When the app is in the foreground there is no problem, but when I kill the app and foreground service starts, it scans for 3-5 minutes with no problem. But waiting enough while screen off it stops scanning still the foreground notification is active.

davidgyoung, 2019-05-23:
@joviden, can you see if you can capture system level logs when this happens?  I'd like to see if there is possibly a system log indicating a scan is blocked

paulpv, 2020-01-27:
Might this be fixed in #941 ?

paulpv, 2020-01-28:
I just noticed the following in the reference app's manifest, maybe it would resolve this issue?
```xml
        <!-- uncomment the following to enable scanning for over 30 minutes on Android 7+ -->
        <service android:name="org.altbeacon.beacon.service.BeaconService"
            tools:node="replace">
            <meta-data android:name="longScanForcingEnabled" android:value="true"/>
        </service>
```

davidgyoung, 2020-01-28:
@paulpv,  The description from @manandharsabir is that it runs for "hours" successfully in the background then stops.  If the longScanForcingEnabled were to be helpful, then @manandharsabir's  problem should have appeared within 30 minutes.  That said, @manandharsabir, you may wish to try this to be sure.