AltBeacon/android-beacon-library #851: The ScanHelper seems to be crashing while trying to register the scanner via ScanManager.registerScanner

agap, 2019-03-28:
### Expected behavior
The app's not crashing.

### Actual behavior

The app using the altbeacon library seems to be crashing for at least single device, LG G6. The stacktraces are as follows:
``` 
Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'void com.android.bluetooth.gatt.ScanManager.registerScanner(java.util.UUID)' on a null object reference
       at android.os.Parcel.readException(Parcel.java:1957)
       at android.os.Parcel.readException(Parcel.java:1897)
       at android.bluetooth.IBluetoothGatt$Stub$Proxy.startScanForIntent(IBluetoothGatt.java:1006)
       at android.bluetooth.le.BluetoothLeScanner.startScan(BluetoothLeScanner.java:252)
       at android.bluetooth.le.BluetoothLeScanner.startScan(BluetoothLeScanner.java:156)
       at org.altbeacon.beacon.service.ScanHelper.startAndroidOBackgroundScan(Unknown Source:79)
       at org.altbeacon.beacon.service.ScanJob.startPassiveScanIfNeeded(Unknown Source:88)
       at org.altbeacon.beacon.service.ScanJob.scheduleNextScan(Unknown Source:30)
       at org.altbeacon.beacon.service.ScanJob.access$300(Unknown Source)
       at org.altbeacon.beacon.service.ScanJob$1$1.run(Unknown Source:4)
       at android.os.Handler.handleCallback(Handler.java:789)
       at android.os.Handler.dispatchMessage(Handler.java:98)
       at android.os.Looper.loop(Looper.java:164)
       at android.app.ActivityThread.main(ActivityThread.java:6710)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:240)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:770)
```
```
Fatal Exception: java.lang.RuntimeException: java.lang.NullPointerException: Attempt to invoke virtual method 'void com.android.bluetooth.gatt.ScanManager.registerScanner(java.util.UUID)' on a null object reference
       at android.app.job.JobServiceEngine$JobHandler.handleMessage(JobServiceEngine.java:121)
       at android.os.Handler.dispatchMessage(Handler.java:105)
       at android.os.Looper.loop(Looper.java:164)
       at android.app.ActivityThread.main(ActivityThread.java:6710)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:240)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:770)
```
### Steps to reproduce this behavior
Not sure, the crashes are coming through the Crashlytics and seem to affect only a single user so far.

### Mobile device model and OS version
LG G6

### Android Beacon Library version
2.16.1

davidgyoung, 2019-03-28:
This issue was logged as #636, and the fix released in 2.16.1.  The current code that handles this is shown below, and it catches a NullPointerException.   I cannot explain how it is possible that such a crash could come from an app using library version 2.16.1.

Are you sure this is from a user with this library version number?  Is it possible that user has an older version of the app?

```
        try {
            final BluetoothManager bluetoothManager =
                    (BluetoothManager) mContext.getApplicationContext().getSystemService(Context.BLUETOOTH_SERVICE);
            BluetoothAdapter bluetoothAdapter = bluetoothManager.getAdapter();
            if (bluetoothAdapter == null) {
                LogManager.w(TAG, "Failed to construct a BluetoothAdapter");
            } else if (!bluetoothAdapter.isEnabled()) {
                LogManager.w(TAG, "Failed to start background scan on Android O: BluetoothAdapter is not enabled");
            } else {
                BluetoothLeScanner scanner = bluetoothAdapter.getBluetoothLeScanner();
                if (scanner != null) {
                    int result = scanner.startScan(filters, settings, getScanCallbackIntent());
                    if (result != 0) {
                        LogManager.e(TAG, "Failed to start background scan on Android O.  Code: " + result);
                    } else {
                        LogManager.d(TAG, "Started passive beacon scan");
                    }
                } else {
                    LogManager.e(TAG, "Failed to start background scan on Android O: scanner is null");
                }
            }
        } catch (SecurityException e) {
            LogManager.e(TAG, "SecurityException making Android O background scanner");
        } catch (NullPointerException e) {
            // Needed to stop a crash caused by internal NPE thrown by Android.  See issue #636
            LogManager.e(TAG, "NullPointerException starting Android O background scanner", e);
        } catch (RuntimeException e) {
            // Needed to stop a crash caused by internal Android throw.  See issue #701
            LogManager.e(TAG, "Unexpected runtime exception starting Android O background scanner", e);
        }

```

agap, 2019-03-29:
Yup, my bad, the issue seems to be fixed in 2.16.1 indeed. Sorry for the confusion!

closed 2019-03-29 by agap