AltBeacon/android-beacon-library #853: Crash recovery attempt is unsuccessful despite promising log statements

mannodermaus, 2019-04-05:
So we received this one complaint from a user who cannot find any nearby beacons with their phone. Their application logs are littered with Bluetooth service crashes, as well as the attempt at resolution through AltBeacon's `BluetoothCrashResolver`. It's ultimately unsuccessful, however there are a couple of logs directly from the Android `BluetoothAdapter` which seem to suggest that the service _did_ recover without telling the layer above?

```
// Shortly after bind() & startRanging()...
04-05 12:50:26.568 15353 15353 W c : a:L0 BLE (w): BluetoothService crash detected
04-05 12:50:26.571 15353 15353 W c : a:L0 BLE (w): Recovery attempt started

// This one is interesting, because yes, it is turned on (according to BluetoothAdapter#isEnabled())
04-05 12:50:26.572 15353 15353 W c : a:L0 BLE (w): Can't start discovery. Is Bluetooth turned on?

// Low-level callbacks?
04-05 12:50:27.482 15353 15364 D BluetoothAdapter: onBluetoothServiceDown: android.bluetooth.IBluetooth$Stub$Proxy@ff47a1c
04-05 12:50:27.482 15353 15364 D BluetoothAdapter: onBluetoothServiceDown: Sending callbacks to 0 clients
04-05 12:50:27.482 15353 15364 D BluetoothAdapter: onBluetoothServiceDown: Finished sending callbacks to registered clients
04-05 12:50:31.580 15353 15353 W c : a:L0 BLE (w): BluetoothAdapter.ACTION_DISCOVERY_STARTED never received. Recovery may fail.

04-05 12:50:31.661 15353 15353 W c : a:L0 BLE (w): BluetoothService crash detected
04-05 12:50:31.714 15353 15353 I c : a:L0 BLE (i): start ranging received
04-05 12:50:31.726 15353 15353 I c : a:L0 BLE (i): Adjusted scanStopTime to be 1100

// Another low-level callback; the Bluetooth service is actually back?
04-05 12:50:31.887 15353 15364 D BluetoothAdapter: onBluetoothServiceUp: android.bluetooth.IBluetooth$Stub$Proxy@787295

04-05 12:50:32.498 15353 15353 I c : a:L0 BLE (i): start ranging received
04-05 12:50:32.498 15353 15353 I c : a:L0 BLE (i): Already ranging that region -- will replace existing region.

// Recovery still ongoing because it wasn't notified of onBluetoothServiceUp
04-05 12:50:32.729 15353 15353 W c : a:L0 BLE (w): Skipping scan because crash recovery is in progress.
```

[This onBluetoothServiceUp/Down callback](https://android.googlesource.com/platform/frameworks/base/+/master/core/java/android/bluetooth/BluetoothAdapter.java#2855) is probably unreachable for us, which makes it quite difficult to say for sure if this is actually a recovered stack at this point. I have suggested a phone reboot to the user, and we'll see if that resolves it. Regardless, I wanted to share this log with you to check if you have any further insight on what might be happening here.

### Expected behavior
After resolving a crash in the underlying Bluetooth service, successful attempts are bubbled up to the app layer through AltBeacon.

### Actual behavior
AltBeacon might miss an internal callback and never resolve the crash recovery attempt.

### Steps to reproduce this behavior
Have this phone in this perfect storm of an unstable BT stack? From our analytics, this doesn't seem to happen on other units of the same model.

### Mobile device model and OS version
Sony SO-04H, Android 8.0.0

### Android Beacon Library version
2.15

davidgyoung, 2019-04-07:
Two points:

* The problem where the library gets stuck in crash recovery was fixed in #768 and that fix is in the latest version of the library.  This change effectively disables the BluetoothCrashResolver on Android 5+, as it is not effective on such devices.  For a fix to getting stuck in this state, simply update your app to the latest library version.

* There is a newer library utility you may use to try to fix problematic bluetooth stacks on 5.0+ devices.  That is called BluetoothMedic.  Here's a description of how it works:  https://stackoverflow.com/a/51450865/1461050

closed 2019-04-07 by davidgyoung

mannodermaus, 2019-04-07:
Thanks for the insight!