AltBeacon/android-beacon-library #857: Scan only one or two times in one hour on Android 7 or lower in background

prsolucoes, 2019-04-16:
### Expected behavior

On Android 8.0 and Android 9.0 appear don't have problems. 

### Actual behavior

But on Android 7.0 or lower it take a long time to scan beacons, 1 or 2 by hour and when scan, in background or closed.

In foreground i don't have problems. The problem is only in background and killed.

### Steps to reproduce this behavior

My flow is:

1 - Use AltBeacon library version 2.6.1
2 - Create a content provider that implement: BeaconConsumer, RangeNotifier and MonitorNotifier.
3 - Content provider is auto-initialized and beacon scan started.
4 - It scan the beacon and send info to our SDK that show or not notifications.
5 - Defined time:

```
beaconManager.setBackgroundBetweenScanPeriod(15 * 60 * 1000);
```

My question is:

What more i need do to use this library and start it services? I can listen in other ways to be more responsive to scans on other devices?

### Mobile device model and OS version

Some devices, i don't know because is from my customer.

### Android Beacon Library version

2.6.1

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2019-04-16:
This does not sound like a significantly different setup than the reference application here:

https://github.com/AltBeacon/android-beacon-library-reference

The main difference is that you configure the background between scan period to 15 minutes, whereas the reference app uses the default 5 minutes.  The reference app certainly does scans every 5 min.  Can you please try that to see if you can reproduce the same problem?  You can even add your line of code to set the background between scan period to 15 minutes.

On Android 7 and lower, the main difference is that a long running service is used to do scans vs. a 15 minute scheduled job on 8+.

One issue you may consider is whether network access is being blocked on Android 7 in Doze mode.  If true, this may mean that beacon scans are happening normally but network access to report them is blocked. This would not be a problem on Android 8+ because the scans are done with the Job Scheduler, which already uses the maintenance windows in Doze mode that allow network access.  You could force the same Job Scheduler use on Android 7 with `beaconManager.setEnableSceduledScanJobs(true)`

prsolucoes, 2019-04-16:
Hi,

I already used this reference project. I make mine based on it.

So, instead of Application i put it on content provider, that is auto started, useful to make libraries.

I set 5 minutes before, and now 15 to see if is not Android blocking system or something like this.

People are walking on shopping center to test, not doze mode.

I think that using this library, all services and other things are auto configured.

What i put now is:
- time back to 5 minutes
- add a services like in reference on manifest (don't know if it make difference):
```
<service android:name="org.altbeacon.beacon.service.BeaconService"
            tools:node="replace">
            <meta-data android:name="longScanForcingEnabled" android:value="true"/>
        </service>
```

Thanks for any help.

davidgyoung, 2019-04-16:
You will probably need to reproduce this problem to get to the bottom of it, and capture a LogCat excerpt over the time of the missing reports.  I understand such a capture is difficult if you can only reproduce the issue walking around a shopping center.  However, it may be possible to do by:

1. Set up a mobile hotspot on the Android phone
2. Connect to that hotspot from a laptop
3. Establish ADB over WiFi (See [here](https://stackoverflow.com/questions/2604727/how-can-i-connect-to-android-with-adb-over-tcp))
4. Capture a log for 60 minutes, or however log is needed to capture missing reports.

If you can do the above please set` beaconManager.setDebug(true);` to capture debug level output. 

If you can get a log trace from above, you should be able to search the log for `CycledLeScanner` and verify it is starting scans every 15 minutes.  If it is, you can then check to see what happens with those scan results such that they are note getting processed as you expect.   If you do not see these happen every 15 minutes, the OS logs may give us a clue as to why they are not -- is the app not running?

prsolucoes, 2019-04-16:
I made a test here, i don't know if it will help:

My sequence:

- Open app with AltBeacon
- iBeacon is detected
- I close the app killing it with swipe
- Open the same app again after kill
- iBeacon is not detected

I use other app to compare (https://play.google.com/store/apps/details?id=net.alea.beaconsimulator) and when always that i open the app my beacon is listed. I can close 10 times, on scanning the beacon is there.

The way of it show the beacon is different from AltBeacon library?

Thanks.

prsolucoes, 2019-04-16:
Hi,

Anyway im capturing 60 minutes of log.

Some message appear here in my test device with android 5:

```
I/CycledLeScanner(31963): This is Android 5.0.  We are using new scanning APIs
I/CycledLeScanner(  577): This is Android 5.0.  We are using new scanning APIs
I/CycledLeScanner(  577): This is Android 5.0.  We are using new scanning APIs
I/CycledLeScanner(  656): This is Android 5.0.  We are using new scanning APIs
I/CycledLeScanner(  656): This is Android 5.0.  We are using new scanning APIs
```

Thanks.

davidgyoung, 2019-04-16:
For any log excerpts, I really need to see the timestamps, otherwise I don't know the context of when the lines were logged.  As for killing the app and detecting afterward, **this is a different problem** than the one opened for this issue, correct?  You may want to add log lines to the point where you turn on beacon ranging or monitoring to make sure those lines execute.

prsolucoes, 2019-04-17:
Hi,

One question related here. The method that you sad:

```
beaconManager.setEnableScheduledScanJobs(true)
```

Need be executed for all version or lower than 8 (7, 6 , 5 ,4)?

Thanks.

davidgyoung, 2019-04-17:
Calling:

```
beaconManager.setEnableScheduledScanJobs(true)
```

Is unnecessary on Android 8+, as it is already the default.

On Android 5-7, it will work.

On Android 4.x it is unfortunately not supported, as the JobScheduler was only added with Android 5.

davidgyoung, 2019-05-29:
closing due to inactivity

closed 2019-05-29 by davidgyoung