AltBeacon/android-beacon-library #860: IllegalStateException: Apps may not schedule more than 100 distinct jobs

ryot, 2019-04-17:
### Expected behavior
No crash reported

### Actual behavior
java.lang.IllegalStateException: Apps may not schedule more than 100 distinct jobs

### Steps to reproduce this behavior
One of our new partners received an crashlytics report of this happening on user devices, and we have never seen or received a report of this type of crash before. We have not been able to reproduce it ourselves, and neither have they. Stacktrace:
```
Caused by java.lang.IllegalStateException: Apps may not schedule more than 100 distinct jobs
       at android.os.Parcel.readException(Parcel.java:2037)
       at android.os.Parcel.readException(Parcel.java:1975)
       at android.app.job.IJobScheduler$Stub$Proxy.schedule(IJobScheduler.java:180)
       at android.app.JobSchedulerImpl.schedule(JobSchedulerImpl.java:44)
       at org.altbeacon.beacon.service.ScanJobScheduler.schedule(ScanJobScheduler.java:249)
       at org.altbeacon.beacon.service.ScanJobScheduler.applySettingsToScheduledJob(ScanJobScheduler.java:85)
       at org.altbeacon.beacon.service.ScanJobScheduler.applySettingsToScheduledJob(ScanJobScheduler.java:92)
       at org.altbeacon.beacon.BeaconManager.applyChangesToServices(BeaconManager.java:1000)
       at org.altbeacon.beacon.BeaconManager.startMonitoringBeaconsInRegion(BeaconManager.java:939)
       at org.altbeacon.beacon.startup.RegionBootstrap$InternalBeaconConsumer.onBeaconServiceConnect(RegionBootstrap.java:203)
       at org.altbeacon.beacon.BeaconManager.bind(BeaconManager.java:425)
       at org.altbeacon.beacon.startup.RegionBootstrap.(RegionBootstrap.java:105)
       at com.x.y.z.LibraryStarter.startService(AreaMetrics.java:149)
       at com.x.y.z.AppSubclass.onCreate(MyBitsApp.java:62)
       at android.app.Instrumentation.callApplicationOnCreate(Instrumentation.java:1127)
       at android.app.ActivityThread.handleBindApplication(ActivityThread.java:6097)
       at android.app.ActivityThread.-wrap1(ActivityThread.java)
       at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1791)
       at android.os.Handler.dispatchMessage(Handler.java:106)
       at android.os.Looper.loop(Looper.java:164)
       at android.app.ActivityThread.main(ActivityThread.java:7000)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:441)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1408)
```
If it helps, here is the library initialization code we use inside of the Application subclass' onCreate:
```
beaconManager = BeaconManager.getInstanceForApplication(myContext);
beaconManager.getBeaconParsers().add(new BeaconParser("ib").setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24"));
beaconManager.setRegionStatePersistenceEnabled(false);
Region region = new Region("AnyBeacon", null, null, null);
regionBootstrap = new RegionBootstrap(this, region);
backgroundPowerSaver = new BackgroundPowerSaver(myContext);
beaconManager.bind(this);

beaconManager.setForegroundScanPeriod(1_100L);
beaconManager.setForegroundBetweenScanPeriod(5_000L);
beaconManager.setBackgroundScanPeriod(8_000L);
beaconManager.setBackgroundBetweenScanPeriod(300_000L);
beaconManager.updateScanPeriods();
```

### Mobile device model and OS version
OS Version: 8.1.0
Device: SM-G610M
RAM Free: 33.9%
Disk Free: 5.4%

### Android Beacon Library version
2.16.1

davidgyoung, 2019-04-21:
This suggests some kind of low-level problem with Android's job system.  "Distinct" jobs are kept distinct using a job identifier, and the Android Beacon Library uses exactly two values for the job id -- one for an immediate ScanJob (used to run right away, in the foreground, or in the background to parse a filtered scan result that comes in asynchronously)  and one for scheduled ScanJobs which come every 15 minutes.  I'm not sure how 100 distinct jobs could possibly be created unless the job identifiers somehow become corrupted by the operating system.

Whatever the root cause, it is unlikely that there is anything that the library could do to stop this if it is indeed an operating system problem.  Nevertheless, I appreciate you logging this hear in case other folks come across it as well.

csgulley, 2019-10-24:
Here is some additional information. We are using version 2.16.3 of the library. I have not been able to reproduce this personally but over 90 days we've recorded 2700 crashes from 160 users. 100% of the crashes have been on Android 9.

Here is the call stack and exception:

`Fatal Exception: java.lang.IllegalStateException: Not allowed to start service Intent { cmp=xxx/org.altbeacon.beacon.service.BeaconService }: app is in background uid UidRecord{4218edc u0a276 CEM  idle change:cached procs:1 proclist:29742, seq(0,0,0)}
       at android.app.ContextImpl.startServiceCommon + 1666(ContextImpl.java:1666)
       at android.app.ContextImpl.startService + 1611(ContextImpl.java:1611)
       at android.content.ContextWrapper.startService + 677(ContextWrapper.java:677)
       at org.altbeacon.beacon.startup.RegionBootstrap$InternalBeaconConsumer.bindService + 219(RegionBootstrap.java:219)
       at org.altbeacon.beacon.BeaconManager.bind + 443(BeaconManager.java:443)
       at org.altbeacon.beacon.startup.RegionBootstrap.<init> + 125(RegionBootstrap.java:125)
       [app frames deleted]
       at android.os.Handler.handleCallback + 873(Handler.java:873)
       at android.os.Handler.dispatchMessage + 99(Handler.java:99)
       at android.os.Looper.loop + 216(Looper.java:216)
       at android.app.ActivityThread.main + 7263(ActivityThread.java:7263)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run + 494(RuntimeInit.java:494)
       at com.android.internal.os.ZygoteInit.main + 975(ZygoteInit.java:975)`

Here is the calling code:

```
if (regionBootstrap == null) {
    manager = BeaconManager.getInstanceForApplication(application);
    final Notification notification = buildNotification(context);
    manager.enableForegroundServiceScanning(notification, VelApplication.NOTIFICATION_ID_BEACON);
    manager.setEnableScheduledScanJobs(false);
    manager.setBackgroundScanPeriod(BEACON_BACKGROUND_SCAN_PERIOD);
    manager.setBackgroundBetweenScanPeriod(BEACON_BACKGROUND_BETWEEN_SCAN_PERIOD);

    manager.getBeaconParsers().clear();
    manager.getBeaconParsers().add(new BeaconParser().setBeaconLayout("m:0-3=4c000215,i:4-19,i:20-21,i:22-23,p:24-24"));
    
    regionBootstrap = new RegionBootstrap(notifier, getRegions());
}
```

As a potential work around, I added code that would catch the IllegalStateException and then try to enable the beacon monitoring again 5 seconds later. I did this thinking maybe the condition that was causing the exception was transitory. However, the work around did not help. When we tried to enable the monitoring 5 seconds later an IllegalStateException exception was thrown again.

Let me know if there is any more information I can provide.

closed 2024-01-14 by davidgyoung