AltBeacon/android-beacon-library #862: delay on detecting nearby beacons sometimes while ranging

shihabal3amri, 2019-04-18:
### Expected behavior
What I am expecting once I open the fragment I'll be able to detect beacons immediately once the configuration is done and then I can do the action that I want to do.

### Actual behavior
What is happening sometimes it detects nearby beacons immediately which is what I want. However, sometimes there is a delay if I opened the fragment again and this delay takes around 6 ~ 18 seconds and what I've noticed in the log I can't detect beacons immediately when it shows for 6 seconds "Not stopping scan because this is Android N and we keep for scanning ... etc" till the time is over and then I can see all the nearby beacons. Moreover the scanning message sometimes increase and start another six seconds and sometimes it increase and maximum that I've seen 18 seconds of delay.

`2019-04-18 16:46:14.374 8789-8789 D/CycledLeScanner: Not stopping scan because this is Android N and we keep scanning for a minimum of 6 seconds at a time. We will stop in 4996 millisconds.
2019-04-18 16:46:15.278 8789-8789 D/CycledLeScanner: Not stopping scan because this is Android N and we keep scanning for a minimum of 6 seconds at a time. We will stop in 4092 millisconds.
2019-04-18 16:46:16.181 8789-8789 D/CycledLeScanner: Not stopping scan because this is Android N and we keep scanning for a minimum of 6 seconds at a time. We will stop in 3189 millisconds.
2019-04-18 16:46:17.085 8789-8789 D/CycledLeScanner: Not stopping scan because this is Android N and we keep scanning for a minimum of 6 seconds at a time. We will stop in 2285 millisconds.
2019-04-18 16:46:17.989 8789-8789 D/CycledLeScanner: Not stopping scan because this is Android N and we keep scanning for a minimum of 6 seconds at a time. We will stop in 1381 millisconds.
2019-04-18 16:46:18.892 8789-8789 D/CycledLeScanner: Not stopping scan because this is Android N and we keep scanning for a minimum of 6 seconds at a time. We will stop in 478 millisconds.
2019-04-18 16:46:20.705 8789-8789 D/CycledLeScanner: Not stopping scan because this is Android N and we keep scanning for a minimum of 6 seconds at a time. We will stop in 5097 millisconds.
2019-04-18 16:46:21.608 8789-8789 D/CycledLeScanner: Not stopping scan because this is Android N and we keep scanning for a minimum of 6 seconds at a time. We will stop in 4194 millisconds.
2019-04-18 16:46:22.541 8789-8789 D/CycledLeScanner: Not stopping scan because this is Android N and we keep scanning for a minimum of 6 seconds at a time. We will stop in 3261 millisconds.
2019-04-18 16:46:23.444 8789-8789 D/CycledLeScanner: Not stopping scan because this is Android N and we keep scanning for a minimum of 6 seconds at a time. We will stop in 2358 millisconds.
2019-04-18 16:46:24.348 8789-8789 D/CycledLeScanner: Not stopping scan because this is Android N and we keep scanning for a minimum of 6 seconds at a time. We will stop in 1454 millisconds.
2019-04-18 16:46:25.251 8789-8789 D/CycledLeScanner: Not stopping scan because this is Android N and we keep scanning for a minimum of 6 seconds at a time. We will stop in 551 millisconds.`

### Mobile device model and OS version
Mobile device: Galaxy Note 8
OS Version: Android 9

### Android Beacon Library version
Altbeacon 2.16.1

davidgyoung, 2019-04-18:
I don't think this problem has anything to do with the "Not stopping scan because this is Android N and we keep scanning for a minimum of 6 seconds at a time." messages, as that is normal on Android 7+.  Even if scanning is not stopped, your app still will receive ranging callbacks at the end of each scan cycle (1.1 secs by default) with a list of beacons detected. 

I suggest adding logging when ever you get a callback to didRangeBeaconsInRegion.  Do you get a callback every 1.1 seconds?  Is it he problem just that the list of beacons is sometimes 0?  If so, that simply means no beacons were detected in that cycle.  That can be normal, say if your beacon does not advertise frequently.noThis happens more often the more radio noise there is and the further away the beacon gets.

As this question is certainly about troubleshooting a custom app and not about any known problem with the library itself, this is really out of scope for this forum. I'm happy to assist troubleshooting your app further on StackOverflow.com if you'd like to post a question there.

closed 2019-04-18 by davidgyoung

shihabal3amri, 2019-04-19:
Thank you @davidgyoung
I've posted my question:
https://stackoverflow.com/questions/55754800/is-there-any-way-to-improve-start-detecting-beacons-using-altbeacon-while-device