AltBeacon/android-beacon-library #863: Beacon not detected when implementing BeaconConsumer in Application Class

sarikas26, 2019-04-21:
### Expected behavior
Should be able to get scanned beacon data in onBeaconServiceConnect method when implementing BeaconConsumer interface inside Application Class everytime.

### Actual behavior
It always show Current State as Outside in didDetermineStateForRegion callback method and beacon size is 0 on most of the devices but working fine only on a single device as expected. On that device also it stopped working after uninstalling and re-installing the app.

### Steps to reproduce this behavior
Implement BeaconConsumer interface in Application class instead of Activity class. Override all the 4 methods of BeaconConsumer  interface inside Application class. 

### Mobile device model and OS version
Moto G6 - android 9.0
Nexus 5X - android 6.0
Xiomi Redmi 4 - android 8.1

### Android Beacon Library version
2.16.1

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2019-04-21:
Please show your Application class definition, your onCreate method with setup, and the BeaconConsumer callback methods.

Please also add LogCat lines to each of these callbacks as well as to your didDetermineStateGmForRegion and didRangeBeaconsInRegion methods so you can verify if they are called, and attach a LogCat excerpt of startup.

sarikas26, 2019-04-22:
@davidgyoung Thanks for the quick reply but i found out the issue. It was not a library issue but some permission issue in my code. Following the sample reference code works fine and i am successfully able to detect beacons in range. So closing this issue.

closed 2019-04-22 by sarikas26