AltBeacon/android-beacon-library #864: Beacon is not detecting after device is reboot without opening the app itself

sitaram-gupta, 2019-04-22:
Log cat:

04-22 11:56:56.046 9160-9160/? I/art: Late-enabling -Xcheck:jni
04-22 11:56:56.097 9160-9160/? D/TidaProvider: TidaProvider()
04-22 11:56:56.126 9160-9160/? W/ReflectionUtils: java.lang.NoSuchMethodException: android.os.MessageQueue#enableMonitor()#bestmatch
        at miui.util.ReflectionUtils.findMethodBestMatch(ReflectionUtils.java:338)
        at miui.util.ReflectionUtils.findMethodBestMatch(ReflectionUtils.java:375)
        at miui.util.ReflectionUtils.callMethod(ReflectionUtils.java:800)
        at miui.util.ReflectionUtils.tryCallMethod(ReflectionUtils.java:818)
        at android.os.BaseLooper.enableMonitor(BaseLooper.java:47)
        at android.os.Looper.prepareMainLooper(Looper.java:112)
        at android.app.ActivityThread.main(ActivityThread.java:6221)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:904)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:794)
04-22 11:56:56.127 9160-9160/com.startup.wellcore V/BoostFramework: BoostFramework() : mPerf = com.qualcomm.qti.Performance@66ce5f8
04-22 11:56:56.631 9160-9160/com.startup.wellcore W/System: ClassLoader referenced unknown path: /data/app/com.startup.wellcore-1/lib/arm64
04-22 11:56:56.640 9160-9160/com.startup.wellcore W/ResourceType: No package identifier when getting name for resource number 0x00000000
04-22 11:56:56.958 9160-9160/com.startup.wellcore D/FirebaseApp: com.google.firebase.auth.FirebaseAuth is not linked. Skipping initialization.
04-22 11:56:56.961 9160-9160/com.startup.wellcore D/FirebaseApp: com.google.firebase.crash.FirebaseCrash is not linked. Skipping initialization.
04-22 11:56:56.962 9160-9160/com.startup.wellcore I/FirebaseInitProvider: FirebaseApp initialization successful
04-22 11:56:56.966 9160-9160/com.startup.wellcore I/InstantRun: starting instant run server: is main process
04-22 11:56:57.002 9160-9160/com.startup.wellcore I/BeaconManager: BeaconManager started up on pid 9160 named 'com.startup.wellcore' for application package 'com.startup.wellcore'.  isMainProcess=true
04-22 11:56:57.021 9160-9160/com.startup.wellcore D/BeaconParser: Parsing beacon layout: m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25
04-22 11:56:57.024 9160-9160/com.startup.wellcore D/BeaconParser: Parsing beacon layout: s:0-1=feaa,m:2-2=10,p:3-3:-41,i:4-21v
04-22 11:56:57.034 9160-9160/com.startup.wellcore D/aaaaBeaconReferenceApplication: setting up background monitoring for beacons and power saving
04-22 11:56:57.042 9160-9160/com.startup.wellcore W/System.err: android.os.RemoteException: The BeaconManager is not bound to the service.  Call beaconManager.bind(BeaconConsumer consumer) and wait for a callback to onBeaconServiceConnect()
04-22 11:56:57.043 9160-9160/com.startup.wellcore W/System.err:     at org.altbeacon.beacon.BeaconManager.applyChangesToServices(BeaconManager.java:1005)
        at org.altbeacon.beacon.BeaconManager.startRangingBeaconsInRegion(BeaconManager.java:849)
        at com.startup.wellcore.BeaconReferenceApplication.onCreate(BeaconReferenceApplication.java:125)
        at android.app.Instrumentation.callApplicationOnCreate(Instrumentation.java:1025)
04-22 11:56:57.044 9160-9160/com.startup.wellcore W/System.err:     at android.app.ActivityThread.handleBindApplication(ActivityThread.java:5506)
        at android.app.ActivityThread.-wrap2(ActivityThread.java)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1593)
        at android.os.Handler.dispatchMessage(Handler.java:102)
04-22 11:56:57.045 9160-9160/com.startup.wellcore W/System.err:     at android.os.Looper.loop(Looper.java:163)
        at android.app.ActivityThread.main(ActivityThread.java:6238)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:904)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:794)
04-22 11:56:57.052 9160-9160/com.startup.wellcore D/BeaconManager: This consumer is not bound.  Binding now: org.altbeacon.beacon.startup.RegionBootstrap$InternalBeaconConsumer@8f6a079
    Binding to service
04-22 11:56:57.059 9160-9160/com.startup.wellcore D/BeaconManager: consumer count is now: 2
04-22 11:56:57.059 9160-9160/com.startup.wellcore D/AppStarter: Waiting for BeaconService connection
04-22 11:56:57.175 9160-9160/com.startup.wellcore D/StartupBroadcastReceiver: onReceive called in startup broadcast receiver
    No consumers are bound.  Ignoring broadcast receiver.
04-22 11:56:57.193 9160-9160/com.startup.wellcore I/CycledLeScanner: This is Android 5.0.  We are using new scanning APIs
04-22 11:56:57.200 9160-9189/com.startup.wellcore I/FA: App measurement is starting up, version: 12780
    To enable debug logging run: adb shell setprop log.tag.FA VERBOSE
04-22 11:56:57.201 9160-9189/com.startup.wellcore I/FA: To enable faster debug mode event logging run:
      adb shell setprop debug.firebase.analytics.app com.startup.wellcore
04-22 11:56:57.205 9160-9160/com.startup.wellcore I/BeaconService: beaconService version 2.16.1 is starting up on the main process
04-22 11:56:57.206 9160-9160/com.startup.wellcore D/BeaconLocalBroadcastProcessor: Register calls: global=1 instance=1
04-22 11:56:57.227 9160-9160/com.startup.wellcore D/ModelSpecificDistanceCalculator: Finding best distance calculator for 7.1.2, N2G47H, Redmi 4X, Xiaomi
04-22 11:56:57.228 9160-9160/com.startup.wellcore D/AndroidModel: Score is 0 for XT1115;Moto X Pro;LXG22.67-7.1;5.0.2 compared to Xiaomi;Redmi 4X;N2G47H;7.1.2
    Score is 0 for LGE;Nexus 5;LPV79;4.4.2 compared to Xiaomi;Redmi 4X;N2G47H;7.1.2
04-22 11:56:57.229 9160-9160/com.startup.wellcore D/AndroidModel: Score is 0 for LGE;Nexus 4;KOT49H;4.4.2 compared to Xiaomi;Redmi 4X;N2G47H;7.1.2
04-22 11:56:57.229 9160-9160/com.startup.wellcore W/ModelSpecificDistanceCalculator: Cannot find match for this device.  Using default
04-22 11:56:57.229 9160-9160/com.startup.wellcore D/ModelSpecificDistanceCalculator: Finding best distance calculator for 7.1.2, N2G47H, Redmi 4X, Xiaomi
04-22 11:56:57.230 9160-9160/com.startup.wellcore D/AndroidModel: Score is 0 for XT1115;Moto X Pro;LXG22.67-7.1;5.0.2 compared to Xiaomi;Redmi 4X;N2G47H;7.1.2
04-22 11:56:57.231 9160-9160/com.startup.wellcore D/AndroidModel: Score is 0 for LGE;Nexus 5;LPV79;4.4.2 compared to Xiaomi;Redmi 4X;N2G47H;7.1.2
04-22 11:56:57.231 9160-9160/com.startup.wellcore D/AndroidModel: Score is 0 for LGE;Nexus 4;KOT49H;4.4.2 compared to Xiaomi;Redmi 4X;N2G47H;7.1.2
04-22 11:56:57.231 9160-9160/com.startup.wellcore W/ModelSpecificDistanceCalculator: Cannot find match for this device.  Using default
04-22 11:56:57.238 9160-9160/com.startup.wellcore D/BeaconService: No org.altbeacon.beacon.SimulatedScanData class exists.
04-22 11:56:57.239 9160-9196/com.startup.wellcore I/DynamiteModule: Considering local module com.google.android.gms.firebase_database:4 and remote module com.google.android.gms.firebase_database:6
04-22 11:56:57.239 9160-9160/com.startup.wellcore I/BeaconService: binding
04-22 11:56:57.240 9160-9196/com.startup.wellcore I/DynamiteModule: Selected remote version of com.google.android.gms.firebase_database, version >= 6
04-22 11:56:57.242 9160-9160/com.startup.wellcore I/BeaconService: starting with intent Intent { cmp=com.startup.wellcore/org.altbeacon.beacon.service.BeaconService }
04-22 11:56:57.254 9160-9160/com.startup.wellcore D/BeaconManager: we have a connection to the service now
    Not synchronizing settings to service, as it is in the same process
[... 66 lines cut ...]
04-22 11:56:58.925 9160-9160/com.startup.wellcore D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
04-22 11:56:58.928 9160-9160/com.startup.wellcore D/CycledLeScanner: stopping bluetooth le scan
04-22 11:56:58.928 9160-9160/com.startup.wellcore D/CycledLeScannerForLollipop: Stopping scan
04-22 11:56:58.931 9160-9198/com.startup.wellcore D/CycledLeScannerForLollipop: Stopping LE scan on scan handler
04-22 11:56:58.931 9160-9160/com.startup.wellcore D/CycledLeScanner: Normalizing between scan period from 300000 to 96163
04-22 11:56:58.932 9160-9160/com.startup.wellcore D/CycledLeScannerForLollipop: This is Android L. Preparing to do a filtered scan for the background.
04-22 11:56:58.932 9160-9198/com.startup.wellcore D/BluetoothAdapter: isLeEnabled(): ON
[... cut ...] 1F:3A:06:C3:BA, notificationTitle=notificationTitleNew}}
04-22 11:57:00.946 9160-9160/com.startup.wellcore D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 94149 milliseconds
04-22 11:57:01.947 9160-9160/com.startup.wellcore D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 93147 milliseconds
04-22 11:57:02.950 9160-9160/com.startup.wellcore D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 92145 milliseconds

sitaram-gupta, 2019-04-22:
beacon is only detecting upon charger connect/disconnect even app is killed.

davidgyoung, 2019-04-22:
@sitaram-gupta, please keep this to one issue at a time.  Reporting multiple problems in one issue makes it very difficult to assist.  Please open a different issue about "wellcore beacon" if needed, but please do some research first to find out the format of that beacon.  Perhaps try to detect it with an off the shelf detector like BeaconScope or Locate from the PlayStore and report results.

For the detection after reboot issue, please fill out all the info in the new issue template including: the device model, operating system version, and library version.  I also need to know what code you tested with.  Was this an unmodified version of the reference app?  Was a foreground service enabled?

Please understand that this open source project is a volunteer community effort, so your help and cooperation are needed to resolve any issues reported.

I am going to close this issue so that you can fill out the new issue template again.  I am happy to help, but please understand I cannot do so unless you provide the info requested.

closed 2019-04-22 by davidgyoung

sitaram-gupta, 2019-04-23:
Thank you will do that as you said.