AltBeacon/android-beacon-library #865: Beacon is not detecting after device is reboot without opening the app itself

sitaram-gupta, 2019-04-23:
### Expected behavior
Detect beacon after device reboot without opening the app again.
### Actual behavior
Its launching the app in background in 5 minutes of device reboot. but not detecting beacons.
### Steps to reproduce this behavior
1) i installed and launched the app beacon is detecting.
2) i killed the app and connect/disconnect charger its working fine i mean beacon is detecting.  
3)   i killed the app ,and without connecting/disconnecting charger its not detecting beacon at all even waiting for 1 hour.

     and     if i reboot the device and waiting for 5 minutes its also not detecting beacon until unless app opened again manually.

### Mobile device model and OS version
Redmi 4x  OS version:-7.1.1
### Android Beacon Library version
2.16.1

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2019-04-23:
For the reboot case:

How do you know the app "is launching the app in background in 5 minutes of device reboot. but not detecting beacons."?  Do you see lines from your app in LogCat after reboot?  If so can you please set `beaconManager.setDebug(true)`, run,  reboot your device, then attach a capture of any lines printed from your app in LogCat.

Are you using a ForegroundService or not?

stephenruda, 2019-05-10:
Is it possible that your device is restricting the app from running in the background or after startup?  A lot of manufacturers add their own battery-saving implementations that will Force Stop apps once in the background for a while or prevent them from running on start up.  It is not specific to the android-beacon-library, it would stop all of your background work.  This is something I have struggled with for a long time with my app.  [See if you can follow this guide to make sure your app is whitelisted on your device.](https://github.com/dirkam/backgroundable-android#xiaomi-auto-start-and-battery-usage-monitoring-and-power-settings)

I came across this project when I was having issues with Samsung Unmonitored Apps and it is a great tool for diagnosing issues like this.  These 'battery managers' that comes pre-built on these manufacturer's devices are extremely frustrating for any developer that is trying to do background work on Android.

tianhui2019, 2019-05-25:
Xiaomi is serious on battery, you can try to test on manufacturers' Android device.

davidgyoung, 2019-10-18:
Tracking this now in related issue #928

closed 2019-10-18 by davidgyoung