AltBeacon/android-beacon-library #868: Exit region event firing after a long time

KeyStarr, 2019-04-30:
I tested the library-reference having modified it only to send notifications on each `didEnterRegion` and `didExitRegion` events (for better visualization).

### Expected behavior
App is in background. Thanks to the [fast BLE background scans ](https://altbeacon.github.io/android-beacon-library/battery_manager.html) i expect the following:

1. On entering a region i expect to receive `didEnterRegion` event in a matter of 10-15 seconds

2. On exiting a region i expect to receive `didExitRegion` event in a matter of 10-15 seconds

### Actual behavior
On entering a region event was received in under 2 seconds, whereas on exiting it came only after 5 minutes. 
On tests that i did not record this time could raise up to 8-10 minutes.

### Steps to reproduce this behavior

I attach my test report, so the steps are clear from it. The device used in this test was Google Pixel 2 (Android 9).

1. 13:41:20 - app in background
2. 13:43:20 beacon was enabled, app launched itself in a foreground, i almost immediately sent it back to the background. 
Log:

`2019-04-30 13:43:25.846 11508-11508/org.altbeacon.beaconreference D/StartupBroadcastReceiver: onReceive called in startup broadcast receiver
2019-04-30 13:43:25.850 11508-11508/org.altbeacon.beaconreference D/StartupBroadcastReceiver: Passive background scan callback type: 1
2019-04-30 13:43:25.850 11508-11508/org.altbeacon.beaconreference D/StartupBroadcastReceiver: got Android O background scan via intent
2019-04-30 13:43:25.850 11508-11508/org.altbeacon.beaconreference D/ScanJobScheduler: scheduling an immediate scan job because last did 1556610205850seconds ago.
2019-04-30 13:43:25.880 11508-11508/org.altbeacon.beaconreference D/ScanState: Scan state restore regions: monitored=1 ranged=0
2019-04-30 13:43:25.881 11508-11508/org.altbeacon.beaconreference D/BeaconLocalBroadcastProcessor: Register calls: global=7 instance=7
2019-04-30 13:43:25.881 11508-11508/org.altbeacon.beaconreference D/ScanState: ScanState says background mode for ScanJob is true
2019-04-30 13:43:25.881 11508-11508/org.altbeacon.beaconreference D/ScanJobScheduler: We just woke up in the background based on a new scan result.  Start scan job immediately.
2019-04-30 13:43:25.881 11508-11508/org.altbeacon.beaconreference D/ScanJobScheduler: Scheduling immediate ScanJob to run in 0 millis
2019-04-30 13:43:25.884 11508-11508/org.altbeacon.beaconreference I/ScanJob: Using immediateScanJobId from manifest: 208352939
2019-04-30 13:43:25.900 11508-11508/org.altbeacon.beaconreference I/ScanJob: Using periodicScanJobId from manifest: 208352940
2019-04-30 13:43:25.900 11508-11508/org.altbeacon.beaconreference W/JobInfo: Requested interval +5m10s0ms for job 208352940 is too small; raising to +15m0s0ms
2019-04-30 13:43:25.900 11508-11508/org.altbeacon.beaconreference W/JobInfo: Requested flex 0 for job 208352940 is too small; raising to +5m0s0ms
2019-04-30 13:43:25.900 11508-11508/org.altbeacon.beaconreference D/ScanJobScheduler: Scheduling ScanJob (job:208352940/org.altbeacon.beaconreference/org.altbeacon.beacon.service.ScanJob) to run every 310000 millis
2019-04-30 13:43:25.928 11508-11508/org.altbeacon.beaconreference D/ScanState: Scan state restore regions: monitored=1 ranged=0
2019-04-30 13:43:25.928 11508-11508/org.altbeacon.beaconreference I/CycledLeScanner: Using Android O scanner
2019-04-30 13:43:25.934 11508-11508/org.altbeacon.beaconreference I/ScanJob: Using immediateScanJobId from manifest: 208352939
2019-04-30 13:43:25.935 11508-11508/org.altbeacon.beaconreference I/ScanJob: Running immediate scan job: instance is org.altbeacon.beacon.service.ScanJob@4768532
2019-04-30 13:43:25.935 11508-11508/org.altbeacon.beaconreference D/ScanJob: Processing 1 queued scan resuilts
2019-04-30 13:43:25.936 11508-11508/org.altbeacon.beaconreference D/ScanJob: Done processing queued scan resuilts
2019-04-30 13:43:25.936 11508-11508/org.altbeacon.beaconreference I/ScanJob: scanJob version 2.16.1 is starting up on the main process
2019-04-30 13:43:25.937 11508-12062/org.altbeacon.beaconreference D/BeaconParser: Ignoring pdu type 01
2019-04-30 13:43:25.937 11508-12062/org.altbeacon.beaconreference D/BeaconParser: Processing pdu type FF: 0201061bff1801beac2f234454cf6d4a0fadf2f4911ba9ffa6007b01c8be6400000000000000000000000000000000000000000000000000000000000000 with startIndex: 5, endIndex: 30
2019-04-30 13:43:25.937 11508-12062/org.altbeacon.beaconreference D/BeaconParser: This is a recognized beacon advertisement -- be ac seen
2019-04-30 13:43:25.937 11508-12062/org.altbeacon.beaconreference D/BeaconParser: Bytes are: 0201061bff1801beac2f234454cf6d4a0fadf2f4911ba9ffa6007b01c8be6400000000000000000000000000000000000000000000000000000000000000
2019-04-30 13:43:25.940 11508-11508/org.altbeacon.beaconreference D/ModelSpecificDistanceCalculator: Finding best distance calculator for 9, PQ2A.190305.002, Pixel 2, Google
2019-04-30 13:43:25.940 11508-11508/org.altbeacon.beaconreference D/AndroidModel: Score is 0 for LGE;Nexus 5;LPV79;4.4.2 compared to Google;Pixel 2;PQ2A.190305.002;9
2019-04-30 13:43:25.940 11508-11508/org.altbeacon.beaconreference D/AndroidModel: Score is 0 for LGE;Nexus 4;KOT49H;4.4.2 compared to Google;Pixel 2;PQ2A.190305.002;9
2019-04-30 13:43:25.941 11508-11508/org.altbeacon.beaconreference D/AndroidModel: Score is 0 for XT1115;Moto X Pro;LXG22.67-7.1;5.0.2 compared to Google;Pixel 2;PQ2A.190305.002;9
2019-04-30 13:43:25.941 11508-11508/org.altbeacon.beaconreference W/ModelSpecificDistanceCalculator: Cannot find match for this device.  Using default
2019-04-30 13:43:25.941 11508-11508/org.altbeacon.beaconreference D/ModelSpecificDistanceCalculator: Finding best distance calculator for 9, PQ2A.190305.002, Pixel 2, Google
2019-04-30 13:43:25.941 11508-11508/org.altbeacon.beaconreference D/AndroidModel: Score is 0 for LGE;Nexus 5;LPV79;4.4.2 compared to Google;Pixel 2;PQ2A.190305.002;9
2019-04-30 13:43:25.941 11508-11508/org.altbeacon.beaconreference D/AndroidModel: Score is 0 for LGE;Nexus 4;KOT49H;4.4.2 compared to Google;Pixel 2;PQ2A.190305.002;9
2019-04-30 13:43:25.941 11508-11508/org.altbeacon.beaconreference D/AndroidModel: Score is 0 for XT1115;Moto X Pro;LXG22.67-7.1;5.0.2 compared to Google;Pixel 2;PQ2A.190305.002;9
2019-04-30 13:43:25.941 11508-11508/org.altbeacon.beaconreference W/ModelSpecificDistanceCalculator: Cannot find match for this device.  Using default
2019-04-30 13:43:25.941 11508-12062/org.altbeacon.beaconreference D/ScanHelper: Beacon packet detected for: id1: 2f234454-cf6d-4a0f-adf2-f4911ba9ffa6 id2: 123 id3: 456 type altbeacon with rssi -77
2019-04-30 13:43:25.945 11508-11508/org.altbeacon.beaconreference D/BluetoothAdapter: isLeEnabled(): ON
2019-04-30 13:43:25.946 11508-12062/org.altbeacon.beaconreference D/ScanHelper: beacon detected : id1: 2f234454-cf6d-4a0f-adf2-f4911ba9ffa6 id2: 123 id3: 456 type altbeacon
2019-04-30 13:43:25.947 11508-12062/org.altbeacon.beaconreference D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.monitor_notification
2019-04-30 13:43:25.948 11508-12062/org.altbeacon.beaconreference D/MonitoringStatus: saveMonitoringStatusIfOn()
2019-04-30 13:43:25.948 11508-11508/org.altbeacon.beaconreference D/CycledLeScanner: Set scan periods called with 10000, 300000 Background mode must have changed.
2019-04-30 13:43:25.948 11508-11508/org.altbeacon.beaconreference D/CycledLeScanner: We are in the background.  Setting wakeup alarm
2019-04-30 13:43:25.952 11508-12062/org.altbeacon.beaconreference D/ScanHelper: looking for ranging region matches for this beacon
2019-04-30 13:43:25.952 11508-11508/org.altbeacon.beaconreference D/CycledLeScanner: Set a wakeup alarm to go off in 300000 ms: PendingIntent{da2a82c: android.os.BinderProxy@7374cfe}
2019-04-30 13:43:25.952 11508-11508/org.altbeacon.beaconreference D/CycledLeScanner: start called
2019-04-30 13:43:25.952 11508-11508/org.altbeacon.beaconreference D/CycledLeScanner: starting a new scan cycle
2019-04-30 13:43:25.953 11508-11508/org.altbeacon.beaconreference D/CycledLeScanner: starting a new bluetooth le scan
2019-04-30 13:43:25.954 11508-11508/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: starting a scan in SCAN_MODE_LOW_LATENCY
2019-04-30 13:43:25.954 11508-11508/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: Using an empty scan filter since this is 8.1+ on Non-Samsung
[... 352 lines cut ...]
....
[... cut ...]
2019-04-30 13:53:58.043 11508-11508/org.altbeacon.beaconreference D/ScanHelper: Started passive beacon scan

(and more about accepting/dropping PDUs)`

### Mobile device model and OS version
Tested on Google Pixel 2.0 (android 9) and Xiaomi Redmi 4X (android 7.1.2).

### Android Beacon Library version
2.16.1 

So, the question - is it okay when app is in background for the `didExitRegion` event to come only after 5-15 minutes? If so, why?

To better understand the use case - i am tracking the amount of time the user spends in a region. Thus, precision is of importance.

davidgyoung, 2019-04-30:
What you describe is expected behaviour.  The hardware filters used by the library provide entry events within a few seconds in the background.  But exit events require a full software scan to check if a beacon is disappeared.  On Android 8+ these are scheduled every 15 minutes using a JobScheduler (the mist frequent time allowed by the OS), which is why it can take ~15 minutes to get an exit event.

I do have an experimental branch that adds the feature you wish in #869, but it requires a source build to try it.  I gave found it to be helpful in many cases but not super reliable.

closed 2019-04-30 by davidgyoung

KeyStarr, 2019-04-30:
Thank you for the quick answer! And for the great library :)

I would suggest adding this piece of knowledge to the corresponding guide, as personally i find it to be unclear.

davidgyoung, 2019-04-30:
There is a table [here](http://www.davidgyoungtech.com/2017/08/07/beacon-detection-with-android-8) that lists the max secs to detect disappear at 450 secs, bit this is actually not correct on Android 8+ as it should be 1500.  

One other alternative for your use case:

1. Use a foreground service to unlock more frrequentt scanning.
2. On region entry, set a backgroundScanPeriod/backgroundBetweenScanPeriod of 5000/25000 ms to give you scans every 25 secs.  This should give you region exits within 60 secs, but battery usage will be higher. You can then change the scan periods back to the defaults on region exit to save battery.

KeyStarr, 2019-04-30:
The table is great! But i meant including this info into the Docs section of the https://altbeacon.github.io/ , because otherwise it is not really searchable (or have i missed it there?).

The feature is for the e-commerce app, so it is not too important to disturb user with a permament notification (1) or to drain too much battery (besides the majority of our users are android 8+). But thank you for the suggestion!

However, i found your solution in #869 interesting. Could you, please, expand a bit more on why it isn't reliable? Does it mean that it will improve precision only for some users while not introducing any other issues/crashes?

davidgyoung, 2019-04-30:
@KeyStarr, to make this easier to find, I added the table to a new page linked off of docs [here](https://altbeacon.github.io/android-beacon-library/detection_times.html)

I also added some notes about testing #869 on that issue.  You are welcome to try it, if you are up to compiling the library from source.