AltBeacon/android-beacon-library #870: Using versions from 2.15.3 to 2.16.1, the Eddystone beacons are no longer detected on Samsung with Android 8.1

manuelmacedo, 2019-05-03:
### Expected behavior
I expected to detect the UID frames of the Eddystone beacons, as it was possible with versions prior to 2.15.3.
In version 2.15.3 a change was made that specifically affected these devices (Samsung with 8.1+), it may be related with that change.
### Actual behavior
Using versions from 2.15.3 to 2.16.1, the Eddystone beacons are no longer detected.
### Steps to reproduce this behavior
- Set a beacon parser for the UID frame of Eddystone beacons.
ex: beaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout(BeaconParser.EDDYSTONE_UID_LAYOUT));
-Create a region with null identifiers
ex: localRegion = new Region("[UNIQUE_ID]", null, bull, null);
-Start ranging beacons in that region
ex: beaconManager.startRangingBeaconsInRegion(localRegion);
### Mobile device model and OS version
Samsung J4+ Android 8.1
### Android Beacon Library version
2.15.3 to 2.16.1

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2019-05-03:
@manuelmacedo can you confirm if this is just in the background?  There was a change that added a scan filter in the background ONLY for 8.1+ samsung versions.  If you see this in the foreground there may be another explanation.

It would be very helpful to getting to the bottom of this if you could turn on debug logging with beaconManager.setDebug(true) and capture two different things:

1. A beacon detection sequence in the foreground (assuming it works there)  It will look something like this:
```
2019-01-31 20:03:18.453 18029-18430 D/BeaconParser: This is a recognized beacon advertisement -- 02 15 seen
2019-01-31 20:03:18.453 18029-18430 D/BeaconParser: Bytes are: 0201061aff4c000215f7826da64fa24e988024bc5b71e0893e00019567c50d0961626561636f6e5f36373935000000000000000000000000000000000000
```

2. setting up the scan filter in the background.  It will look something like this:
```
05-03 10:41:42.535 17749 17749 D ScanFilterUtils: Set up a scan filter: BluetoothLeScanFilter [mDeviceName=null, mDeviceAddress=null, mUuid=null, mUuidMask=null, mServiceDataUuid=null, mServiceData=null, mServiceDataMask=null, mManufacturerId=4c, mManufacturerData=02 15 8F DF 29 81 FD 3A 00 00 56 E7 4F 03 A0 A2 6A 38 00 01, mManufacturerDataMask=FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF]
```

manuelmacedo, 2019-05-06:
Hi David, the behavior seems to be the same with the screen On or Off.
I've made a test with the old and new version of the library.
I'm sorry for the log length, specially for the old version, but I didn't know exactly which part would be more important.

The log for both versions is the following:

**Version 2.16.1**

2019-05-06 14:13:59.036 14041-14041/ D/CycledLeScannerForLollipop: starting a scan in SCAN_MODE_LOW_LATENCY
2019-05-06 14:13:59.037 14041-14041/ D/CycledLeScannerForLollipop: Using a non-empty scan filter since this is Samsung 8.1+
2019-05-06 14:13:59.037 14041-14041/ D/ScanFilterUtils: Set up a scan filter: BluetoothLeScanFilter [mDeviceName=null, mDeviceAddress=null, mUuid=null, mUuidMask=null, mServiceDataUuid=null, mServiceData=null, mServiceDataMask=null, mManufacturerId=118, mManufacturerData=BE AC, mManufacturerDataMask=FF FF]
2019-05-06 14:13:59.038 14041-14041/ D/ScanFilterUtils: making scan filter for service: 0000FEAA-0000-1000-8000-00805f9b34fb 0000feaa-0000-1000-8000-00805f9b34fb
2019-05-06 14:13:59.039 14041-14041/ D/ScanFilterUtils: making scan filter with service mask: FFFFFFFF-FFFF-FFFF-FFFF-FFFFFFFFFFFF ffffffff-ffff-ffff-ffff-ffffffffffff
2019-05-06 14:13:59.039 14041-14041/ D/ScanFilterUtils: Set up a scan filter: BluetoothLeScanFilter [mDeviceName=null, mDeviceAddress=null, mUuid=0000feaa-0000-1000-8000-00805f9b34fb, mUuidMask=ffffffff-ffff-ffff-ffff-ffffffffffff, mServiceDataUuid=null, mServiceData=null, mServiceDataMask=null, mManufacturerId=ffffffff, mManufacturerData=null, mManufacturerDataMask=null]
2019-05-06 14:13:59.039 14041-14041/ D/CycledLeScanner: Waiting to stop scan cycle for another 5000 milliseconds
2019-05-06 14:13:59.040 14041-14041/ D/CycledLeScanner: Scan started
2019-05-06 14:13:59.040 14041-14041/ D/IntentHandler: got ranging data
2019-05-06 14:13:59.042 14041-14041/ D/ BLEAltBeaconManager_didRangeBeaconsInRegion: No beacons found

**Version 2.13.1**
2019-05-06 14:23:44.762 15132-15132/I/BeaconService: start ranging received
2019-05-06 14:23:44.764 15132-15132/D/BeaconService: Currently ranging 1 regions.
2019-05-06 14:23:44.764 15132-15132/D/CycledLeScanner: start called
2019-05-06 14:23:44.776 15132-15132/D/CycledLeScanner: starting a new scan cycle
2019-05-06 14:23:44.778 15132-15132/D/CycledLeScanner: starting a new bluetooth le scan
2019-05-06 14:23:44.780 15132-15132/D/CycledLeScannerForLollipop: starting non-filtered scan in SCAN_MODE_LOW_LATENCY
2019-05-06 14:23:44.783 15132-15132/D/CycledLeScannerForLollipop: Making new Android L scanner
2019-05-06 14:23:44.785 15132-15132/D/BluetoothAdapter: STATE_ON
2019-05-06 14:23:44.787 15132-15132/D/BluetoothAdapter: STATE_ON
2019-05-06 14:23:44.790 15132-15132/D/CycledLeScanner: Waiting to stop scan cycle for another 1100 milliseconds
2019-05-06 14:23:44.790 15132-15132/D/CycledLeScanner: Scan started
2019-05-06 14:23:44.791 15132-15560/D/BluetoothAdapter: STATE_ON
2019-05-06 14:23:44.792 15132-15132/D/CycledLeScanner: Set scan periods called with 5000, 5000 Background mode must have changed.
2019-05-06 14:23:44.792 15132-15132/D/CycledLeScanner: We are not in the background.  Cancelling wakeup alarm
2019-05-06 14:23:44.792 15132-15560/D/BluetoothAdapter: STATE_ON
2019-05-06 14:23:44.793 15132-15132/D/CycledLeScanner: cancel wakeup alarm: null
2019-05-06 14:23:44.795 15132-15560/D/BluetoothAdapter: STATE_ON
2019-05-06 14:23:44.796 15132-15560/D/BluetoothAdapter: STATE_ON
2019-05-06 14:23:44.802 15132-15560/D/BluetoothLeScanner: Start Scan with callback
2019-05-06 14:23:44.803 1241-2158/? D/SamsungAlarmManager: setInexact Intent (T:2/F:0/AC:false) 19310225T051702 - CU:10165/CP:15132
2019-05-06 14:23:44.806 15132-15132/D/CycledLeScanner: Set a wakeup alarm to go off in 9223372036847580406 ms: PendingIntent{5b18c4: android.os.BinderProxy@3ed73ad}
2019-05-06 14:23:44.807 1798-2342/? I/bt_stack: [INFO:gatt_api.cc(1014)] GATT_Register
2019-05-06 14:23:44.808 1798-2342/? I/bt_stack: [INFO:gatt_api.cc(1055)] allocated gatt_if=8
2019-05-06 14:23:44.809 1823-15564/? I/SKBD: aph [PF_KL][SIIC] getCursorCapsMode done  121
2019-05-06 14:23:44.809 15132-15201/D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=8 mScannerId=0
2019-05-06 14:23:44.810 1823-1823/? I/SKBD: aph [PF_KL][SIIC] getCursorCapsMode done  104
2019-05-06 14:23:44.811 1823-1823/? I/SKBD_UnifiedIME: ToolBarManager setToolbarVisibility, visible = false
2019-05-06 14:23:44.813 1823-1823/? I/SKBD: bhq startInputViewShowTips: true true
2019-05-06 14:23:44.813 1823-1823/? I/SKBD: bhq startInputViewShowTips: conditions for show dialog : true true true
2019-05-06 14:23:44.813 1823-1823/? I/SKBD: bbk getDefaultInputRange : UNKNOWN
2019-05-06 14:23:44.814 1823-1823/? I/SKBD: aoq sendBroadcastSIPShown()
2019-05-06 14:23:44.820 1823-15567/? I/SKBD: ant no need to check Hint, Limit exceed or Bixby is Off
2019-05-06 14:23:44.820 1823-1823/? I/SKBD_Performance: SamsungKeypad [PF_OP] onStartInputView 314421510
2019-05-06 14:23:44.822 1798-2310/? D/BluetoothAdapter: STATE_ON
2019-05-06 14:23:44.822 1241-2158/? D/InputMethodManagerService: isImeSwitcherDisabledPackage : false
2019-05-06 14:23:44.822 1241-2158/? D/InputMethodManagerService: shouldShowImeSwitcherLocked : checking vis : 3
2019-05-06 14:23:44.824 1241-2158/? D/InputMethodManagerService: mDesktopModeManager null!
2019-05-06 14:23:44.824 1241-2158/? D/InputMethodManagerService: updateSystemUiLocked 0
2019-05-06 14:23:44.825 1798-2310/? D/BluetoothAdapter: STATE_ON
[... cut ...]
2019-05-06 14:23:45.931 1798-2342/? I/bt_stack: [INFO:gatt_api.cc(1055)] allocated gatt_if=8
2019-05-06 14:23:45.933 15132-15576/D/IntentHandler: got ranging data
2019-05-06 14:23:45.934 15132-15570/D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=8 mScannerId=0
2019-05-06 14:23:45.936 15132-15576/D/BLEAltBeaconManager_didRangeBeaconsInRegion: 1 beacons found
2019-05-06 14:23:45.938 15132-15576/D/BLEAltBeaconManager_didRangeBeaconsInRegion: BLE scan result found: 00:1E:C0:62:85:30

davidgyoung, 2019-05-06:
Thanks for the logs, @manuelmacedo.  Very helpful.

Okay, so this looks like a bug in the library.  The scan filters the library creates for Eddystone beacons look like they have *always* been wrong.  Nobody reported this until now because there was always a periodic backup scan that didn't use scan filters until Samsung came out with their custom variant of Android 8.1, which required non-empty scan filters to be used in the background.  The library was updated to do this as Samsung requires, but since the scan filter was incorrect for eddystone, this never matches anything.

The specific problem with the scan filter is that it includes a manufactuererId in the filter, something that service advertisements like Eddystone do not have.  

```
2019-05-06 14:13:59.039 14041-14041/ D/ScanFilterUtils: making scan filter with service mask: FFFFFFFF-FFFF-FFFF-FFFF-FFFFFFFFFFFF ffffffff-ffff-ffff-ffff-ffffffffffff
2019-05-06 14:13:59.039 14041-14041/ D/ScanFilterUtils: Set up a scan filter: BluetoothLeScanFilter [mDeviceName=null, mDeviceAddress=null, mUuid=0000feaa-0000-1000-8000-00805f9b34fb, mUuidMask=ffffffff-ffff-ffff-ffff-ffffffffffff, mServiceDataUuid=null, mServiceData=null, mServiceDataMask=null, mManufacturerId=ffffffff, mManufacturerData=null, mManufacturerDataMask=null]
```

The fix will be to remove the manufacturer id.  Specific steps.

1. Make the manufacturer field nullable (make it an Integer vs. int) here:
https://github.com/AltBeacon/android-beacon-library/blob/master/lib/src/main/java/org/altbeacon/beacon/service/scanner/ScanFilterUtils.java#L24

2. In `createScanFilterDataForBeaconParser`, set the manufacturer to null if the service uuid is not null.

3. Don't set manufacturer data if manufacturer is null here:
https://github.com/AltBeacon/android-beacon-library/blob/master/lib/src/main/java/org/altbeacon/beacon/service/scanner/ScanFilterUtils.java#L93

4. Update the unit test to verify the filter does not set manufacturer:
https://github.com/AltBeacon/android-beacon-library/blob/master/lib/src/test/java/org/altbeacon/beacon/service/scanner/ScanFilterUtilsTest.java#L76

@manuelmacedo, if you need a quick fix here, I might ask that you help out with making this change and issue a pull request.   If you are not in a hurry, I will get to it before the next release.

manuelmacedo, 2019-05-07:
Hi David,
I understood the necessary changes to the library. The changes are not urgent, I can use the previous version a little longer until the next release. However, if I can get some free time in the next few days, I will be happy to contribute and make the change. In that case, I'll let you know. 

Thank you for your help and for the great work with this library.

davidgyoung, 2019-05-29:
@manuelmacedo, I just got to working on this today.  I did some tests on my Samsung Galaxy S9 with Android 9.0, and I found that I cannot repeat this problem.   Using the reference app (with Eddystone-UID parser configured and the foreground service setup code uncommented to enable it).  I used the latest Android Beacon Library version 2.16.1.   I was able to detect Eddystone beacons with the screen off and scan filters active as shown in the log below:

```
05-29 12:59:45.637 16578 16578 I BeaconService: beaconService version 2.16.1 is starting up on the main process
05-29 12:59:45.768 16578 16578 D CycledLeScannerForLollipop: Using a non-empty scan filter since this is Samsung 8.1+
05-29 12:59:45.769 16578 16578 D ScanFilterUtils: making scan filter for service: 0000FEAA-0000-1000-8000-00805f9b34fb 0000feaa-0000-1000-8000-00805f9b34fb
05-29 12:59:45.769 16578 16578 D ScanFilterUtils: making scan filter with service mask: FFFFFFFF-FFFF-FFFF-FFFF-FFFFFFFFFFFF ffffffff-ffff-ffff-ffff-ffffffffffff
05-29 12:59:45.769 16578 16578 D ScanFilterUtils: Set up a scan filter: BluetoothLeScanFilter [mDeviceName=null, mDeviceAddress=null, mUuid=0000feaa-0000-1000-8000-00805f9b34fb, mUuidMask=ffffffff-ffff-ffff-ffff-ffffffffffff, mServiceDataUuid=null, mServiceData=null, mServiceDataMask=null, mManufacturerId=ffffffff, mManufacturerData=null, mManufacturerDataMask=null]
05-29 12:59:45.851 16578 16606 D ScanHelper: beacon detected : id1: 0x01000000000000000002 id2: 0x010000000002
05-29 12:59:45.945 16578 16609 D ScanHelper: beacon detected : id1: 0x01000000000000000002 id2: 0x010000000002
05-29 12:59:46.053 16578 16610 D ScanHelper: beacon detected : id1: 0x01000000000000000002 id2: 0x010000000002
05-29 12:59:46.258 16578 16611 D ScanHelper: beacon detected : id1: 0x01000000000000000002 id2: 0x010000000002
05-29 12:59:46.365 16578 16614 D ScanHelper: beacon detected : id1: 0x01000000000000000002 id2: 0x010000000002
```

I realized that my comment above about a bug in the code is incorrect.  I based this on the log lines coming from the Android OS saying a manufacturer id of FFFFFFFF was configured in the filter.  I learned in troubleshooting that this simply indicates that no manufacturer id is configured.  The filter is therefore correct.

I suspect that the problem may be that the scan filters simply are not working on your J4+.   It is not clear if this is a constant problem or an intermittent one -- it is certainly true that scan filters are a limited hardware resource and it may be that other apps had exhausted them all when you tested.

If they never work on the J4+, this would be a big problem because when the screen is off Samsung now requires the filters to be present on Android 8+, otherwise the scans are blocked.  If scan filters do not work on your J4+, the you can not do scans at all with the screen off.

With the screen on, the current 2.16.1 library release will also use ScanFilters with Samsung on Android 8+, and as a result, this would also block scans with the screen on if the scan filters do not work.  This, however, will be fixed in the next version of the library by not using scan filters with the screen on.   This change has been put in #855, and is planned for the next 2.16.2 release, expected later this week.  I suspect that #855 will fix your problem, at least with the screen on.

davidgyoung, 2019-05-29:
Closing this issue as the best fix available is already implemented in #855.  Please reopen and/or comment if you have any other ideas.

closed 2019-05-29 by davidgyoung