AltBeacon/android-beacon-library #872: Foreground service Not Scanning

joviden, 2019-05-06:
### Expected behavior
Scanning continue after app is killed and while foreground service is active

### Actual behavior
After killing app it goes on scanning with foreground service but after a while it stops scanning.

  
    @Override
    public void onCreate() {
        super.onCreate();
        Toast.makeText(getApplicationContext(), "On create.", Toast.LENGTH_SHORT).show();

        BackgroundPowerSaver backgroundPowerSaver = new BackgroundPowerSaver(this);

    }

    @Override
    public int onStartCommand(Intent intent, int flags, int startId) {
        Toast.makeText(getApplicationContext(), "On start command.", Toast.LENGTH_SHORT).show();

        try{
            beaconManager.unbind(this);
        }catch (Exception e) {
            e.printStackTrace();
        }

        beaconManager = BeaconManager.getInstanceForApplication(this);
        setBeaconLayouts();

        beaconManager.setForegroundScanPeriod(9500l);
        beaconManager.setForegroundBetweenScanPeriod(500l);
        beaconManager.setBackgroundScanPeriod(9500l);
        beaconManager.setBackgroundBetweenScanPeriod(500l);
       // beaconManager.setAndroidLScanningDisabled(true);

        try {
            beaconManager.updateScanPeriods();
        } catch (RemoteException e) {
            e.printStackTrace();
        }

        startForegroundService();

        return START_STICKY;

    }

    /* Used to build and start foreground service. */
    private void startForegroundService()
    {
        Toast.makeText(getApplicationContext(), "Search started.", Toast.LENGTH_SHORT).show();
        String CHANNEL_ID = "erdem";
        NotificationManager notificationManager = (NotificationManager) getApplicationContext().getSystemService(Context.NOTIFICATION_SERVICE);

        try {

            NotificationCompat.Builder builder;
            if (Build.VERSION.SDK_INT >= 26) {
                CharSequence name = "Channel Name";
                int importance = NotificationManager.IMPORTANCE_NONE;
                NotificationChannel channel = new NotificationChannel(CHANNEL_ID, name, importance);

                if (notificationManager != null) {
                    notificationManager.createNotificationChannel(channel);
                }

                builder = new NotificationCompat.Builder(getApplicationContext(), CHANNEL_ID)

                        .setSmallIcon(R.drawable.logo_transparent)
                        .setChannelId(CHANNEL_ID)
                        .setContentTitle("Looking for kidies.")
                        .setAutoCancel(true);

            } else {

                int icon = Build.VERSION.SDK_INT >= Build.VERSION_CODES.LOLLIPOP ? R.drawable.logo_transparent : R.mipmap.ic_launcher;
                builder = new NotificationCompat.Builder(getApplicationContext(), "erdem")
                        .setSmallIcon(icon)
                        .setContentTitle("Looking for kidies")
                        .setAutoCancel(true);

            }

            Notification notification = builder.build();
            startForeground(1, notification);
           // Çift notifikasyon yapıyor! beaconManager.enableForegroundServiceScanning(notification, 123);

            beaconManager.setEnableScheduledScanJobs(false);
            beaconManager.setRegionStatePersistenceEnabled(false);
            beaconManager.bind(this);
            beaconManager.setDebug(true);

        } catch (Exception e) {
            e.printStackTrace();
        }
    }

    @Override
    public void onBeaconServiceConnect() {
        Log.i("serviceWorking", "Service connected");

        beaconManager.removeAllMonitorNotifiers();
        beaconManager.addMonitorNotifier(this);
        try {
            beaconManager.startMonitoringBeaconsInRegion(new Region("myRangingUniqueId", null, null, null));
        } catch (RemoteException e) {
            e.printStackTrace();
        }

    }

    private void setBeaconLayouts()
    {

        final String IBEACON_LAYOUT = "m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24";
        final String IBEACON_LAYOUT2 = "m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25";

        beaconManager.getBeaconParsers().clear();
        beaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout(IBEACON_LAYOUT2));
        beaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout(IBEACON_LAYOUT));
        beaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout(BeaconParser.ALTBEACON_LAYOUT));
        beaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout(BeaconParser.EDDYSTONE_UID_LAYOUT));
        beaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout(BeaconParser.EDDYSTONE_URL_LAYOUT));
        beaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout(BeaconParser.EDDYSTONE_TLM_LAYOUT));

    }

    @Override
    public void onDestroy() {

        try {
            beaconManager.stopMonitoringBeaconsInRegion(new Region("myRangingUniqueId", null, null, null));
            beaconManager.stopRangingBeaconsInRegion(new Region("myRangingUniqueId", null, null, null));
        } catch (RemoteException e) {
            e.printStackTrace();
        }
        beaconManager.removeAllMonitorNotifiers();
        beaconManager.removeAllRangeNotifiers();
        beaconManager.unbind(this);
    }

    @Override
    public void didRangeBeaconsInRegion(Collection<Beacon> collection, Region region) {

        if (collection.size() > 0) {
         
            Log.i("checkBeacon", "The first beacon I see is about "+ String.format("%.2f", collection.iterator().next().getDistance())+"" +" meters away.");
            Toast.makeText(getApplicationContext(),"Beacon uzaklığı: "+String.format("%.2f", collection.iterator().next().getDistance())+" metre.",Toast.LENGTH_SHORT).show();
        }else {
            Toast.makeText(getApplicationContext(), "Beacon bulunamadı.", Toast.LENGTH_SHORT).show();
        }

    }

    @Override
    public void didEnterRegion(Region region) {

    }

    @Override
    public void didExitRegion(Region region) {
        Toast.makeText(getApplicationContext(),"Beacon bölgesinden çıkıldı.",Toast.LENGTH_SHORT).show();
        try {
            beaconManager.stopRangingBeaconsInRegion(region);
        } catch (RemoteException e) {
            e.printStackTrace();
        }
        beaconManager.removeAllRangeNotifiers();
       

    }

    @Override
    public void didDetermineStateForRegion(int i, Region region) {

        Toast.makeText(getApplicationContext(),"Beacon bölgesine girildi",Toast.LENGTH_SHORT).show();
        try {
            beaconManager.stopRangingBeaconsInRegion(region);
        } catch (RemoteException e) {
            e.printStackTrace();
        }

        beaconManager.removeAllRangeNotifiers();
        beaconManager.addRangeNotifier(this);
        try {
            beaconManager.startRangingBeaconsInRegion(region);
        } catch (RemoteException e) {
            e.printStackTrace();
        }

    }
IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2019-05-06:
Where is the shown code placed?  Is this in an Application class or a custom Service?

joviden, 2019-05-06:
Its a custom Service.

public class Scan_ForeGround_Activity extends Service implements BeaconConsumer,RangeNotifier, MonitorNotifier {}

davidgyoung, 2019-05-06:
@joviden,  I am happy to help you get this working, but I think this is not the proper forum, as this is likely largely a custom app debugging issue and not a specific problem with the library itself.

To summarize, you are building a custom foreground service, and inside of it starting scanning with the Android Beacon Library setting it up to use a *second* foreground service started from the first foreground service.  Then when you kill the app with the task switcher, you are seeing one or both of the foreground services not automatically restart, even though you see the icon associated with one or both.

I can confirm that I do not see these symptoms when using the library as a foreground service as recommended by the library reference app, where the service is started by a custom application class and not a second foreground service.

My first reaction is that this is an unusual architecture, and I'm not surprising you are seeing problems like this.  I think to solve the problem you need to simplify your tests case so you don't have to paste 200 lines of custom code just to explain what you are doing.  I suggest one of the following troubleshooting steps:

1. Remove the Android Beacon Library from being started by your custom foreground service.  Does the problem go away?  If not, then the problem is with your custom foreground service.
2. If no, Make a second dummy custom foreground service that is empty (just for testing).  Start it from inside your existing custom foreground service, then try to reproduce the problem.  Can you reproduce the problem?  If yes, then the problem is starting a  foreground service inside a second foreground service.

I'm going to close this issue as out of scope for this forum, but please do open a new issue if you can use the suggested steps described above to identify a more precise issue with the library.  Alternatively, if you want help troubleshooting your custom app, you might consider opening a new question on StackOverflow.com.

closed 2019-05-06 by davidgyoung