AltBeacon/android-beacon-library #876: CycledLeScanner crashes on setWakeUpAlarm

JonatanPlesko, 2019-05-15:
### Expected behavior
No crash
### Actual behavior
Fatal Exception: java.lang.SecurityException
get application info: Neither user 1010129 nor current process has android.permission.INTERACT_ACROSS_USERS.

Fatal Exception: java.lang.SecurityException
Permission Denial: getIntentForIntentSender() from pid=20646, uid=10159 requires android.permission.GET_INTENT_SENDER_INTENT

Fatal Exception: java.lang.NullPointerException
println needs a message
### Steps to reproduce this behavior
Not sure about the reproduction, we noticed these crashes on Firebase. Here's the stack trace for LENOVO VIBE X3 Lite, Android 5.1:
```
Fatal Exception: java.lang.SecurityException: get application info: Neither user 1010129 nor current process has android.permission.INTERACT_ACROSS_USERS.
       at android.os.Parcel.readException(Parcel.java:1546)
       at android.os.Parcel.readException(Parcel.java:1499)
       at android.app.IAlarmManager$Stub$Proxy.set(IAlarmManager.java:244)
       at android.app.AlarmManager.setImpl(AlarmManager.java:425)
       at android.app.AlarmManager.set(AlarmManager.java:224)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.setWakeUpAlarm(CycledLeScanner.java:489)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:368)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner$2.run(CycledLeScanner.java:374)
       at android.os.Handler.handleCallback(Handler.java:815)
       at android.os.Handler.dispatchMessage(Handler.java:104)
       at android.os.Looper.loop(Looper.java:194)
       at android.app.ActivityThread.main(ActivityThread.java:5763)
       at java.lang.reflect.Method.invoke(Method.java)
       at java.lang.reflect.Method.invoke(Method.java:372)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:960)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:755)
```

There was another similar crash on Galaxy S5 Neo (6.0.1):
```
Fatal Exception: java.lang.SecurityException: Permission Denial: getIntentForIntentSender() from pid=20646, uid=10159 requires android.permission.GET_INTENT_SENDER_INTENT
       at android.os.Parcel.readException(Parcel.java:1620)
       at android.os.Parcel.readException(Parcel.java:1573)
       at android.app.IAlarmManager$Stub$Proxy.set(IAlarmManager.java:217)
       at android.app.AlarmManager.setImpl(AlarmManager.java:484)
       at android.app.AlarmManager.set(AlarmManager.java:260)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.setWakeUpAlarm(CycledLeScanner.java:489)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:368)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner$2.run(CycledLeScanner.java:374)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:158)
       at android.app.ActivityThread.main(ActivityThread.java:7230)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1230)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1120)
```

And another one on LENOVO A6010 (5.0.2):
```
Fatal Exception: java.lang.NullPointerException: println needs a message
       at android.util.Log.println_native(Log.java)
       at android.util.Slog.v(Slog.java:28)
       at android.app.AlarmManager.setImpl(AlarmManager.java:429)
       at android.app.AlarmManager.set(AlarmManager.java:228)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.setWakeUpAlarm(CycledLeScanner.java:489)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:368)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:338)
       at org.altbeacon.beacon.service.scanner.CycledLeScannerForLollipop$1.run(CycledLeScannerForLollipop.java:153)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:135)
       at android.app.ActivityThread.main(ActivityThread.java:5233)
       at java.lang.reflect.Method.invoke(Method.java)
       at java.lang.reflect.Method.invoke(Method.java:372)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:898)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:693)
```
### Mobile device model and OS version
LENOVO VIBE X3 Lite (5.1)
Galaxy S5 Neo (6.0.1)
LENOVO A6010 (5.0.2)
### Android Beacon Library version
2.15.4

davidgyoung, 2019-05-15:
It looks like this happens on devices that allow multiple Android users, and occurs when the first user who started the app is no longer logged in, yet the app is still running.  The AlarmManager throws the first exception in that case.  

Unfortunately, I can find no discussion of how this exception can be avoided in this case.

https://stackoverflow.com/questions/37062702/setting-alarms-on-devices-with-multiple-users

davidgyoung, 2019-05-22:
It looks like it is possible to detect when a user switches on Android:  https://stackoverflow.com/questions/15392126/how-to-detect-switching-between-users

This bug might be fixed by adding in code like that and cancelling all alarms in the event of a user switch. That said, this is not a common crash, and its cause is pretty obscure.  So implementing this is not a high priority for the library's development track.

If it is important to prevent this crash in a third party app, that app can also look for a user switch using the technique above, and stop all monitoring/ranging when it happens, and the service will stop and the alarm will be cancelled.

davidgyoung, 2019-05-29:
Fixed in #886

PR #886 (2019-05-29): Prevent crash on alarms going off with a different user active
This attempts to fix #876, which reports a SecurityException crash when a library Alarm goes off after somebody switched users on the phone.  The fix is to detect the user switch and cancel the alarm.

Testing this on the Google Pixel 3a, I start the reference app (with a foreground service enabled), then switch to the Guest User.  I see that the logs show the alarm gets cancelled:

```
05-29 13:44:14.588  3760  3760 W CycledLeScanner: User switch detected.  Cancelling alarm to prevent potential crash.
05-29 13:44:14.588  3760  376 [... cut]
closed 2019-05-29 by davidgyoung