AltBeacon/android-beacon-library #877: Couldn't detect Android's iBeacon signal using a firmware

Serry92, 2019-05-15:
I am using the library for two purposes.
first, I  make my android device to transmit an iBeacon signal with a specific UUID, major, minor.
I expect that the firmware's light is to blink but the actual result nothing happened. The firmware doesn't detect my signal.

The application works as expected on some Android devices, and doesn't work on others.

Note: At the same time, my app searches for available iBeacons (ranges for beacons).

An example of a device that doesn't work (The firmware doesn't detect this mobile's signal):

-Model number: Huawei P10 
-Android OS:8.1
-Android beacon library version:2.15..4

An example of a device that works:
Model number: Aquaris E4.5
Android Version: 5.0

davidgyoung, 2019-05-15:
@Serry92,  I will surely need more information to be able to help here:

* Please show the code you use to transmit.  
* Please confirm if you check for errors on starting transmission, and if you get errors on some devices, please show what they are.
* Please use a second Android device to try and detect the transmission (not just the firmware) and confirm whether the second Android device can detect the transmission.

Serry92, 2019-05-16:
@davidgyoung Thank you for your quick reply.

I contacted the client (the firmware producer) and after they debugged the frames they receive from the Android app they noticed a difference between the frames sent by different mobiles.  

The following image is an example of a (correct) frame sent by an Android device and received by the firmware.

The firmware expects the iBeacon UUID to start at position 9, and this is exactly the case in the attached image.

However, some Android devices send the frame in a different structure where the UUID starts at position 6 not 9.

For example:

-Model number: Huawei P10
-Android OS:8.1
-Android beacon library version:2.15..4

sends the UUID starting from position 6.

While:

Model number: Aquaris E4.5
Android Version: 5.0

sends the UUID starting from position 9.

How can we force all Android devices to send the data in the same structure?

Thank you very much once again for your support. 

[image]

davidgyoung, 2019-05-22:
I would have to see the bytes of the other case to be sure (the case where the uuid starts at position 6), but I suspect that device does not send out the first three bytes 02 01 06.  These three bytes are a Flags AD type, and while typical, I don't think it is absolutely required for a iBeacon frame.  What's more, I do not believe there is any way to programmatically tell an Android device whether or not to include the Flags AD type in the advertising packet.  Most devices will have it, but some (as you have seen) will not.

The *right* way to fix this is to make your firmware a bit more sophisticated, and let it look for the right AD type to parse.  The algorithm would be like this:

1. Set a pointer to the start of the advertisement
2. Read the length of the first PDU from the byte at the pointer position
3. Read the AD type from the byte after the pointer position
4. Is the AD type == FF (manufacturer data) ?  If no, advance the pointer by the length from step 2 and go to step 2.  If yes, go to step 5.
5. Advance the pointer by 6 and start reading the UUID

closed 2019-05-23 by davidgyoung