AltBeacon/android-beacon-library #879: stop advertising

Serry92, 2019-05-20:
### Expected behavior
I expect to stop immediately when I stop advertising
### Actual behavior
when I stop advertising it takes 3 or 4 seconds to stop.

davidgyoung, 2019-05-20:
What device model and operating system version are you using for the transmitter?  What are you using to measure the time it takes for the transmission to stop?

Serry92, 2019-05-21:
@davidgyoung thank you for reply 

### Mobile device model and OS version
-Model number: Huawei P10
-Android OS:8.1

### Android Beacon Library version
-Android beacon library version:2.15..4

I know it takes that time by reading logs of other devices ranging my signals even if after I stop advertising

davidgyoung, 2019-05-22:
Unfortunately, this is probably a Huawei P10-specific issue.  The Android Beacon Library's BeaconTransmitter class is simply a convenience wrapper around built-in Android APIs that start and stop beacon advertising.  The library class does nothing to affect the timing of starting and stopping the advertisement -- it just provides easy conversion of beacon identifiers of specific beacon layouts to proper beacon BLE advertising frames.

I do not see such delays in stopping transmission on several other Android device models I have tested, including the Nexus 4, 5, 5X, Pixel 3A, Samsung Galaxy S7-S9, Nokia 6.1+ and Huawei P9 w/ Android 7.  On those devices, stopping transmission happens very quickly -- certainly within 100ms.

closed 2019-05-29 by davidgyoung