AltBeacon/android-beacon-library #880: The txpower value does not appear.

wodud95, 2019-05-22:
### Expected behavior
When you use getTxPower () in ScanResult.class, only 127 txpower values ​​will pop up.

### Actual behavior

### Steps to reproduce this behavior

### Mobile device model and OS version
android 8.0.0
### Android Beacon Library version
android-beacon-library-2.15.4

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2019-05-22:
The ScanResult class is an internal Android class, not one this library provides.  If you look at the documentation for it, it indicates that a value of 127 indicates there is no TxPower present -- this means that the Bluetooth LE AD type is not in the advertising packet.  This is expected for standard beacon frames like AltBeacon, iBeacon and Eddystone frames.

https://developer.android.com/reference/kotlin/android/bluetooth/le/ScanResult#getTxPower()

closed 2019-05-22 by davidgyoung

prayagupd, 2019-11-10:
I was having same issue with android standard library. and happens stdlib does not provide `txPower`. `127` is default value.

```java
    public ScanResult(BluetoothDevice device, ScanRecord scanRecord, int rssi,
            long timestampNanos) {
        mDevice = device;
        mScanRecord = scanRecord;
        mRssi = rssi;
        mTimestampNanos = timestampNanos;
        mEventType = (DATA_COMPLETE << 5) | ET_LEGACY_MASK | ET_CONNECTABLE_MASK;
        mPrimaryPhy = BluetoothDevice.PHY_LE_1M;
        mSecondaryPhy = PHY_UNUSED;
        mAdvertisingSid = SID_NOT_PRESENT;
        mTxPower = 127;
        mPeriodicAdvertisingInterval = 0;
    }
```